indilib/indi-3rdparty #863: indi-rpicam and aarch64

aaronwmorris, 2023-12-04:
I was attempting to compile 2.0.5 on a Raspberry Pi OS 12 (bookworm) 64-bit aarch64 system and it was failing to compile the indi-rpicam server due to libmmal being an incorrect binary format.

It is my understanding that the libmmal library is 32-bit only on the armv6l and armv7l platforms.

From CMakeLists.txt
```
# MMAL Required for Raspberry PI camera driver
if (CMAKE_SYSTEM_PROCESSOR MATCHES "armv+" OR CMAKE_SYSTEM_PROCESSOR MATCHES "aarch64")
find_package(MMAL)
endif()

IF (MMAL_FOUND)
option(WITH_RPICAM "Install Raspberry Pi High Quality Camera Driver" On)
ELSE ()
option(WITH_RPICAM "Install Raspberry Pi High Quality Camera Driver" Off)
ENDIF(MMAL_FOUND)
```

Should the MMAL library detection be removed for the aarch64 platform?

As a side note, I have libcamera compiled from source on this system.  It might not be installed otherwise.

knro, 2023-12-05:
indi-rpicam is deprecated and is partially replaced by indi-libcamera. I think we should remove indi-rpicam from the repo.

aaronwmorris, 2023-12-05:
No argument here.  :-)

closed 2024-01-17 by knro