indilib/indi-3rdparty #867: ZWO ASI Camera - GPSStartLine invalid value

aaronwmorris, 2023-12-17:
INDI 2.0.5  (may affect 2.0.4 as well)

The latest ZWO ASI SDK introduces some new variables which do not have sane defaults, resulting in the camera not operating properly.

The two new variables under the `CCD_CONTROLS` property are `GPSStartLine` and `GPSEndLine`.  The default value of GPSStartTime is invalid causing other variables like Gain to be unable to be set unless GPSStartLine is set to a valid value, such as 0.

There are no GPS devices, so I do not think the "GPS" string refers to global positioning system.

```
ZWO CCD ASI224MC.CCD_CONTROLS.Gain=200
ZWO CCD ASI224MC.CCD_CONTROLS.WB_R=52
ZWO CCD ASI224MC.CCD_CONTROLS.WB_B=95
ZWO CCD ASI224MC.CCD_CONTROLS.Offset=1
ZWO CCD ASI224MC.CCD_CONTROLS.BandWidth=80
ZWO CCD ASI224MC.CCD_CONTROLS.AutoExpMaxGain=300
ZWO CCD ASI224MC.CCD_CONTROLS.AutoExpMaxExpMS=30000
ZWO CCD ASI224MC.CCD_CONTROLS.AutoExpTargetBrightness=100
ZWO CCD ASI224MC.CCD_CONTROLS.HardwareBin=0
ZWO CCD ASI224MC.CCD_CONTROLS.HighSpeedMode=0
ZWO CCD ASI224MC.CCD_CONTROLS.MonoBin=0
ZWO CCD ASI224MC.CCD_CONTROLS.GPSStartLine=53248
ZWO CCD ASI224MC.CCD_CONTROLS.GPSEndLine=259
ZWO CCD ASI224MC.CCD_CONTROLS_MODE.AUTO_Gain=Off
ZWO CCD ASI224MC.CCD_CONTROLS_MODE.AUTO_WB_R=Off
ZWO CCD ASI224MC.CCD_CONTROLS_MODE.AUTO_WB_B=Off
ZWO CCD ASI224MC.CCD_CONTROLS_MODE.AUTO_BandWidth=Off
<snip>
ZWO CCD ASI224MC.SDK.VERSION=1, 32, 0, 0
```

The variables are only available on certain camera modules and not others.  So far, I have seen reference to it affecting the ASI224MC and ASI1600.

It apparently does **not** affect the ASI678 and ASI2600.

```
Dec 14 10:45:40 hostname [INFO] Capture-29-8763/Dummy-3 indi.newMessage(): new Message 2023-12-14T08:45:40: Error: Invalid range for GPSStartLine (GPSStartLine). Valid range is from 0 to 975. Requested value is 53248
```

aaronwmorris, 2023-12-17:
Reference to the issue in the forums:

https://www.indilib.org/forum/ccds-dslrs/14057-asi1600-and-gpsstartline.html

knro, 2023-12-18:
Thank you for the report. Should we explicitly set this to zero on startup ?

knro, 2023-12-18:
This is documented in the header file:
```
ASI_GPS_DATA* startLineGPSData: the GPS data of the start line. the start line number is set by ASISetControlValue(..., ASI_GPS_START_LINE,...). the default value is 0
ASI_GPS_DATA* endLineGPSData: the GPS data of the end line. the end line number is set by ASISetControlValue(..., ASI_GPS_END_LINE,...). the default value is MaxHeight - 1

```

so this only affects GPS-enabled cameras or what?

jpaana, 2023-12-18:
The controls are directly enumerated from the SDK as are their default values so we don't specifically handle the GPS controls in any way. It definitely seems like a bug in the SDK that they offer those controls for cameras that don't have GPS functionality at all. I guess I could add a workaround to initialize those two controls to the values they are supposed to contain (0 and max height - 1) if they are present.

aaronwmorris, 2023-12-18:
As far as I know, there are no current ZWO cameras with GPS functionality.  Maybe this is an indication that some new cameras are going to have GPS functionality.

I wonder if the binning setting would affect these settings since it would change the geometry of the sensor.

daleghent, 2023-12-18:
Yes we ran into this in NINA when updating the SDK for its native ZWO driver recently. It's very annoying. I posted about the issue of every camera acting like it has a GPS on the ZWO forums but essentially got a non-answer regarding it. There appears to be an additional bug with the error code it throws sometimes.

closed 2024-01-17 by knro