indilib/indi-3rdparty #873: PlayerOne Ceres-M - Ekos capture hangs on sub second exposure loop

MarcinL1977, 2023-12-21:
I cannot loop sub second exposures in Ekos with PlayerOne Ceres-M camera. I am using Raspberry Pi5 board as a control machine. From the software side I have tested both: Stellarmate and Astroarch. Connection to Raspberry is through VNC.

Exact steps to reproduce the behavior:
1. Kstars / Indi server run with driver playerone_ccd
2. Exposure time set to below 1 second (1 second and above are fine)
3. Single exposures are fine, but capturing will freeze when loop is being used
4. Indi log in camera section produced a warning saying that "PlayerOne firmware might require an update to *compatible mode - I have found no info on how to potentially update firmware
5. Then there is a repeated error occurring saying POA_ERROR_EXPOSING 

I have tried to change USB ports from 3.0 to 2.0 and vice versa. I have also used powered usb hub, with no luck. I can get to sub seconds exposures when I set delay option to 1, but this, I believe is not helping my case, as the goal is to get shorter than 1sec exposures for guiding. I also played a bit with fast exposure options and usb bandwidth in Indi driver , but with no effect.

To my surprise, connecting to Indi server from remote client, does allow to produce sub second exposures with no issues. There is however 1sec download time in my case (wifi), so it won't help.

DMX Capture software can run this camera with 13fps with no problems, so it seems that hardware is fine.

I apologise if this is not the right place to ask for help. I tried to post on the forum, but new users registration seems broken at the moment.

ijessen, 2023-12-21:
+1.  I also have a Ceres-M.  I also experience the same behavior.  I haven't dug into it yet but will try to next week.

(I also can't sign up for the forum either - are we twins?)

PlayerOneAstronomy, 2023-12-22:
@MarcinL1977 @ijessen Sorry for the issue. I have informed our software engineer about this issue and he has already started working on it.

hiro3110i, 2023-12-29:
@MarcinL1977 Could you please give me a little more information about your conditions?
What kind of OS/Linux are you using on Raspberry Pi5?
Actually, RasPi5 is very new so I'm not sure about it.
Are you testing the latest KStars/Ekos v3.6.8 (latest version is build 2023-12-26) with indi_playerone_ccd v1.12?
Is your capture loop running on "CCD tab (camera icon)" of Ekos?

I don't have Ceres-M but I can test with Sedna-M on my Raspberry Pi4 (Ubuntu Mate 64bit).
I also use VNC from my MacBook Air or Windows11 PC.
On my Raspberry Pi4, shorter exposure than one second is working well.
However, Ceres-M might have different behavior I understand.
So I need more information about your reproduce environment/procedure.
Debug logs related to firmware updates are leftovers but there is no need to worry about them at this moment.

Thank you for your cooperation and apologize for any inconvenience.

MarcinL1977, 2023-12-29:
I appreciate you looking into it. I am out for a couple of days, will report back next week.

I am using archlinux for the tome being, thanks to astroarch release, but can also switch to stellarmate.
I am experimenting a bit. I will try to install pure debian or kubuntu and see how it works.

I have some further observations from one night of testing:

camera works fine in a single capture modes with fast and long exposures. Loop in ekos capture module will not work below 1 sec. Guide module with 1sec and above works, but loop in guide module doesn’t at all, no matter the exposure times. Polar alignment fails, as refresh sooner or later stops and I tried that with longer exposures - initial plate solving is ok, but then, when you need to refresh to see the effect of corrections it will fail, after say image number 5 or 6. Polar alignment with my old Atik 383 worked fine.

Next week, I will try to provide more systematic observations and more importantly… logs.

Happy New Year!
Marcin

MarcinL1977, 2023-12-29:
Missed your question, apologies. It is recommended to use shorter exposures for AM3/AM5 mounts. I tested with 1 sec and dark guiding set to 0.5 sec and I got to RMS of say 1-1.3, which for very light mount like AM3 is, I think, very good and ok for my needs. I may also not get enough stars with 0.5sec exposure, so perhaps my problem is academic 😉

hiro3110i, 2023-12-30:
@MarcinL1977 Thank you for your report.
I'm discussing with POA software engineer.
New SDK will release soon, I'll test it as soon as possible at the beginning of next year.

Have a happy new year!

hiro3110i, 2024-01-09:
@MarcinL1977 indi_playerone_ccd driver was updated by SDK v3.6.1.
It was tested with Sedna-M, Neptune-C II, Uranus-C, Saturn-C and Poseidon-C on my Raspberry Pi4 (Ubuntu Mate 20.04).
Could you please try updated driver on your side?

MarcinL1977, 2024-01-10:
I managed to build libplayerone, but indi-playerone fails. It seems it complains about CFITSIO missing. Here is the build process console output:

sudo cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-playerone/
-- The CXX compiler identification is GNU 13.2.0
-- The C compiler identification is GNU 13.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PLAYERONE: /usr/lib/aarch64-linux-gnu/libPlayerOneCamera.so;/usr/lib/aarch64-linux-gnu/libPlayerOnePW.so
-- CFITSIO not found.
-- Found INDI: /usr/lib/aarch64-linux-gnu/libindidriver.so;/usr/lib/aarch64-linux-gnu/libindiAlignmentDriver.so (found version "2.0.5") 
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found version "1.2.13")  
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Found USB1: /usr/lib/aarch64-linux-gnu/libusb-1.0.so  
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Configuring done (1.2s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CFITSIO_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
   used as include directory in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
CFITSIO_LIBRARIES (ADVANCED)
    linked by target "indi_playerone_single_ccd" in directory /home/marcin/Projects/indi-3rdparty/indi-playerone
    linked by target "indi_playerone_ccd" in directory /home/marcin/Projects/indi-3rdparty/indi-playerone

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "indi_playerone_ccd":

    "CFITSIO_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "indi_playerone_single_ccd":

    "CFITSIO_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "indi_playerone_wheel":

    "CFITSIO_INCLUDE_DIR-NOTFOUND"

CMake Error in CMakeLists.txt:
  Found relative path while evaluating include directories of
  "playerone_camera_test":

    "CFITSIO_INCLUDE_DIR-NOTFOUND"

MarcinL1977, 2024-01-10:
Seems that libcfitsio-dev was needed,; build succeeded.
Testing the driver now.

hiro3110i, 2024-01-11:
@MarcinL1977 Thank you for testing.
Both of previous v3.6.0 and the latest v3.6.1, your issue was not reproduced with
Sedna-M, Neptune-C II, Uranus-C, Saturn-C and Poseidon-C at least in my environment (Ubuntu Mate 20.04).
Sorry, I don't have Ceres-M.
Manufacture's engineer had tested with Ceres-M in KStars/Ekos but not reproduced it, I heard.
Do you have any other PlayerOne cameras?
Does anyone have Ceres-M and can test it?

Difference is camera itself, RasPi5 and distribution of linux.
Actually, Ceres-M is not using sony's sensor but others which I tested are sony sensor camera.
Do you have RasPi4? I'm testing on RasPi4/4GB SBC.

MarcinL1977, 2024-01-11:
@hiro3110i 

Unfortunately, this is the only PlayerOne camera I own.
Platform is RasPi5/4GB.
System used yesterday was Ubuntu 23.10.
Indi was pulled from Jasem's PPA, libplayerone and indi-playerone were built based on updated git.
Same behaviour on Archlinux with AstroArch was observed.

PlayerOneAstronomy, 2024-01-12:
@MarcinL1977 
Hi, our engineer have not reproduced this problem on AstroArch and PiOS on Pi4, however, we currently do not have a Pi5 available.
In order to find out what is causing this problem we would like to do some tests on your Pi5 remotely,  is it convenient?
If you agree, tell me your email and our engineers will contact you later.

MarcinL1977, 2024-01-12:
My email: marcin.laba@gmail.com,
Regards

hiro3110i, 2024-02-27:
@MarcinL1977 Finally, I could buy RasPi5/4GB here in Japan. I have just started testing on RasPi5.
At the starting point, I'm trying Ubuntu 23.10. Generally Ubuntu 23.10 is working well on RasPi5.

However, I found RasPi5 is heavy electricity eater and USB bus power is limited when I give poor power source from USB-C port.
Power source problem is also very sensitive for camera operation in Ekos.
Poseidon-C needs 12V power source even if cooling system is off. (give 12V first and then connect USB cable) 
RasPi4 can drive Poseidon-C only with bus power but RasPi5 can not when I use very general AC adapter.
Other non-cooled cameras (Saturn-C/Uranus-C/Neptune-C II/Sedna-M) can work only bus power but I had to use enough power source when I want to use short exposure.
My conclusions are that POWER SOURCE IS SUPER SENSITIVE for RasPi5 anyway.
Instead, RasPi5 has the advantage of significantly reducing image download time.

Could you please check the capabilities of power source for RasPi5?
For example, I could loop very short exposure when I use AC adaptor of MacBook Pro.
But I could not drive the cameras with iPhone's AC adaptor.

MarcinL1977, 2024-02-29:
Hi, my board is embedded into Astrolink by Astrojolo device at the moment and I do not want to disasseemble it. It has a very decent power source. Did PlayerOne engineer reach out to you after the testing? We did remote test as you requested. If I am not mistaken the conclusion was that there are dropped frames, which Ekos does not handle well. I have no issues with short exposures in other software. AstroDMX runs stable with 13fps.

hiro3110i, 2024-02-29:
@MarcinL1977 I'm glad you found an alternative, thank you very much for your effort.
Since then, I have continued to consider this while consulting with PlayerOne engineers.
Now my Sedna-M is running stably and continuously at around 35fps@8bits/20fps@16bits with Ekos on Ubuntu 23.10/RasPi5 (at full resolution of Sedna-M bin1=3096×2078 pixels which is much bigger picture)
After all, my conclusion is that it strongly depends on the RasPi5's power supply.
(The code has not changed at all)

On the other hand, Ubuntu 23.10 is still in the development phase and I think it would be wise to wait for 24.04 LTS in general.

Could you please close this topic if you never mind?

closed 2024-03-01 by MarcinL1977