indilib/indi-3rdparty #878: Touptek camera resolution issue on RPi

vladtcvs, 2024-01-02:
**Describe the bug**
When I use my Omegon Vetec Pro 16000M camera on PC (amd64), I see full resolution 4640x3506. But when I use it on RPi (armhf), I see resolution 1536x1160. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect camera to RPi
2. Run driver (omegon or toupcam)
3. See low resolution on RPi

**Expected behavior**
Full resolution on RPi

**Screenshots**
PC: 
[image]

RPi (remote connection):
[image]

**Desktop (please complete the following information):**
 - OS: Debian on PC and Raspbian on RPi
 - Version: PPA version 2.1+t202311131302~ubuntu23.04.1 on PC
                   [4ddc7a0](https://github.com/indilib/indi-3rdparty/commit/4ddc7a01881948fd4f1a33f03baf14184cbd449c) on RPi

vladtcvs, 2024-01-02:
I've tested `demosimplest` sample from Omegon SDK, both on PC and RPi, and for bot cases I see `4640x3506` resolution:
```
event callback: 0x0001
...
press ENTER to exit
event callback: 0x0001
pull image ok, total = 1, res = 4640 x 3506
```

vladtcvs, 2024-01-02:
Sorry, after some searching inside configs I've founded that it was config for use low resolution

closed 2024-01-02 by vladtcvs

knro, 2024-01-03:
I made this change on purpose because capture on high resolution was failing by default on armhf. You can still change resolution but it just defaults to low-res.