indilib/indi-3rdparty #881: RpiGPIO driver crashes when loading a saved configuration

FireNWater, 2024-01-11:
After Saving a Configuration under the Options tab, any attempt to Load results in driver crash.  If there are no Configuration files in the /home/stellarmate/.indi/ folder, the driver does not crash.  

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Save a Configuration <Options> tab
3. Load a Configuration <Options> tab
4. Crash

**Expected behavior**
1. I expected the driver to load the configuration I had set up just before clicking "Save"

**Screenshots**
I have included copies of a fresh configuration file(s) that the driver produces on the first run after deleting the previous configuration files from the /.indi/ folder.

**Desktop (please complete the following information):**
-StellarMate OS ver 1.8.0 / Stable / SM-110
-Rpi 4 / 8GB

**Log Files**
Make sure to enable logging and include log files [Rpi_GPIO_INDI_Logs_Crash.zip](https://github.com/indilib/indi-3rdparty/files/13907463/Rpi_GPIO_INDI_Logs_Crash.zip)

knro, 2024-01-12:
Thank you for the report. I also think the driver is missing configuration for Raspberry PI 5 + CM4. I hope @ken-self knows what's going on, I'll try to also debug from my side.

ken-self, 2024-01-12:
Looks like pigpio does not work with RPi 5 due to archiitectural incompatibility.
https://github.com/joan2937/pigpio/issues/589

knro, 2024-01-14:
It appears that this driver should probably migrate to libgpiod. Yesterday, I migrated a driver for my rolloff roof to this and it was pretty easy. For example:

```
auto chip = gpiod::chip("gpiochip0");
auto open_line = chip.get_line(FULL_OPEN_PIN);
open_line.request({"ikarus_roof", gpiod::line_request::DIRECTION_INPUT, 0});
full_open_state   = open_line.get_value();
```

I've added [FindGPIOD.cmake](https://github.com/indilib/indi-3rdparty/blob/master/cmake_modules/FindGPIOD.cmake) so that 3rd party projects can link against the C++ GPIO bindings.

knro, 2024-01-23:
Anyone willing to volunteer migrating this to the new libgpiod?