indilib/indi-3rdparty #886: Ubuntu 23.10 packages

ajeb78, 2024-01-22:
I'm not sure if this is the right place to report this or not, but here goes...
On current kubuntu 23.10 with the mutlaqja ppa added and git update done, `apt install indi-full` fails with the error:
```
The following packages have unmet dependencies.
  indi-full : Depends: indi-toupbase but it is not installable
E: Unable to correct problems, you have held broken packages.
```
`apt search toup` reports no packages named indi-toup*, however there is a libtoupcam package available. Are there some packaging dependencies that need updating?

MarcinL1977, 2024-01-24:
I can't provide a solution, but I have applied a workaround, having the same issue with fresh 23.10 installation. 

Indi-full is a meta package -  `sudo apt show indi-full` should show you the list of packages it tries to install. You can install couple of core ones - I can't check now, but from memory: indi-core, indi-data, indi-bin (libindi-... might be the correct prefix). Installation should go ok and then if you need any specific driver from 3rdparty, you can also install it manually. 

I have a working indi instance on ubuntu 23.10 with Atik, PlayerOne, ZWO mount, EAF and and QHY wheel installed the above described way.

ajeb78, 2024-01-25:
Yes, that's what I ended up doing, I installed the specific packages I needed and then built libtoupcam and ind-toupbase from source. I just thought I'd report it in case nobody had noticed that the packages were broken...

knro, 2024-03-17:
This should be fixed in the PPA.

closed 2024-03-17 by knro