indilib/indi-3rdparty #887: Player One Poseidon-C gain setting

ijessen, 2024-01-23:
**Describe the bug**
Setting full gain (550) does not appear to have the expected effect of producing .003 e-/ADU.  

driver: indi_playerone_ccd v1.13
library: libPlayerOneCamera.so.3.6.1

Imaging results suggest the actual gain is something 190 or less (no less than .125 e-/ADU).  

It's unclear to me whether the issue is related to driver, SDK, or camera hardware.  My instinct says SDK, but I figured I'd open the issue here since I know both @PlayerOneAstronomy and @hiro3110i may be able to chime in.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Pull latest, build, and install both library and driver
2. Run driver, launch and connect Ekos
3. Set gain to 550
4. Do not touch offset (left at default, out of the box, value of 20)
5. Enable LRN mode (not sure if necessary, but that was the mode I was shooting in)
6. Confirm "auto-gain" is disabled
7. Image in 16-bit RAW mode

**Expected behavior**
Images should reflect .003 e-/ADU by exhibiting a minimum step size between differently-illuminated pixels of 333 ADU (with expected variance due to fixed pattern and read noise).  Instead, examination of the resulting images put a ceiling on the minimum step size of only 8 ADU, corresponding to a minimum .125 e-/ADU.

**Screenshots**
The following screenshot shows statistics of a small selection of pixels within the vignetted corner of a 30" image.  This image was shot at 550 gain and 20 offset.  Testing shows that each offset unit corresponds to an 8 ADU increase in bias, therefore the min value of each pixel in this scenario (absent noise) is 160 (this is confirmed by darks).  Note the mean and median values in this selection are 168, min 161, and max 175.  This would make sense for pixels that are being illuminated by a small number of photons, where the e-/ADU value is anywhere above .125.  An e-/ADU of .125 would correspond to most pixels in this selection being illuminated by a single photon, some being illuminated by two, and some by none.

[image]

The next screenshot shows statistics of a small selection of pixels at the heart of a bright star (located below the statistics window) from the same image as above.  The max pixel value within this star is 702.  If the gain of 550 were in fact applied, that would correspond to maaaaybe 2 photons received from this bright star in 30".  There also would be no explanation for the appearance of gradients and other signal (see the nearby galaxies?) at brightnesses between the heart of this star and the background.

[image]

FITS header confirms the gain setting of 550

[image]

Finally, the raw FITS file is attached.

[M_86_Light_427.zip](https://github.com/indilib/indi-3rdparty/files/14027278/M_86_Light_427.zip)

ijessen, 2024-01-23:
[This ](https://github.com/indilib/indi-3rdparty/blob/3bb903c36c37352856c86b2e5ee3dcfe60adc66f/indi-playerone/playerone_base.cpp#L789) at least explains why trying to reset the value to the current value doesn't  work.

ijessen, 2024-01-23:
Root cause identified.

The new SDK exposes capabilities to enable auto mode for gain and the 3 white balance channels.  Existing driver code, which is now triggered by these new capabilities, treats all values as floats.  In fact, all the existing auto-able config parameters (gain, WB_R, WB_G, WB_B) are integer valued (see [here](https://github.com/indilib/indi-3rdparty/blob/3bb903c36c37352856c86b2e5ee3dcfe60adc66f/libplayerone/PlayerOneCamera.h#L93)).

I'll open a PR with the extremely simple fix, but if future revisions of the SDK ever introduce auto capabilities for float-type values, new logic will be required.

PlayerOneAstronomy, 2024-01-24:
@ijessen 
Sorry for this problem. 
I have forwarded this issue to our engineer, and he is investigating the cause of the problem.

knro, 2024-01-24:
Thank you for the PR @ijessen Can you recheck?

ijessen, 2024-01-24:
@hiro3110i at least for my Poseidon-C the issue is only triggered by the update to SDK 3.6.1.  The issue doesn't exist when I downgrade the SDK to 3.6.0 (while leaving the driver version at 1.13).  The new SDK exposes auto gain for the Poseidon-C, which in turn triggers the driver code path with the bug.

@knro looks good.  I built and installed the driver from master and the issue is resolved.  One question from me - since I'm by no means a C++ expert - why doesn't the compiler balk at `confVal.intValue = num.value`?  Shouldn't it require an explicit cast from the double to the long?  [Elsewhere in the driver ](https://github.com/indilib/indi-3rdparty/blob/d88b4b60553b7486e4399fed1b2238b38b73b8c4/indi-playerone/playerone_base.cpp#L795)the same operation uses an explicit `static_cast`.

closed 2024-01-24 by ijessen