indilib/indi-3rdparty #889: celestronaux: flipped SCT focuser direction

ijessen, 2024-01-24:
Unlike a refractor, which focuses by shifting the imager to align with a fixed focal plane, an SCT focuses by shifting the focal plane to align with a fixed imager.  This requires some mental gymnastics to align with INDI's refractor-based focusing convention:

> By definition, position 0 (zero) is where the focuser is completely retracted (i.e. closest to the OTA) and it increases positively as it moves outwards

Moving an SCT primary mirror forwards towards the corrector plate - by turning the focus knob CCW - shifts the focal plane backwards.  This is equivalent to moving the imager forwards in relation to the focal plane (inside focus).  

With an SCT, the final focus movement should always be towards the corrector plate, i.e. anti-gravity.  This takes up backlash and ensures that the mirror support is firmly engaged with the upper profile of the focuser shaft threads, and therefore cannot shift down under the influence of gravity.

The Ekos AF Overscan applies a final inwards movement to all outwards focusing - exactly what is needed in the SCT case.  But only as long as the directions are all properly aligned to the same convention.  Unfortunately, the encoder of the Celestron SCT Focuser counts _down_ for CCW rotation of the focus knob, exactly opposite the INDI convention.

knro, 2024-01-25:
Thank you for the PR. I think we need to update the definition to accommodate SCTs and other optical systems.

ijessen, 2024-01-25:
It's not really user facing - only appears in the dev documentation.  But if it's helpful, here's some possible language:

By definition, position 0 (zero) is the intrafocal limit, where the imager is positioned on the OTA side of the focal plane.  This corresponds to a fully retracted focuser draw tube.  In the case of an SCT with moving primary mirror, the intrafocal limit is achieved when the mirror is fully traversed toward the corrector plate.  Focus values increase positively in the extrafocal direction (i.e. extending draw tube or SCT mirror moving toward rear cell).

closed 2024-01-25 by ijessen