indilib/indi-3rdparty #891: Player One Saturn-C times out every second capture attempt.

glundby, 2024-01-30:
**Describe the bug**
Camera Player One Saturn-C (IMX533)
indi_playerone_ccd     version 1.12   interface 6    sdk 3.6.0  
Times out every second capture attempt after a few minutes. Next capture proceeds normal with download, next hangs after count down to zero and times out after several minutes and so on.

**To Reproduce**
Make sequence of 10 subs a 10s (1s does not reveal the problem!)
Start sequence and observe every second sub to timeout

**Expected behavior**
Rolling back to indi_playerone_ccd    version 1.0   interface 6   SDK 3.1.1.0 
The camera behaves as expected and captures any sequence without any problems!

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: StellarMate 1.8.0
 - Version Kstars 3.6.8 Stable

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

glundby, 2024-01-30:
I upgraded to RPi5 to see if the better HW might solve the problem, but no!

hiro3110i, 2024-01-30:
@glundby Thank you for reporting the issue.
Now I have Saturn-C in my hand, and I tested it with the latest indi_playerone_ccd v1.13 (SDK v3.6.1).
It is working with no problem in my environment (Ubuntu Mate 20.04 on RasPi4/4GB).
Could you please contact to PlayerOne Support? They will have a solution for individual Saturn-C.

By the way, RasPi5 cannot tested at this moment because I cannot buy it in Japan now due to legal issue of wifi radio.
I'll test RasPi5 as soon as possible if I'll be able to buy it here.

glundby, 2024-01-30:
Interesting!

How can I update one step to v1.13 and SDK to 3.6.1?
I would like to try it.

As for RPi5, It runs everything but much faster.  I'm even able to run FireCapture with the PO Saturn on it.

Regards! Gustav Lundby

hiro3110i, 2024-01-31:
@glundby At first, you should contact to PlayerOne support to update the firmware.
It will be the most important thing to solve the Saturn-C issue.

Current distributed version is v1.12 which is included in KStars v3.6.1.
You have to build it from source code if you want to use v1.13 now.

glundby, 2024-02-01:
Hello!

Thank you! version 1.13 of the driver works! Problem solved!

Regards!

Gustav Olav Lundby

hiro3110i, 2024-02-01:
@glundby I'm glad to hear the words "SOLVED"!
Thanks again for your contribution.
Could you please close this thread?

hiro3110i, 2024-02-15:
@knro This issue was solved by firmware update.
Could you please close this issue?

closed 2024-02-15 by knro