indilib/indi-3rdparty #893: Non functional PI5 gpio

dgedgedge, 2024-02-02:
**Describe the bug**
Indi seems to crash or does not respond anymore to kstars.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run kstars
2. Ekos add indi gpio then run
3. Indi windows does not come up

**Expected behavior**
Indi windows pops up and show gpio.

Note: There has been some change in gpio manageent in recent kernels, and in particular with GPIO.
Note: With Python RPI.GPIO does not work anymore. Need to use gpiod.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: StellarMate OS 1.8.1
 - Version : Updated to INDI Library v2.0.6 and KStars 3.6.9

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2024-02-02:
A new driver is required.

dgedgedge, 2024-02-03:
OK : gpiod driver for example ?

knro, 2024-08-07:
Please try the new INDI GPIO driver.

closed 2024-08-07 by knro