indilib/indi-3rdparty #894: EQMod park does sometimes not move to park position

pmneo, 2024-02-03:
**Describe the bug**
My EQ6R does sometimes not move to the park position. It just stops at the last position and is marked as "parked".

This happens 1 of 4-5 times in the last couple of nights. I have attached the video of my roof cam of this night and attached the logs about the time, when the mount will be parked

**To Reproduce**
It "feels" like it happens, if the current ekos run has to be interupted because of weather alert

**Expected behavior**
The mount should move always to the park position

**Screenshots**
See attachments

**Desktop (please complete the following information):**
 - OS: 22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Jan 12 18:54:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
 - Version 
Usage: indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 2.0.5
Code 2.0.5-tgz. Protocol 1.7.

**Log Files**
See Attachments

https://github.com/indilib/indi-3rdparty/assets/8696493/9631f786-f361-4d20-bfee-ac13c5429dc1

[parkerror.txt](https://github.com/indilib/indi-3rdparty/files/14152287/parkerror.txt)

pmneo, 2024-02-03:
Just to proof, that this works sometimes as expected:

Uploading noIssuesFast.mp4…

knro, 2024-02-04:
Did you set a custom parking position or it is the default home position ?

pmneo, 2024-02-04:
No, I'm using the default position and settings

pmneo, 2024-02-07:
So it happend again this night:

```
[2024-02-07T01:38:40.978 CET DEBG ][           org.kde.kstars.indi] - ISD:Telescope: Parking... 

[2024-02-07T01:38:40.978 CET INFO ][ org.kde.kstars.ekos.scheduler] - "Parking mount in progress..."
[2024-02-07T01:38:40.998 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] Compute local time: lst=10.52646354 (10:31:35.27) - julian date=2460347.52685271 "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j1\", 4 bytes written "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=F14073\", 8 bytes read "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] GetRAEncoder() = 7553265 "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j2\", 4 bytes written "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=BD7DC2\", 8 bytes read "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] Current encoders RA=7553265 DE=12746173 "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f1\", 4 bytes written "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=111\", 5 bytes read "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f2\", 4 bytes written "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=101\", 5 bytes read "
[2024-02-07T01:38:41.004 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] Scope RA ( 6:42:07) DE ( 9:46:57) Aligned RA ( 6:42:07) DE ( 9:46:57) AZ (251:12:50) ALT (28:45:42), PierSide (East) "
[2024-02-07T01:38:41.010 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] StopRA() : calling RA StopWaitMotor "
[2024-02-07T01:38:41.010 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f1\", 4 bytes written "
[2024-02-07T01:38:41.010 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=111\", 5 bytes read "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[MOUNT] StopWaitMotor() : Axis = 1 "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":K1\", 4 bytes written "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=\", 2 bytes read "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f1\", 4 bytes written "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=101\", 5 bytes read "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] StopDE() : calling DE StopWaitMotor "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f2\", 4 bytes written "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=101\", 5 bytes read "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[MOUNT] StopWaitMotor() : Axis = 2 "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":K2\", 4 bytes written "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=\", 2 bytes read "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f2\", 4 bytes written "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=101\", 5 bytes read "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j1\", 4 bytes written "
[2024-02-07T01:38:41.011 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=F14073\", 8 bytes read "
[2024-02-07T01:38:41.016 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j2\", 4 bytes written "
[2024-02-07T01:38:41.016 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=BD7DC2\", 8 bytes read "
[2024-02-07T01:38:41.016 CET INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Parking mount: RA increment = 835343, DE increment = -2053565 "
[2024-02-07T01:38:41.016 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] SlewTo() : deltaRA = 835343 deltaDE = -2053565 "
[2024-02-07T01:38:41.016 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[MOUNT] SetMotion() : Axis = 1 -- dir=forward mode=goto speedmode=highspeed "
[2024-02-07T01:38:41.016 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] CheckMotorStatus() : Axis = 1 "
[2024-02-07T01:38:41.016 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f1\", 4 bytes written "
[2024-02-07T01:38:41.017 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=101\", 5 bytes read "
[... cut ...]
[2024-02-07T01:38:44.023 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] IsRARunning() = false "
[2024-02-07T01:38:44.024 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] CheckMotorStatus() : Axis = 2 "
[2024-02-07T01:38:44.024 CET DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[DEBUG] IsDERunning() = false "
[2024-02-07T01:38:44.024 CET INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Mount is parked. "
``` 

Which seems to be declared as park finished. This may could be anhanced to check target encoder positions?

closed 2024-03-17 by knro