indilib/indi-3rdparty #910: Touptech 678 M

pdro871, 2024-03-29:
hi, I tried to upload the touptech 678 m camera to the ekos profile. But ekos doesn't see the room.

knro, 2024-03-30:
That makes no sense, please elaborate. You can't locate the profile? or the profile has the driver but it's not recognized?

pdro871, 2024-03-30:
The profile has the driver but the camera is not detected. I also tried to install the entire driver library, but I already have everything installed, I assume that these latest touptech cameras are not recognized. On windows the camera works. obviously there are other drives. Some advice?

Inviato dall'app Tiscali Mail.

knro, 2024-03-30:
@touptek Can you please check?

pdro871, 2024-03-30:
I didn't understand

blemasle, 2024-06-07:
Well, I also have issues with a brand new 678M that make the camera absolutely unusable with INDI.
For instance, building a dark library with PHD2 always lead to a failure at some point and the camera seems to be stuck in a loop.

I already tried to play to usb speed settings and set all the way down to 0 to no avail.
This is on INDI v2.0.8

MaierJoh, 2024-10-04:
I had the same problems, 
since the last update (INDI 3rd Party Drivers v2.1.0 Released) everything works again.