indilib/indi-3rdparty #912: indi_mi_ccd_usb for Moravian C3-26000C does not show cooler ON/OFF and camera temperature

noldi20, 2024-04-05:
**Describe the bug**
After updating my linux kstar version to 3.7.0 the cooler ON/OFF is disabled and the actual temperature for my Moravian C3-26000C CMOS Cam is not displayed, value is always zero. The previous driver was working. Indi driver is V1.9 indi_mi_ccd_usb. The cooling temperature can be set and also the cooling seems to be working. No error messages. 

**To Reproduce**
1. start kstar
2. Run Ekos with camera driver MI CCD (USB)
3. Go to CAM module and check temperature and cooler button

**Expected behavior**
1. Cooler buttons ON / OFF are enabled and can be selected
2. Actual Cam temperature is displayed e.g. 22.00 : field "Current CCD Temperature"

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu 
 - Version 22.04.4 LTS

**Log Files**
[log_08-36-11.txt](https://github.com/indilib/indi-3rdparty/files/14881067/log_08-36-11.txt)

knro, 2024-04-05:
@sWski Can you please check?

sWski, 2024-04-10:
Hi guys,

I looked at this problem yesterday and I'm pretty bummed about it. Summary of steps I've tried:

- installed KStars 3.7.0 + indi-full from Jasem's PPA - *buttons are disabled*
- downgrade indi-mi to previous version [1.8+t202306010454](https://launchpad.net/~mutlaqja/+archive/ubuntu/ppa/+sourcepub/14893226/+listing-archive-extra) - *buttons are disabled*
- downgrade libmicam to previous version [1.0+202305311922](https://launchpad.net/~mutlaqja/+archive/ubuntu/ppa/+sourcepub/14892740/+listing-archive-extra) - *buttons are disabled*
- downgrade KStars to version  [3.6.5](https://launchpad.net/~mutlaqja/+archive/ubuntu/ppa/+sourcepub/14999796/+listing-archive-extra) - *buttons are disabled*
- all the downgrades above together - *buttons are disabled*

Then I've reverted these packages back to the latest version and:

- compiled indi-mi from master - *buttons work*
- compiled indi-mi 3b62bcb (2 years ago) - *buttons work*
- downgraded libmicam to version mensioned above and compiled both master and 3b62bcb - *buttons work*

Looking at the differences in the logs, when the buttons do not work, this is not in the listing:  
```
Driver indi_mi_ccd: read <defSwitchVector device='MI C4-16000' name='CCD_COOLER'>
Client 9: queuing <defSwitchVector device='MI C4-16000' name='CCD_COOLER'>
```
This property is handled [here](https://github.com/indilib/indi-3rdparty/blob/master/indi-mi/mi_ccd.cpp#L223) and [here](https://github.com/indilib/indi-3rdparty/blob/master/indi-mi/mi_ccd.cpp#L257).
[HasCooler](https://github.com/indilib/indi/blob/master/libs/indibase/indiccd.h#L232) is quite straightforward. 
And CCD_HAS_COOLER is set by driver [here](https://github.com/indilib/indi-3rdparty/blob/master/indi-mi/mi_ccd.cpp#L369).

So it's not clear to me how when I compile the driver, it works. The library decides whether the camera has a cooler or not and it's version remained the same as the distributed version.

knro, 2024-04-10:
Yes I can confirm it's a driver compile issue, sorry to waste your time, I will rebuild all packages next.

closed 2024-04-10 by knro

peter-englmaier, 2024-04-27:
@knro Hi Jasem, I am facing the same issue now, with moravian camera C4: cooling buttons do not show up with current, latest release `indi-mi/jammy,now 1.8+t202402030654~ubuntu22.04.1 amd64`. When I compile the driver myself, it works again.

knro, 2024-04-28:
I initiated a [rebuild](https://code.launchpad.net/~mutlaqja/+recipe/indi-mi-stable) for this driver in the PPA

peter-englmaier, 2024-04-28:
> I initiated a [rebuild](https://code.launchpad.net/~mutlaqja/+recipe/indi-mi-stable) for this driver in the PPA
I just downloaded (using `apt update`) the new version and the cooling shows up again.
Thanks!