indilib/indi-3rdparty #917: can't connect to 2" esatto focuser

wbendick, 2024-04-27:
Hello I am trying to connect some astro equiptment to kstars through ekos and indi on a fresh install of astroberry on a rpi4. Most of it is connecting fine but I am not able to connect to my 2" Esatto focuser. lsusb sees the focuser is present.

[image]

It is the cygnal integrated products line. I have tried adding both Esatto Arco and just the Esatto in the device manager web page.

[image]

I see errors that esatto is offline in the log. 

[image]

Selecting the port doen't appear to do anything.

[image]

Does anyone have any suggestions on what I can try to get this connected? I know the hardware is functional, I can control it successfully from windows.

knro, 2024-04-28:
Firmware up to date?

wbendick, 2024-04-28:
yes, i updated it to the latest

knro, 2024-04-29:
Enable focuser logging and attach the log, apparently the handshake failed.

sonny486, 2024-04-29:
You do have the 12 Volt power supply hooked up?

wbendick, 2024-04-29:
@sonny486 Today I tried adding 12v power from a Celestron powertank to the esatto in addition to the usb cable. I didn't notice any difference. It still would not connect. Just powering esatto from the usb cable always worked fine from windows control.

wbendick, 2024-04-29:
@knro I don't see any options for enabling any additional logging in the INDI control panel for esatto. Is this hiding somewhere else?

[image]

wbendick, 2024-05-01:
I tried installing indi from source on a fresh ubuntu install and the connection problem went away. Seems like it has been fixed since the last release of astroberry.