indilib/indi-3rdparty #919: Uranus-c PRO PlayerOne cooling not working

kris969, 2024-05-07:
**Describe the bug**
From Ekos, buttons to turn on and of camera cooling are greyed, so not usable.
When trying to set temperature at indi driver level, the set temperature command is sent to the driver, but no information are given back. With the actual behavior, the camera temperature is unknown

**My setup**
==Raspberry==
- Raspberry pi 4 - 4Mb
- Fresh install of Ubuntu 23.10 headless, generate from rpi-imager + sudo apt update && sudo apt full-upgrade
- a SSD directly connected to an USB3 port
- a reduce set of packages & including Indi driver installed to fit my needs : gpsd, indi-bin, indi-gpsd, indi-playerone
  (version given at the end of this post)
- Raspi4 INDI Web Manager
- no other software installed on the Raspberry
- Raspberry is connected to my local network using a wired Ethernet connection

==Laptop==
- Ubuntu 23.10
- Kstars/Ekos
- I use a connection to the Raspberry pi server threw the network using ports 7624 & 8624

**To Reproduce**
From the laptop, connection to playerone is ok:
[image]

General informations are:
[image]

From Ekos : impossible to activate the temperature cooling process : button is greyed:
[image]

On the laptop, from PlayerOne indi driver, i did the following steps:
-1- when setting temperature to -5°C, the following text is present in the log file on the raspberry.
-2- when setting Max dT to: 1 °C/mn and Threshold to: 0.2°C, the following text is added in the log file on the raspberry.
-3- when waiting with no action, the following lines are added in the log file on the raspberry.
Log:
[image]

On the latptop the indi driver interface is still showing a temperature of 0.00
[image]

It seems nothing is working about cooling...   

**Expected behavior**
Just to have cooling working.
I haven't done yet all detailed tests for this indi driver, but it seems other functions are working.

**Indi version used**
indi-bin:
[image]

indi-playerone:
[image]

kris969, 2024-05-08:
I started trying to locate where this issue could be coming from...

To do this:
- Firstly, I remove the indi-playerone ppa package using: sudo apt remove indi-playerone.
- I build the indi-playerone driver based on this 3rd Party Drivers repo, strictly following the recommendations
- To be sure, I did a reboot

My idea was to add some debug log, if needed, to try to understand the origin of this issue.

From my laptop, first of all, I checked the driver information at the indi driver interface. It appears that the version number is exactly the same but:
- the running driver is: "indi-playerone-ccd". Before it was "indi-playerone-single-ccd", as shown on the previous post. 
- some extra information are present: SDK, Serial Number an Nickname.
All theses information are in the following screenshot:
[image]

But a major difference is that:  **IT WORKS !!!**
Maybe, the only remaining minor issues concern for "cooling power" and "refroidisseur". For me, if I am wrong let me know, these LED should be green. 
Here is the view from the indi driver interface:
[image]

At the Ekos level, everything is OK, as you can see in the following screenshot:
[image]

So to conclude:
- From sources from https://github.com/indilib/indi-3rdparty: indi-playerone works well relative to cooling system (tested with  Uranus-c PRO camera).
- There is something wrong for this package coming from the ppa.
- Maybe it  doesn't seem normal to have the same version number for these two package...

I hope my modest contribution will help?
Have fun, looking at the sky ;)

knro, 2024-05-08:
Looks like many issues with packages in last release. The compiled version works OK but there is an issue with the Debian packages which is so far unknown. I could rebuild all packages but this could be disruptive at this stage. In 3 weeks we will rebuild all package for next INDI release.

kris969, 2024-05-08:
Great decision!
I can continue so far with my workaround.
If you want I will be able to test the next release on my side to check if this issue will be effectively fixed.
Let me know ;)

kris969, 2024-05-09:
Another point that could be useful to know: it's mandatory to install the libplayerone library.

knro, 2024-05-20:
It's a packaging issue that affected many drivers. Hopefully resolved by 2.0.8 release due June 1st.

closed 2024-05-20 by knro