indilib/indi-3rdparty #928: libricohcamerasdk arm64 avalability

knro, 2024-05-20:
libricohcamerasdk is not available on arm64 but most modern operating systems migrated to arm64 now. Is there any hope of getting this running on arm64? Without it, can indi-pentax driver operation on arm64 architectures?

@karlrees Any ideas?

karlrees, 2024-05-21:
The indi-pentax should still be able to function on arm64 work _if your camera is in MSC mode_.  For that, it uses the pktriggermode-based driver, which should be able to compile on arm64.

The librricohcamerasdk driver was only needed for PTP mode.  Despite repeated requests, Ricoh never once responded to me about the source code.  I think they have even taken the library down now, as I can't find the site anymore.  Without some serious hacking, I don't see a path forward in PTP mode.

knro, 2024-05-22:
Thank you Karl for the clarification. I think we'll just add this tidbit to the driver documentation so that everybody is aware of this limitation.

closed 2024-05-22 by knro