indilib/indi-3rdparty #931: Using Indi_ddw_dome with DDW 5.2 and it crashes immediately with a sigsegv

jmocenigo, 2024-06-05:
I will gladly describe the bug in more detail (I know it is sigsegv from syslog), and I will even attempt to fix it (if need be), but before I get source, do a build, and use gdb to get more detailed info on where it is happening, I would like to ask: is there a link to a page that describes how to run just a module (e.g., indi_ddw_dome) from the command-line so I can simplify debugging it? 

As far as reproducing it for those that have DDW 5.2, which I should say is fully working from the hand controller. I should also add: this is my first time trying to use DDW with INDI (or at all) so I have no history of successful use to be able to say when things turned sour. My set up of DDW is: I switch from "OFF" to "REMOTE", connect RS-232 output cord that comes with DDW to a Prolific serial-to-usb (PL2303) converter, and have that connected to Ubuntu 22.04.4 LTS. I am using the latest release of INDI:
       indi-full is already the newest version (2.0.8+202406011048~ubuntu22.04.1)
When I tell the INDI DDW module to connect via KStars/EKOS, it connects and immediately crashes. This happens whether the DDW is the only non-simulated device or if all the devices are real and already running and connected.

**Describe the bug**
A clear and concise description of what the bug is.  Provide a general summary of the issue in the Title above as well as affected program name like `indiserver` or `indi_toupcam_ccd`

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform action
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2024-06-06:
Can you please use [Ekos Debugger](https://stellarmate.com/support/tutorials/ekos-debugger.html) to send us back the backtrace from the driver crash?

jmocenigo, 2024-06-06:
Wow! I had no idea ekosdebugger existed, but it clearly has been around for more than 2 years based on the forums. So easy to use, too! Thanks.

Log attached.

[indi_logs_24-06-06T11-38-05.zip](https://github.com/user-attachments/files/15649330/indi_logs_24-06-06T11-38-05.zip)

knro, 2024-06-06:
Can you try deleting ~/.indi/ParkData.xml and restarting the driver again?

jmocenigo, 2024-06-06:
Sorry for the delay in responding.

Yes, thank you so much, removing that file, which was for 2022, fixed the problem!

closed 2024-06-06 by knro