indilib/indi-3rdparty #949: Posible error Driver mi_ccd.cpp set temperature Moravian Gx-xxxx?

aeskartin, 2024-07-23:
HI

Could this be the issue that several Moravian camera users are experiencing when we turn off the camera's cooling from EKOS and it activates at 100% instead? At least 4 users have had this problem for a long time, and we cannot use the CCD warming function at the end of the session.

---

### Issue with INDIlib Driver Handling Moravian Camera Cooling

We are experiencing an issue with the INDIlib driver for Moravian cameras where the cooling does not turn off properly. Instead of shutting down, the cooling is activated at maximum power. To help diagnose and resolve this issue, we have reviewed and modified the driver files `mi_ccd.cpp`, `mi_ccd.h`, and the Moravian library `gxccd.h`.

### Analysis of the Code

#### Header File: `mi_ccd.h`

In the header file, we have the declarations related to the cooling system:

```cpp
ISwitch CoolerS[2];
ISwitchVectorProperty CoolerSP;
INumber CoolerN[1];
INumberVectorProperty CoolerNP;
```

#### Implementation File: `mi_ccd.cpp`

In the `initProperties` function, the cooler properties are defined:

```cpp
IUFillSwitch(&CoolerS[0], "COOLER_ON", "ON", ISS_ON);
IUFillSwitch(&CoolerS[1], "COOLER_OFF", "OFF", ISS_OFF);
IUFillSwitchVector(&CoolerSP, CoolerS, 2, getDeviceName(), "CCD_COOLER", "Cooler", MAIN_CONTROL_TAB, IP_WO, ISR_1OFMANY, 0, IPS_IDLE);

// CCD Regulation power
IUFillNumber(&CoolerN[0], "CCD_COOLER_VALUE", "Cooling Power (%)", "%+6.2f", 0.0, 1.0, 0.01, 0.0);
IUFillNumberVector(&CoolerNP, CoolerN, 1, getDeviceName(), "CCD_COOLER_POWER", "Cooling Power", MAIN_CONTROL_TAB, IP_RO, 60, IPS_IDLE);
```

The `SetTemperature` function seems to set the desired temperature correctly:

```cpp
int MICCD::SetTemperature(double temperature)
{
    TemperatureRequest = temperature;
    if (gxccd_set_temperature(cameraHandle, temperature) != 0)
    {
        LOGF_ERROR("Error setting temperature to %.2f", temperature);
        return -1;
    }
    LOGF_INFO("Temperature set to %.2f", temperature);
    return 0;
}
```

#### Handling Cooler Switch

The `ISNewSwitch` function handles the switch states, including those for cooling:

```cpp
bool MICCD::ISNewSwitch(const char *dev, const char *name, ISState *states, char *names[], int n)
{
    if (!strcmp(name, "CCD_COOLER"))
    {
        if (states[0] == ISS_ON)
        {
            if (gxccd_set_temperature(cameraHandle, TemperatureRequest) != 0)
            {
                LOGF_ERROR("Error turning on cooler.");
                return false;
            }
        }
        else
        {
            if (gxccd_set_temperature(cameraHandle, TEMP_COOLER_OFF) != 0)
            {
                LOGF_ERROR("Error turning off cooler.");
                return false;
            }
        }
        LOGF_INFO("Cooler %s", states[0] == ISS_ON ? "ON" : "OFF");
        return true;
    }
    // Handle other switches...
    return INDI::CCD::ISNewSwitch(dev, name, states, names, n);
}
```

### Possible Problem

The issue could be related to the `TEMP_COOLER_OFF` value set to 100 degrees Celsius. This high value might be causing the cooler to activate at maximum power instead of turning off.?

### Solution

We propose adjusting the `TEMP_COOLER_OFF` value to a more appropriate setting, such as `-1` degrees Celsius, to ensure the cooler turns off correctly:

```cpp
#define TEMP_THRESHOLD  0.2  /* Differential temperature threshold (°C) */
#define TEMP_COOLER_OFF -1   /* Temperature to turn off the cooler (°C) */
```

This change should help in correctly turning off the cooler when needed.

Alex
Josep Lluis
Eduardo
Ramon

knro, 2024-07-23:
@sWski Can you please take a look?

sWski, 2024-07-26:
Hello guys,

I have investigated this problem again, because it seemed to me that we have already fixed this bug...
I found that this was the case in the firmware of the new Cx series cameras, but the bug still exists in the old Gx series cameras. 

Basically the camera has some temp. offset in EEPROM, which is set by calibration, but if this offset is negative, then when calculating the value in FW it can overflow the maximum value and the camera starts cooling.

Solutions are two:
1. Set `TEMP_COOLER_OFF` to some reasonable temperature, like 50, but we don't know what kind of offsets the cameras have in them, so some of them may still overflow
2. gxccd library will read the offset from EEPROM and compensate this bug in FW, so it will not overflow

Option 2. seems more reasonable to me, I already have this compensation programmed and I think it works as it should. Next week I would just test it more and release a new version of the library.

aeskartin, 2024-07-26:
Sure, here is the translation with the added names:

---

Hi Jakub,

Thank you for your message and for investigating this problem again. If you consider it necessary and don't have the means to test it yourself, I would like to offer to perform a test on one of our Gx series cameras using a remote setup. For this, I would need detailed instructions to compile the code with the solution and a straightforward way to revert the changes if necessary.

Here’s what I need:

1. **Instructions to compile the solution code**: Please provide the necessary steps to download, compile, and implement the updated version of the gxccd library with the programmed compensation. Include any specific dependencies or configurations needed.

2. **Procedure to revert to the previous version**: In case the solution does not work as expected, it would be helpful to have clear instructions on how to revert the changes and restore the previous version of the firmware and library. This could include specific commands or detailed steps to undo the update.

Once I have these instructions, I can perform the test and provide you with detailed feedback on the camera's behavior with the new compensation.

Thank you for your help, and I look forward to your instructions.

Best regards,

Alex

aeskartin, 2024-09-07:
Here is the translation of the text you provided:

---

Hello everyone,  
I assume that this correction is implemented in the latest version of KStars 3.7.2. 

I want to share the tests I have conducted with my Moravian G2-8300, using KStars version 3.7.2 and INDI 2.0.9. Some other colleagues are also experiencing similar issues where it doesn't work correctly.

I am using KStars 3.7.2 and INDI/EKOS 2.0.9 on a Raspberry Pi 4 with Ubuntu 22.04.4 LTS aarch64, with MATE 1.26.

Here are the tests I've done with the Moravian G2-8300:
By default, when everything is connected and the cooling is on in EKOS, the CCD remains at ambient temperature and the cooling power is at 0%.
[image]

If I turn off the cooling in EKOS, the CCD temperature drops to the minimum, the cooling power remains at 100%, and it goes down to -30ºC.
[image]

If I turn the cooling back on in EKOS, the temperature now stays at 0ºC, and the cooling power at 20%. To bring it back to ambient temperature and the cooling power to 0%, I have to set the target temperature to +30ºC.
[image]

---

Let me know if you need any further adjustments or assistance!

Alex

knro, 2024-12-02:
@aeskartin Checking the code, it appears that when toggling the cooler off, it sets temperature to +100 which should WARM up the camera. Are you testing against latest drivers?

closed 2024-12-02 by knro

aeskartin, 2024-12-06:
Hice pruebas con la versión 3.7.2 de Kstars e INDI 2.0.9 y sigue haciendo
lo mismo.
[image: imagen.png]
Iniciando la prueba, posición apagando la refrigeración

[image: imagen.png]
despues de 10 min llega a la temperatura minima y el cooling power se
mantiene a 100%

[image: imagen.png]
Posición encendido la refrigeración el cooling a 0% la temperatura sube.

[image: imagen.png]
Al llegar la temperatura de refrigeración a 0º, el cooling la mantiene

Con la version 3.7.3 y

aeskartin, 2024-12-06:
> I tested with Kstars version 3.7.2 and INDI 2.0.9 and it still does the
> same thing.
> [image: imagen.png]
> Starting the test, position turning off cooling
>
> [image: imagen.png]
> after 10 min it reaches the minimum temperature and the cooling power
> stays at 100%.
>
> [image: imagen.png]
> Position the cooling power to 0% and the temperature rises.
>
> [image: imagen.png]
> When the cooling temperature reaches 0°C, the cooling power maintains it.
>
> [image: imagen.png]
> If we want the cooling power to be deactivated we have to confirm a
> temperature higher than room temperature, 20º.
>
> With version 3.7.3 and INDI 2.1.0 I have not been able to test because
> this version does not recognise the camera.

Is it possible to update the drivers in version 2.0.9? My knowledge of
compiling code is limited, but with a few hints I could test it.

Thank you

ALex
AAVSO: EALA
aavbae.net

aeskartin, 2024-12-06:
I have tested with Kstars version 3.7.2 and INDI 2.0.9 and the problem remains

[image] 
Start of the test, cooling off position 

[image] 
After 10 min it reaches the minimum possible temperature and the cooling power is kept at 100% 

[image] 
We move to ON position, the cooling power is at 0%, the temperature rises. 

[image]
In cooling ON position it reaches 0º, the cooling power rises to 19% and keeps it that way. 

[image] 
If we want to deactivate the cooling power we have to confirm a temperature higher than the ambient temperature, for example 20º. 

With version 3.7.3 and INDI 2.1.0 I have not been able to test because this version does not recognise the Moravian camera. 

Is it possible to update to the new drivers in version 2.0.9? I guess by recompiling the local code. 
My knowledge of compiling code is limited, but with some hints I could try it. 

Thanks 
Alex 
AAVSO: EALA 
[aavbae.net](http://aavbae.net/)

knro, 2024-12-06:
INDI v2.1.0 and v2.1.1 (release last week) should both support MI cameras. What do you mean it does not recognize MI?

aeskartin, 2024-12-06:
Hello Jasem

 I am referring to this post.

https://github.com/indilib/indi-3rdparty/issues/995#issuecomment-2408818651

When updating my Raspberry Pi 4 with Ubuntu to Indi version 2.1.0, it did
not detect the Moravian G2-8300 CCD camera.

Downgrade to version 2.0.9 to make it work.

Alex

aeskartin, 2024-12-24:
Hi Jasem, I have some updates.

I finally managed to set up an SD card and send it to the observatory. It has the following configuration: INDI 2.1.1 with KStars 3.7.4 stable, running on Ubuntu 22.04.5 LTS Jammy Jellyfish 64-bit, Linux 5.15.0-1029 Raspi aarch64.

This is a somewhat uncommon configuration. After updating, I performed some tests, and now both issues #995 and #949 have been resolved—the detection issue with the Moravian G2-8300 camera and the temperature control issue, respectively.

Today, I was able to complete a scheduled observation without having to operate manually. I will conduct more tests if the weather permits, and if I encounter any problems, I will open a new thread.

Thank you, Jasem, for your hard work and dedication to the project!

Alex