indilib/indi-3rdparty #951: Does not compile

HaraldStauss, 2024-07-29:
Since today, indi-3rdparty does not compile any more.
There seems to be a problem in the indi-eqmod module. 
Perhaps the recent updates in the CMakeLists.txt file is the problem.
Compiler stops with an error when working on the StarAdventurer driver.

Thanks for looking into this.

knro, 2024-07-29:
You need to recompile indi library first.

closed 2024-07-29 by knro

HaraldStauss, 2024-07-30:
I did recompile the indi libriary first. It is still not compiling. Here is what I got:

[  2%] Building CXX object indi-eqmod/CMakeFiles/indi_staradventurergti_telescope.dir/staradventurergti.cpp.o
In file included from /home/pi/Projects/indi-3rdparty/indi-eqmod/azgtibase.h:21,
                 from /home/pi/Projects/indi-3rdparty/indi-eqmod/azgti.cpp:21:
/home/pi/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:245:35: error: 'TelescopeHomeAction' has not been declared
  245 |         IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                                   ^~~~~~~~~~~~~~~~~~~
In file included from /home/pi/Projects/indi-3rdparty/indi-eqmod/staradventurergtibase.h:21,
                 from /home/pi/Projects/indi-3rdparty/indi-eqmod/staradventurergti.cpp:21:
/home/pi/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:245:35: error: 'TelescopeHomeAction' has not been declared
  245 |         IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                                   ^~~~~~~~~~~~~~~~~~~
/home/pi/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:245:17: error: 'IPState EQMod::ExecuteHomeAction(int)' marked 'override', but does not override
  245 |         IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                 ^~~~~~~~~~~~~~~~~
/home/pi/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:245:17: error: 'IPState EQMod::ExecuteHomeAction(int)' marked 'override', but does not override
  245 |         IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                 ^~~~~~~~~~~~~~~~~
make[2]: *** [indi-eqmod/CMakeFiles/indi_staradventurergti_telescope.dir/build.make:108: indi-eqmod/CMakeFiles/indi_staradventurergti_telescope.dir/staradventurergti.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1305: indi-eqmod/CMakeFiles/indi_staradventurergti_telescope.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/build.make:108: indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/azgti.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  2%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/azgtibase.cpp.o
In file included from /home/pi/Projects/indi-3rdparty/indi-eqmod/azgtibase.h:21,
                 from /home/pi/Projects/indi-3rdparty/indi-eqmod/azgtibase.cpp:19:
/home/pi/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:245:35: error: 'TelescopeHomeAction' has not been declared
  245 |         IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                                   ^~~~~~~~~~~~~~~~~~~
/home/pi/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:245:17: error: 'IPState EQMod::ExecuteHomeAction(int)' marked 'override', but does not override
  245 |         IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                 ^~~~~~~~~~~~~~~~~
make[2]: *** [indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/build.make:121: indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/azgtibase.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1332: indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/all] Error 2
make: *** [Makefile:171: all] Error 2

knro, 2024-07-30:
Did you install the compiled libindi to the system?