indilib/indi-3rdparty #956: indi-gphoto: Unreliable long exposures on Canon DSLR (EOS6D in particular, but also others)

jochym, 2024-08-10:
**Describe the bug**
Long exposures (above 10s) on Canon DSLR are very unreliable. It is very obvious on my 6D but also 40D suffers similar problems. The capture hangs, repeats multiple times etc.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Just set exposure to 30-40s
3. The problem will show up very soon

**Expected behavior**
Well, reliable captures ;)

**Desktop (please complete the following information):**
 - OS: Linux debian/raspbian 
 - Version bookworm, but it was like this for a long time

**Analysis**
It seems with long exposures the data may be available after some, sometimes long delay. In case of 6D the delay may be as long as the exposure (I suspect the camera is doing dark framing internally). I have figured out that simple constant wait suggested in gphoto docs is not correct. After long search and trial-and-error, I have figured out the correct sequence for EOS 6D in gphoto CLI (example script - captures a frame with given exposition):
```
#!/bin/bash

LANG=C 

gphoto2 --filename "capt_exp=${1}s_%Y-%m-%d_%H:%M:%S.%C" \
        --set-config eosremoterelease=Immediate \
        --wait-event=${1}s \
        --set-config eosremoterelease="Release Full" \
        --wait-event-and-download=CAPTURECOMPLETE
```
In case of gphoto2 tool, the LANG=C is crucial, but the real solution is `--wait-event-and-download=CAPTURECOMPLETE` which waits for the actual end of capture. This script essentially "works". Unfortunately, I do not know how to port this solution to indi-gphoto. I guess it may be easy for the person familiar with the code (just wait for this event instead of fixed timeout, although some configurable timeout on top of this may be useful).

jochym, 2024-08-10:
Just to add some more data. The problem may be worked around for 6D with setting "Long time exposure noise reduction" to off in camera. However, I think this is still a bug. We should wait for a complete exposure, not a particular timeout. Actually, I think we should wait on complete exposure *and* complete file showing up in the camera. The automatic dark frame in 6D is actually a very neat feature. It makes the scope shutter unnecessary. It will be a shame not to use it.

jochym, 2024-08-10:
And it fixed the problem. This problem. Unfortunately, there is also a problem with focusing - but that is for another day. That is - it is worked-around. It still needs proper fixing...

knro, 2024-08-11:
Thank you for your investigation. I suppose we could wait full-exposure but perhaps not by default. Maybe can add a toggle for timeout: Default vs Max

jochym, 2024-08-11:
Great idea! My tests last night showed that it really works over a wide range of times - so it is definitely worth it. Maybe just a switch: wait for exposure+timeout/wait for complete in-camera processing + timeout with adjustable timeout. It seems this was a problem before - the timeout used to be 5s (from the comment) now it is 10s...

Additionally, I spotted a problem with focusing in EKOS. Autofocusing leaves the camera in hang state, while executing the same steps manually with buttons in EKOS is fine - I guess something is not closed properly at the end. The focusing is problematic with canon, but *eventually* it works (see the attached frame). But it is a problem with DSLRs and I do not think much can be done about that. 
One thing though - the canon focussing works only in viefinder mode. I had to switch it on manually in the INDI control dialog - it will be easier on the user if this was integrated into EKOS focusing page. Shall I open a separate issue somewhere for this?
  
[image]