indilib/indi-3rdparty #976: Indi-gphoto: can't capture images:  Failed to open serial port: usb:001,020

apalazzi, 2024-09-18:
**Describe the bug**
I cannot acquire images from my Canon EOS 5d MKII: the camera is detected and I can get videos, but when I try to capture images I get an "failed to open serial port" error.
The permissions on the port seem correct:
crw-rw-rw-+ 1 root plugdev 189, 129 18 set 21.16 /dev/bus/usb/002/020

**To Reproduce**
1. Connect and turn on the camera
2. start the indi server
3. try to take a picture with kstars/ekos or CCDCiel

**Expected behavior**
The camera should take a picture.

**Desktop (please complete the following information):**
 - OS: Debian/Linux  Trixie/sid
 - Version latest git version

**Log Files**
2024-09-18T19:00:34: [ERROR] Error starting exposure 
2024-09-18T19:00:34: [ERROR] Failed to open serial port: usb:001,020 
2024-09-18T19:00:34: [INFO] Starting 1 seconds exposure. 
2024-09-18T19:00:34: [ERROR] Error starting exposure 
2024-09-18T19:00:34: [ERROR] Failed to open serial port: usb:001,020 
2024-09-18T19:00:34: [INFO] Starting 1 seconds exposure. 
2024-09-18T19:00:34: [ERROR] Error starting exposure 
2024-09-18T19:00:34: [ERROR] Failed to open serial port: usb:001,020 
2024-09-18T19:00:34: [INFO] Starting 1 seconds exposure. 
2024-09-18T18:56:02: [INFO] World Coordinate System is enabled. 
2024-09-18T18:56:02: [INFO] Device configuration applied. 
2024-09-18T18:56:02: [INFO] Force BULB is enabled. All expsures shall be captured in BULB mode except for subsecond captures. 
2024-09-18T18:56:02: [INFO] Images downloaded from camera will saved in the camera internal storage. 
2024-09-18T18:56:02: [INFO] Capture target set to Internal RAM 
2024-09-18T18:56:02: [INFO] Upload settings set to client only. 
2024-09-18T18:56:02: [INFO] Loading device configuration... 
2024-09-18T18:56:02: [INFO] Detected Canon Inc. Model Canon EOS 5D Mark II. 
2024-09-18T18:56:02: [INFO] Canon DSLR EOS 5D Mark II is online.

hitoshi869, 2024-09-19:
Hello
I have the same problem with the latest indilib.
On my canon 600D, the save of the image setting on indilib fails and the red mark stays on.
This happens on both ghoto-ccd and canon DSLR.
CCDCiel shows 600D error
I hope it will be improved. Any advice would be appreciated.

knro, 2024-09-19:
It's already fixed in GIT. Should be released with INDI 2.1.0 on Oct 1st.

closed 2024-09-19 by knro

hitoshi869, 2024-09-19:
HelloUnfortunately I used the latest code from the Git clone yesterday. Once I got to the latest version, the problem occurred.

 
 

Closed #976 as completed.

hitoshi869, 2024-09-23:
I rebuilt both indilib and 3rdparty this morning with the latest code and gphoto_ccd works in my environment.
I was able to control it from CCDCiel.
Thank you.