indilib/indi-3rdparty #981: Indilib indi-3rd party making eqmod fails #980

zac-old, 2024-09-22:
Continuing my previous bug report which has been closed with the advice that  I must install the latest indi first.
I had installed the latest indi - that is how I discovered I needed the third party drivers.
Please see my original bug report.

zac-old, 2024-09-24:
As I indicated in a private communication - since you said that I must install indi-core before the 3rd party elements -  I thought the fact that I had deleted the indi-core installation files before I installed eqmod might have been why my install failed. This has proved not to be the case. I have reinstalled again retaining all the installation files from indi-core and the make failed exactly as before. Error messages as follows:

[  3%] Building C object CMakeFiles/indi_eqmod_telescope.dir/align/htm.c.o
[  3%] Building C object CMakeFiles/indi_staradventurergti_telescope.dir/align/htm.c.o
[  5%] Building C object CMakeFiles/indi_azgti_telescope.dir/align/htm.c.o
[  6%] Building C object CMakeFiles/indi_staradventurer2i_telescope.dir/align/htm.c.o
[ 10%] Building C object CMakeFiles/indi_staradventurergti_telescope.dir/align/chull/chull.c.o
[ 10%] Building C object CMakeFiles/indi_eqmod_telescope.dir/align/chull/chull.c.o
[ 11%] Building C object CMakeFiles/indi_staradventurer2i_telescope.dir/align/chull/chull.c.o
[ 13%] Building C object CMakeFiles/indi_azgti_telescope.dir/align/chull/chull.c.o
[ 15%] Building CXX object CMakeFiles/indi_eqmod_telescope.dir/eqmod.cpp.o
[ 16%] Building CXX object CMakeFiles/indi_staradventurer2i_telescope.dir/staradventurer2i.cpp.o
[ 18%] Building CXX object CMakeFiles/indi_staradventurergti_telescope.dir/staradventurergti.cpp.o
[ 20%] Building CXX object CMakeFiles/indi_azgti_telescope.dir/azgti.cpp.o
In file included from /home/zac/Projects/indi-3rdparty/indi-eqmod/eqmod.cpp:20:
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                               ^~~~~~~~~~~~~~~~~~~
In file included from /home/zac/Projects/indi-3rdparty/indi-eqmod/staradventurer2i.h:21,
                 from /home/zac/Projects/indi-3rdparty/indi-eqmod/staradventurer2i.cpp:21:
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                               ^~~~~~~~~~~~~~~~~~~
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:13: error: ‘IPState EQMod::ExecuteHomeAction(int)’ marked ‘override’, but does not override
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |             ^~~~~~~~~~~~~~~~~
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:13: error: ‘IPState EQMod::ExecuteHomeAction(int)’ marked ‘override’, but does not override
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |             ^~~~~~~~~~~~~~~~~
In file included from /home/zac/Projects/indi-3rdparty/indi-eqmod/staradventurergtibase.h:21,
                 from /home/zac/Projects/indi-3rdparty/indi-eqmod/staradventurergti.cpp:21:
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                               ^~~~~~~~~~~~~~~~~~~
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:13: error: ‘IPState EQMod::ExecuteHomeAction(int)’ marked ‘override’, but does not override
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |             ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/indi_staradventurer2i_telescope.dir/build.make:104: CMakeFiles/indi_staradventurer2i_telescope.dir/staradventurer2i.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:167: CMakeFiles/indi_staradventurer2i_telescope.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/indi_eqmod_telescope.dir/build.make:104: CMakeFiles/indi_eqmod_telescope.dir/eqmod.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 22%] Building CXX object CMakeFiles/indi_eqmod_telescope.dir/eqmodbase.cpp.o
In file included from /home/zac/Projects/indi-3rdparty/indi-eqmod/azgtibase.h:21,
                 from /home/zac/Projects/indi-3rdparty/indi-eqmod/azgti.cpp:21:
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                               ^~~~~~~~~~~~~~~~~~~
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:13: error: ‘IPState EQMod::ExecuteHomeAction(int)’ marked ‘override’, but does not override
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |             ^~~~~~~~~~~~~~~~~
[ 23%] Building CXX object CMakeFiles/indi_staradventurergti_telescope.dir/staradventurergtibase.cpp.o
make[2]: *** [CMakeFiles/indi_staradventurergti_telescope.dir/build.make:104: CMakeFiles/indi_staradventurergti_telescope.dir/staradventurergti.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 25%] Building CXX object CMakeFiles/indi_azgti_telescope.dir/azgtibase.cpp.o
make[2]: *** [CMakeFiles/indi_azgti_telescope.dir/build.make:104: CMakeFiles/indi_azgti_telescope.dir/azgti.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.cpp:33:
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
[... cut ...]
/home/zac/Projects/indi-3rdparty/indi-eqmod/eqmodbase.cpp:3622:34: error: ‘TelescopeHomeAction’ was not declared in this scope; did you mean ‘TelescopeConnection’?
 3622 | IPState EQMod::ExecuteHomeAction(TelescopeHomeAction action)
      |                                  ^~~~~~~~~~~~~~~~~~~
      |                                  TelescopeConnection
make[2]: *** [CMakeFiles/indi_eqmod_telescope.dir/build.make:118: CMakeFiles/indi_eqmod_telescope.dir/eqmodbase.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:89: CMakeFiles/indi_eqmod_telescope.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

zac-old, 2024-10-03:
Using the new versions 2.1.0 of indi-core and indi-3rdparty it still doesn't work.

indi-eqmod still fails to make for the same reason Error report as follows.

zac@rpj2:~/Projects/build/indi-eqmod $ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty-2.1.0/indi-eqmod
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found INDI: /usr/lib/aarch64-linux-gnu/libindidriver.so;/usr/lib/aarch64-linux-gnu/libindiAlignmentDriver.so (found version "1.9.9") 
-- Found NOVA: /usr/lib/aarch64-linux-gnu/libnova.so
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found version "1.2.13") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Found GSL: /usr/include (found version "2.7.1") 
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- Could NOT find GMock (missing: GMOCK_LIBRARY GMOCK_INCLUDE_DIR) 
-- GTEST not found, not building unit tests
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zac/Projects/build/indi-eqmod
zac@rpj2:~/Projects/build/indi-eqmod $ make -j4
[  6%] Building C object CMakeFiles/indi_staradventurer2i_telescope.dir/align/htm.c.o
[  6%] Building C object CMakeFiles/indi_staradventurergti_telescope.dir/align/htm.c.o
[  6%] Building C object CMakeFiles/indi_eqmod_telescope.dir/align/htm.c.o
[  6%] Building C object CMakeFiles/indi_azgti_telescope.dir/align/htm.c.o
[ 13%] Building C object CMakeFiles/indi_azgti_telescope.dir/align/chull/chull.c.o
[ 13%] Building C object CMakeFiles/indi_staradventurer2i_telescope.dir/align/chull/chull.c.o
[ 13%] Building C object CMakeFiles/indi_staradventurergti_telescope.dir/align/chull/chull.c.o
[ 13%] Building C object CMakeFiles/indi_eqmod_telescope.dir/align/chull/chull.c.o
[ 18%] Building CXX object CMakeFiles/indi_staradventurergti_telescope.dir/staradventurergti.cpp.o
[ 18%] Building CXX object CMakeFiles/indi_azgti_telescope.dir/azgti.cpp.o
[ 18%] Building CXX object CMakeFiles/indi_eqmod_telescope.dir/eqmod.cpp.o
[ 20%] Building CXX object CMakeFiles/indi_staradventurer2i_telescope.dir/staradventurer2i.cpp.o
In file included from /home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/staradventurergtibase.h:21,
                 from /home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/staradventurergti.cpp:21:
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                               ^~~~~~~~~~~~~~~~~~~
In file included from /home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/azgtibase.h:21,
                 from /home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/azgti.cpp:21:
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                               ^~~~~~~~~~~~~~~~~~~
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.h:247:13: error: ‘IPState EQMod::ExecuteHomeAction(int)’ marked ‘override’, but does not override
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |             ^~~~~~~~~~~~~~~~~
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.h:247:13: error: ‘IPState EQMod::ExecuteHomeAction(int)’ marked ‘override’, but does not override
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |             ^~~~~~~~~~~~~~~~~
In file included from /home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmod.cpp:20:
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |                               ^~~~~~~~~~~~~~~~~~~
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.h:247:13: error: ‘IPState EQMod::ExecuteHomeAction(int)’ marked ‘override’, but does not override
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
      |             ^~~~~~~~~~~~~~~~~
In file included from /home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/staradventurer2i.h:21,
                 from /home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/staradventurer2i.cpp:21:
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.h:247:31: error: ‘TelescopeHomeAction’ has not been declared
  247 |     IPState ExecuteHomeAction(TelescopeHomeAction action) override;
[... 211 lines cut ...]
 3599 |             LOGF_INFO("Start Tracking (%s).", TrackModeSP.findOnSwitch()->getLabel());
      |             ^~~~~~~~~
/home/zac/Projects/indi-3rdparty-2.1.0/indi-eqmod/eqmodbase.cpp:3607:62: error: ‘ISwitchVectorProperty’ {aka ‘struct _ISwitchVectorProperty’} has no member named ‘findOnSwitch’
[... cut ...]
 3622 | IPState EQMod::ExecuteHomeAction(TelescopeHomeAction action)
      |                                  ^~~~~~~~~~~~~~~~~~~
      |                                  TelescopeConnection
make[2]: *** [CMakeFiles/indi_eqmod_telescope.dir/build.make:118: CMakeFiles/indi_eqmod_telescope.dir/eqmodbase.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:89: CMakeFiles/indi_eqmod_telescope.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
zac@rpj2:~/Projects/build/indi-eqmod $ 

Please don't tell me I need to install indi-core.It is already installed.

knro, 2024-10-04:
Where is indi-core installed to? /usr or /usr/local? It must be the former.

Astro-Otter-Space, 2024-11-14:
Got error too after trying to rebuild indi-eqmod. 
For indi-core, i try to find it : 
```bash
sudo find / -type d -name 'indi-core'
/home/astroloutre/Projects/build/indi-core
```
But got only build directory. Should i rebuild it too ?

closed 2024-12-02 by knro