indilib/indi-3rdparty #994: QHY174GPS camera cooling doesn't work

MatiasLM, 2024-10-11:
Im using QHY174GPS camera + HYCFW3 filter wheel connected to a Raspberry Pi 4B 8GB with AstroArch (with INDI and Kstars). Camera and filter wheel controla and adquisición seems to be working well however i'snt possible to control camera temperature.

Even though the set point is set to 0° the camera temperature remains at ambient temperature (no fixed). I tried to do the same from the configuration menu but I get the same results. Even if I set it to automatic mode with Collong Power at 100% the camera remains at room temperature. See attachments.
[image]
[image]

I enabled logging, start INDI, power on and configure the camera cooler both from Ekos capture panel and INDI control panel. I attach the log file
[image]
[log_15-09-51.txt](https://github.com/user-attachments/files/17345931/log_15-09-51.txt)

It is worth noting that the camera is powered externally and the cooling system works and the issue occurs with and without filter wheel connected. Whitout using INDI, If I use the Raspberryas a USB server hub (virtualhere) and start other software such as Shapcap on the client PC, it is possible to control the temperature correctly.

Full hardware configuration

QHY174GPS camera + HYCFW3 filter wheel connected to camera
ZWO EAF
Sky Whatchet HEQ5(EQMOD)
QHY5III290M (guiding)
Raspberry Pi 4B 8GB
Full software configuration

AstroArch SO (https://github.com/devDucks/astroarch)
Kstars 3.7.2
indi 2.0.9

Best Regards,

PD: I published this issuen on [AstroArch Gitlab](https://github.com/devDucks/astroarch/issues/109) and [sourceforge](https://sourceforge.net/p/indi/bugs/99/) but although I have gotten some answers, the solution has not yet been found.

MatiasLM, 2024-10-13:
Some updates,

I tried in same hardware setup on other SD Card with Raspberry Pi OS Bookworm (64 bit) and I still having the same problem.  Software deployed with [nou astro-soft-build script](https://gitea.nouspiro.space/nou/astro-soft-build) :
- INDI Library: 2.1.0 Code v2.1.0. Protocol 1.7.
- kstars 3.7.3 Stable

Here you can find the log [log_12-34-32.txt](https://github.com/user-attachments/files/17357001/log_12-34-32.txt)

knro, 2024-10-14:
I believe QHY is already following up with this? We can only fix issues in the INDI driver itself, not the SDK.

MatiasLM, 2024-10-14:
I think a QHY developer is following the issue here https://sourceforge.net/p/indi/bugs/99/. However, I think he is saying that indi-3rdparty is using an unofficial version of the SDK.