indilib/indi-3rdparty #995: Help Please, Moravian G2-8300 controler Failure with INDI 2.1.0

aeskartin, 2024-10-12:
Hi

A week ago, my Raspberry Pi 4 with Ubuntu updated to INDI 2.1.0, and the Moravian G2-8300 camera stopped working. When starting INDI, it loads all the equipment in the remote observatory setup except the main camera, and it gives a driver failure error. 

I reported this issue with the logs from KStars and Ekos Debugger in the INDILIB forum; here is the link: 
https://www.indilib.org/forum/ccds-dslrs/15003-help-after-the-update-the-g2-8300-camera-is-not-working-driver-fail.html#102488, but Jasem advised me to report it here to see if a solution can be found. 

If more information is needed or if I should include anything else, please let me know. 

Detail driver INDI:
2024-10-05T00:07:13: startup: indiserver Usage: indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 2.1.0
Code 2.1.0-tgz. Protocol 1.7.

Version OS:

Code 2.1.0-tgz. Protocol 1.7.
         `:+shmNNMMNNmhs+:`              AAVBAE@trev
        .odMMMMMMMMMMMMMMMMMMdo.                 /dMMMMMMMMMMMMMMMmMMMMMMMMd/         OS: Ubuntu MATE 22.04.5 LTS aarch64
    :mMMMMMMMMMMMMNNNNM/`/yNMMMMMMm:       Host: Raspberry Pi 4 Model B Rev 1.4
  `yMMMMMMMMMms:..-::oM:    -omMMMMMy`     Kernel: 5.15.0-1029-raspi
 `dMMMMMMMMy-.odNMMMMMM:    -odMMMMMMd`    Uptime: 1 hour, 34 mins
 hMMMMMMMm-.hMMy/....+M:`/yNm+mMMMMMMMh    Packages: 3091 (dpkg), 18 (snap)
/MMMMNmMN-:NMy`-yNMMMMMmNyyMN:`dMMMMMMM/   Shell: bash 5.1.16
hMMMMm -odMMh`sMMMMMMMMMMs sMN..MMMMMMMh   Resolution: 1920x1080
NMMMMm    `/yNMMMMMMMMMMMM: MM+ mMMMMMMN   DE: MATE 1.26.0
NMMMMm    `/yNMMMMMMMMMMMM: MM+ mMMMMMMN   WM: Metacity (Marco)
hMMMMm -odMMh sMMMMMMMMMMs oMN..MMMMMMMh   Theme: Yaru-bark-dark [GTK2/3]
/MMMMNNMN-:NMy`-yNMMMMMNNsyMN:`dMMMMMMM/   Icons: Yaru-bark-dark [GTK2/3]
 hMMMMMMMm-.hMMy/....+M:.+hNd+mMMMMMMMh    Terminal: mate-terminal
 `dMMMMMMMMy-.odNMMMMMM:    :smMMMMMMd`    Terminal Font: Monospace 14
   yMMMMMMMMMms/..-::oM:    .+dMMMMMy      CPU: BCM2835 (4) @ 1.800GHz
    :mMMMMMMMMMMMMNNNNM: :smMMMMMMm:       Memory: 1450MiB / 7632MiB
      /dMMMMMMMMMMMMMMMdNMMMMMMMd/
        .odMMMMMMMMMMMMMMMMMMdo.                                   
           `:+shmNNMMNNmhs+:`                         

Thank you.

knro, 2024-10-13:
@sWski Can you please take a look?

aeskartin, 2024-10-18:
Hi, good morning. 

@sWski ,  @knro 

I’d like to know if the issue with the Moravian driver for the G2 cameras will be fixed soon. If you think the solution will be ready shortly, I can wait. But if it’s going to take more than another week, I can’t have the remote setup down for that long and will need to consider an alternative. Could you give me an idea of when the fix might be ready? 

On my end, if you need more information, need me to run tests, or pull more logs, I’m available to help with whatever you need. 

Thanks for everything!