indilib/indi-3rdparty #1001: Erratic errors in MaxDome driver.

fcasarramona, 2024-11-22:
**Describe the bug**
During the dome operation, numerous "Dome errors" appears from time to time, with no apparent reason.
This didn't happen in previous versions. See my source code review at the end

**To Reproduce**
Not a fixed procedure o know cause. 

**Expected behavior**
No errors happen in previous versions.

**Desktop (please complete the following information):**
 - OS: Stellarmate OS
 - Version 1.8.5

**Log Files**
Attached Log file. Se at [2024-11-21T22:16:59.133 CET INFO ] and beyond.

[log_22-13-11.txt](https://github.com/user-attachments/files/17871702/log_22-13-11.txt)

**Source code review**
I reviewed the source code and detected some issues "Fix pending" and one issue not marked to fix in the last commit.
See my review notes: https://github.com/indilib/indi-3rdparty/pull/925/commits/176924a56b0ed24f5f92bcc433fc6e7a44e0d8b7

knro, 2024-11-22:
@naheedsa Please take a look

knro, 2024-12-02:
@fcasarramona Can you please check now?

closed 2024-12-02 by knro