indilib/indi-3rdparty #1013: Player One Poseidon-M: wrong driver automatically determined

AstroFromHome, 2024-12-15:
**Describe the bug**
I am running test for the to be released Touptek Astrostation. The driver that is determined is the Player One Camera1.
This is the wrong driver that the code determines. I have investigated in all the files for the Player One cameras.
The Camera1 is for a Sedna-M camera. 
 Camera2 is for the Poseidon-C, so the color version.
From my point of view the 3rdparty/playerone package also needs to refer the Poseidon-M or Poseidon-M Pro to the Player One Camera2 package.
With the wrong package that seems to be selected as a fallback solution as the Posidon-M does not have a driver associated. Maybe the driver package also finds Camera1 for the Sedna-M the most fitting one as both are monochrome cameras.

With the wrong driver selected I am stuck in the Gain range 0-100, the readout mode cannot be selected nor the dew heater can be activated. Camera2 driver does not start the camera as the camera name is not found.

From my limited view as a non developer I think the Camera2 just needs to be enhanced by the Poseidon-M and that the Bayer matrix availability would need to be checked.

Sorry for mixing packages, driver, etc but I am no developer.

Nevertheless I hope you understand the problem that I have and what I think could fix it.

AstroFromHome, 2024-12-15:
@hiro3110i I could see that you have been heavily involved into other issues with Player One cameras. Maybe you know an easy fix that the correct driver package is determined.

knro, 2024-12-16:
indi_playerone_ccd should be the driver used.  I think indi_playerone_single_ccd should be removed unless it is really necessary. We removed it for ZWO since it led to more problems than solutions.

AstroFromHome, 2024-12-16:
Thanks a lot for the information! 
Let see if I can make them to use the indi_playerone_ccd driver instead.

I am going to report back here.

hiro3110i, 2024-12-16:
@AstroFromHome Thank you for reporting your issue.

First of all, the playerone_ccd driver and playerone_single_ccd are the same implementation and are not tied to any specific camera model. The only difference is that the playerone_single_ccd driver is limited to opening only a single camera device. The playerone_ccd driver can open multiple camera devices at the same time.

Therefore, I recommend using the playerone_ccd driver for most general purposes.

Now, let me explain why I have left the playerone_single_ccd driver instead of deleting it.

In many cases, the palyerone_ccd driver, which supports multiple devices, is sufficient, but due to the logic for automatically discovering devices, you cannot open specify camera (specific serial number of camera device) manually. This is because it depends on the order in which the devices are discovered. In this way, I have left the playerone_single_ccd in case you need to specify the intended correspondence one-to-one with the camera device.
This is also very useful if you want to give nicknames to specific cameras to distinguish them.
These settings are stored in an .xml file in the ~/.indi folder.
If you want to reset them, you can do so by deleting the .xml file for that camera.

I know that the asi_single_ccd camera driver has been removed, but I don't understand the real reason.

AstroFromHome, 2024-12-16:
@hiro3110i thank you very much for your analysis. So trying the playerone_ccd driver should generally fix it if I understand correct. Just giving nicknames would not work then because of the missing one to one correspondance. I guess not having nicknames for the cameras is ok. Else I guess the _single_ driver could be enhanced with all other camera models that are needed.

As a non-developer I then misunderstood that the _single_ driver would automatically fall back to the Sedna-M camera.

Do you think it would then work once the playerone_ccd driver is impelemented that also my Poseidon-M (Poseidon-C is in the _single_driver) would have all settings available? Other people may use the Uranus, Ares, Artemis and so on cameras from PlayerOne. Those should also work with full availability of the settings.
Could you please also guide me where to check that the settings like Gain, High and Low conversion modes, etc are then available once a camera is connected?
Just being able to select Gain 0-100 is pretty crappy on the PlayerOne camera as most of them switch to HCG at Gain 125.

Having a PlayerOne driver in place for the most general purpose is what is intended.

hiro3110i, 2024-12-16:
@AstroFromHome Once again, playerone_single_ccd driver has no specific camera model information.
If you feel it, please check ~/.indi folder. Such settings are memorized in the .xml files.
If you don't have any .xml files for PlayerOne Cameras, playerone_single_ccd driver will discover connected camera for any model. Could you please check .xml files first?

knro, 2024-12-16:
The reason asi driver was removed is that the single driver didn't work nicely with other single drivers for ZWO, it was causing errors. If it is working for PlayerOne then great, keep it!

AstroFromHome, 2024-12-16:
@hiro3110i thank you very much for your feedback it is highly appreciated! As said I initially thought that there is a fallback which you have well explained that this is not present.
Unfortunately I do not have any access to the ~\.indi folder that I cannot check the .xml file.

By the way are the entries PlayerOne Camera1, Camera2 and Camera3 also visible in case the playerone_ccd driver was used or is it something specific to the playerone_single_ccd driver only?

I have already tried to decompile the .apk but I cannot find any indi components in it. It looks like they are managing those seperately.
So I need to wait until one of the developers is coming back to me. Maybe he can also tell me if the .xml in the ~\.indi folder is deleted during shutdown or if it stays. If the .xml remains I need to get to know how I can delete it.

I just see that the PlayerOne Camera1 entry is being exchanged with Player One Poseidon-M Pro after selecting the PlayerOne Camera1. The Camera2 or Camera3 entries do not connect my camera at all. I would be fine with it in case I would be able to set Gain, readout mode, etc properly. Then I would not care about what kind of driver is running.

pana3110, 2024-12-17:
@AstroFromHome What is your platform/OS running indi?

PlayerOne Camera1, Camera2, Camera3 is playerone_single_ccd driver,
and PlayerOne CCD means playerone_ccd which is the multi device driver.

If you're not sure, simply choose the multi-device PlayerOne CCD.
This should work with any camera model without any issues.
If the camera model is correctly recognized the first time, the settings for connected camera will be displayed.

The playerone_single_ccd driver is for advanced users who can access XML settings in ~/.indi folder.
Let's say you connect two cameras of the same camera at the same time.
The camera model name alone won't allow you to distinguish between them on the UI.
In that case, by using the single driver and giving them nicknames, you can reliably identify which one is which.

AstroFromHome, 2024-12-17:
@pana3110 thank you so much for all the information!
I already handed over all this valueable information to my contact. The user should not have any access to the .xml in the ~/.indi folder.
So I proposed them to use the playerone_ccd driver only. The device is similar to the AsiAir where you have just one imaging camera and one guiding camera. So the user will never use two same cameras at the same time with the need to distinguish them. Except they use a guide camera for imaging.

Maybe I have been the wrong messanger here as I am just an unaffilitaed tester for the device. But as you may imagine I would like to have my gear working as well. 

I was now also able to find the playerone_ccd driver on the device. But the same result. I assume prior to the compilation of the driver inappropriate values have been configured. The playerone_ccd driver shows the same misbehavior on the device.
It looks like they are giving this back to the developers to rework the driver.

In paralell I have been trying to quickly setup astroberry on a loose RasPi4 but the driver compiled into the build is unable to recognize my camera. Most likely the driver used in the downloadable version of astroberry is too old to recognize my camera and the automatic update function does not work anymore. I have by far too little knowlege to update the driver manually on my RasPi in order to find out if a newer version will do the trick.

I have also seen that someone with the Ares camera from PlayerOne had issues with Gain using SDK 3.6.1 but I can nowhere find informatio if SDK 3.7.0 resolved the issue.

I could see from what you have posted in the past you have a Uranus-C II camera as the only PlayerOne camera. If you like I could try with my Poseidon-M if the current playerone_ccd works with this one with some help how to update the driver in astroberry.

Last but not least I will just wait upon the company's solution.

hiro3110i, 2024-12-18:
@AstroFromHome Astroberry is still 32bit based, indi driver has been fully migrated to 64bit and no longer supports 32bit.

I'm using indi driver v2.1.0 with KStars/Ekos v3.7.4 on my Raspberry Pi4/Ubuntu Mate 22.04 and Pi5/Ubuntu 24.04.
The latest driver should be used because new models of camera were not supported by older driver.
At least, playerone_ccd driver v2.1.0 can work with Neptune-C II, Uranus-C, Saturn-C, Proseidon-C and Sedna-M when I tested with them.
I believe Poseidon-M should be also work fine, but I'm asking the test with Poseidon-M to PlayerOne's engineer.

AstroFromHome, 2024-12-18:
@hiro3110i thank you for the information. Thank you as well to get in touch with the P1 engineer to verify.
Today I have installed Raspberry OS 64bit on my Pi4.

As someone with 0 knowledge on Linux I was trying to install indi drivers and kstars. The indi drivers seem to be ready to be used. Unfortunately when trying to install kstars with the steps provided on the relevant githup page I end with erros that two ECMconfig.cmake files or so are missing.
Now I see it as a challenge to prove that the problem that I have discribed in the first post is an issue with the driver that was implemented.

Additionally with the script it looks like I was able to install the stellarsolver. With the XISF package I am also still struggeling.

Maybe I need to give it another try using Ubuntu Mate like you do.

hiro3110i, 2024-12-20:
@AstroFromHome I don't recommend building it from source code without sufficient prior knowledge.
I recommend installing pre-built packages from the apt repository.
`sudo apt-add-repository ppa:mutlaqja/ppa`
`sudo apt-get update`
`sudo apt-get install indi-full kstars-bleeding`

AstroFromHome, 2024-12-20:
@hiro3110i thank you very much for the advise.

I am going to install Ubuntu over the weekend and I am going to try installing the pre-built packagabes. I switch from a Debian based OS to Ubuntu as I have seen that kstars should be easier to install via apt should work better how I could read online.

Building from source code worked at least for the indi drivers quite good. Just two drivers failed during cmake. But the easier the installation works the better for me as someone without any prior knowledge. I will keep everyone posted here.

AstroFromHome, 2024-12-20:
@hiro3110i your advice was brilliant for me as a person without any prior knowledge.
I installed Ubuntu Mate 24.05 LTS or so and then Indi plus KStars are precompiled packaged.

What shall I say? The playerone_ccd driver just works absolutely fine with the Poseidon-M. To be I could not imagine that the monochrome woud not run as the color version works and both share the same hardware except the not present bayer matrix on the monochrome camera.
Attached a screenshot in which you can see that the driver is capable to do the proper Gain settings. On top a photo of my test setup on which you can see the tiny P4 on the right hand side. The control box (black with antenna) that I am doing user tests for that is like the AsiAir is attached to the telescope on the finder shoe which is attached to the focuser.

So the issue should be resolved now. The problem is a too old driver or that the SDK needs to be integrated as seperate SDK like those for ZWO or QHY.

Nevertheless thank you so much for all your help!
Trying to find the root cause and if INDI is really able to control the camera was so much fun that I am thinking about to afford a mini PC or so to run it with Linux and then learn how all this compiling from source and so one really works. You have opened up a completely new point of view for me.
 
[image]
[image]

hiro3110i, 2024-12-21:
@AstroFromHome Congratulations!  I'm happy to hear your successful setup.
I usually use KStars/Ekos for serious photography with its internal guiding control on Raspberry Pi.
Auto focus and plate solve are also working fine.
RasPi5 has very good speed to transfer and plate solve for full resolution pictures.
Enjoy!

AstroFromHome, 2024-12-21:
@hiro3110i thank you very much! I feel really feel full of self-esteem that I was able to do the full test that I have intended to do. Even though I would have struggled a lot more without all the information from you.

The Pi4 seems to be a little weak for a full Ubuntu distribution. So I will look into the Pi5.
My Pi4 just has 4GB RAM so even the test was filled long lags prior to the next possible mouse click.
I use AsiAir, Touptek Astrostation and NINA on a mini PC. Such a mini PC I guess would be the best option to run Ekos/Kstars.

Having had this success yesterday made me eager to test out some Linux to broaden my point of view.

I have passed on all that I was able to find out with your deep and patient help.
As my problem is resolved that newer drivers and SDK are needed I am closing this thread.

closed 2024-12-21 by AstroFromHome