indilib/indi-3rdparty #1016: SVBONY Driver of OEM by ToupTek

jctk, 2024-12-30:
Hi @knro
Cc: @touptek 

A representative from SVBONY has asked us to implement an INDI driver for SVBONY's new SC715C (IMX715 CMOS), which will be ToupTek's OME product.
Could you please add it to one of the drivers for ToupTek and several OEMs?
Our contact at SVBONY provided us with the svbonycamsdk 20241119 file.
Looking at the API Document, it appears that the Touptek SDK v57 symbols have just been changed to SVBONY.

It is important to note that the existing SVBONY driver will also survive.
You need to ensure that the INDI driver naming does not duplicate.

The attached file is a subset of the original svbonycamsdk, leaving only the documentation and the library for linux x64.
The full set is too large to attach.
[svbonycamsdk.20241119.subset.zip](https://github.com/user-attachments/files/18272367/svbonycamsdk.20241119.subset.zip)

**Provide link to product page**
This camera has not yet been released.

**Describe device family**
* Cameras (CMOS)

**Provide device communication protocol**
The documentation in the attached file describes it.

**Hardware Sample**
I do not currently have any equipment on hand.
I will contact SVBONY if needed.

knro, 2024-12-30:
Thank you. I need to ask @touptek to add it to their library.

jctk, 2025-01-12:
Hi @knro
Cc: @touptek

Could you please let me know the status of this request?
If you can't add svbony driver as an OEM driver for touptek, I will need to add libsvbonycam independent of the touptek library and develop an INDI driver with a ToupCam clone.

knro, 2025-01-12:
@jctk I spoke to Touptek and they said they will work on it. I think you need to get in touch with them directly to move this forward.

jctk, 2025-01-12:
@knro 
Thanks for the explanation to ToupTek.

@touptek 
I have no direct contact information for Touptek. I hope Touptek will participate in this Issue #1016.

jctk, 2025-01-18:
Hi @touptek 
The SVBONY rep explained to me that they will provide a device for testing if I need it for testing.
Do you need it?

jctk, 2025-01-22:
@knro 
I would like to contact a @Touptek about providing SVBONY prototype devices for testing during development.
Can you tell me how to contact @Touptek?

knro, 2025-01-22:
Sure, can you send me an email and I'll put you in touch? mutlaqja at ikarustech dot com

knro, 2025-02-02:
This is now integrated with latest touptek PR.

closed 2025-02-02 by knro