indilib/indi-3rdparty #1017: libasi/armv6/libASICamera2.bin not an armv6 file

HaraldStauss, 2025-01-09:
**Describe the bug**
The file libasi/armv6/libASICamera2.bin not an armv6 file is not an armv6 file.

**To Reproduce**
use:
file libASICamera2.bin
it will show that the file is a 64bit file. The armv6 processor is 32bit only.

**Expected behavior**
The correct file from the ASI SDK need to be provided.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: DietPi on Raspberry Zero W
 - Version: Don't recall

**Log Files**
N/A

The current SDK from the ASI download site contains the correct file. I replaced the file on my system and INDI-3rd party would compile properly. Thank you for switching out the file with the correct one. 

Regards,
Harald Stauss

HaraldStauss, 2025-01-11:
I see that you have exchanged the file. Thank you. I have not yet tested it. But I assume this will fix the problem on the Raspberry Pi Zero W.

Thank you,
Harald

knro, 2025-02-02:
At any rate, we get the binaries from ZWO. So you might need to follow up with ZWO if there is an issue with architecture.

closed 2025-02-02 by knro