indilib/indi-3rdparty #1018: aag_cloudwatcher_ng not working with latest CloudWatcher hardware and firmware

jmocenigo, 2025-01-09:
**Describe the bug**
So, this may be me or it may be a bug, but indi_aagcloudwatcher_ng is not working for me.  I just received my CloudWatcher directly from Lunatico Astronomia within the last month. It has the latest firmware. The specific error I see (repeatedly) is:
```
2025-01-09T19:57:00: [ERROR] Can not get data from device 
2025-01-09T19:57:00: [ERROR] ERROR in getIRSkyTemperature 

```
Looking at the code, specifically "CloudWatcherController::getValues" does not seem to match the latest documentation found at:
   [Fifth Addenda](https://lunaticoastro.com/aagcw/TechInfo/Rs232_Comms_v140.pdf)
In that the parse order of the first "sscanf" function call  (and others, perhaps) seems wrong.

Note: I can run the Lunatico supplied Windows software on my Linux box under "wine" and it works fine. Screenshot attached (FYI: readings shown in F and mph). So, the hardware is properly attached.

**To Reproduce**
Get a recent CloudWatcher (with SQM sensor) and 5.89 firmware and try to start "AAG Cloudwatcher NG" weather module.

**Expected behavior**
After a short delay, the above error messages will appear cyclically and all readings in the "Device Raw Readings" tab will be zero.

**Screenshots**
A screenshot of the Device Raw Reading tab and error log is attached.

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04LTS
 - Version: KStars 3.7.4 Stable

**Log Files**
log_15-11-11.txt attached.

[image]
[image]
[log_15-11-11.txt](https://github.com/user-attachments/files/18366955/log_15-11-11.txt)

jmocenigo, 2025-01-10:
P.S. If you agree that it is a bug, then I will be happy to fix it. Just let me know. Thanks.

knro, 2025-01-10:
It would be great if you can find the root cause. I tried to fix these issues a while ago but I don't have an actual unit to test with. So a PR would be greatly appreciated!

jmocenigo, 2025-01-10:
Is this still the proper process to follow?
[How to Contribute to INDI Github Development](https://indilib.org/develop/tutorials/181-how-to-contribute-to-indi-github-development.html)
If not, please send an updated link.

knro, 2025-01-11:
Sure, you can still use this guide.

jmocenigo, 2025-01-14:
Jasem: So, I believe I am making good progress, but I have a question: there is a Site Management panel that contains Lat/Long/Elevation information. If I have access to elevation, then I can compute relative pressure in addition to absolute pressure. So, how can I query for the elevation value? Can you point to an example (or provide one)?

Incidentally, I get the feeling that in practice one would not use this driver as it stands since you don't want to start and stop the weather station, but rather leave it running continuously. So, I suspect one with CloudWatcher hardware would prefer to be doing one of the following rather than use this driver directly, since it runs in so-called "master" mode:

1. Run in "slave" mode (such an option to allow this ability would need to be added to this driver) though  option #2 below is likely the better way to go in this regard;
2. Use the Weather Watcher driver (assuming it can process the AAG json file intended for "slave" mode);
3. If the user has the Lunatico Astro Solo, then use the Weather Watcher driver to access its URL (assuming Weather Watcher can handle that format format).

I have not tried Weather Watcher yet, so I don't know if it already has these abilities.

I will say that with some additions, this driver could provide value (aside from session operation of CloudWatcher) as a way for non-Windows users to update the CloudWatcher firmware (options to do that would have to be added, but at least the serial communication protocol supports it). This driver could also be extended to allow non-Windows users to alter CloudWatcher's parameters that are stored in non-volatile memory.

Jack-Slingshotter, 2025-01-30:
I'm just commenting here to say that my company was experiencing similar issues with this driver. By cloning this repo and debugging for a few days, I was able to write a patch that got this driver working. My manager is currently in the process of checking whether or not I'm allowed to share this patch with you guys

jmocenigo, 2025-01-31:
I am still working on this… there are a lot of things to fix as it was out-of-date and did not take advantage of the latest features (but still need to keep backwards compatibility).

Life occupied me elsewhere for a while, but I am back at it now.

I am also adding Source/Replica option (what Lunatico docs refer to with old terminology as master/slave) so CW does not need to one restarted with each EKOS session.

P.S. I resolved my question about obtaining elevation information.

knro, 2025-01-31:
> I'm just commenting here to say that my company was experiencing similar issues with this driver. By cloning this repo and debugging for a few days, I was able to write a patch that got this driver [...]

Please let us know as we we going to tag next INDI release tomorrow.

jmocenigo, 2025-01-31:
I am not ready for tomorrow. Don’t know about Jack, of course. Sent from my iPhone

jmocenigo, 2025-03-03:
I have been trying out Weather Watcher and I am now successfully using it to connect to my Solo CloudWatcher via the Solo's cgi-bin/cgiLastData access method, it seems to me that it will handle things just fine after adjusting for the parameter names (see below).

So, rather than try to extend aag_cloudwatcher to handle data from a serial connection or from a file, I will just concentrate on its original function, namely the serial connection. Most of that work is done, but I now need time to physically disconnect the CloudWatcher from the Solo and attach it to my NUC to confirm the code changes work.

Sorry for dragging this out so long.

Output of http://<solo-lan-address>/cgi-bin/cgiLastData with latest parameter names:

>    ataGMTTime=2025/03/03 21:15:42
>    cwinfo=Serial: XXXX, FW: 5.89
>    clouds=-32.310000
>    cloudsSafe=1
>    temp=4.480000
>    wind=4
>    windSafe=1
>    gust=5
>    rain=3340
>    rainSafe=1
>    lightmpsas=9.95
>    lightSafe=2
>    switch=0
>    safe=0
>    hum=24
>    humSafe=1
>    dewp=-14.360000
>    rawir=-31.500000
>    abspress=995.362500
>    relpress=1025.170493
>    pressureSafe=1
>

jmocenigo, 2025-03-14:
Finally pushed my changes to GitHub. Hopefully the commit comments explain changes. If not, please contact.

jmocenigo, 2025-04-15:
Left open by mistake.

closed 2025-04-15 by jmocenigo