indilib/indi-3rdparty #1019: Force pushing master branch lost some commits

jpaana, 2025-01-09:
**Describe the bug**
Some commits made at the end of December got lost due to force pushing to master branch.

**To Reproduce**
Looking at git log from an older copy of the  repo that hasn't been force pushed to shows commits that are missing from the current origin branch.

**Expected behavior**
Force push shouldn't be used as that usually messes commit history.

**Screenshots**
[image]

jpaana, 2025-01-09:
Missing commit hashes are: 
- 7f612f096dbff00d439b49968ca1c784668d97dc
- 963c8ac9cb8ea933ed7f8b2ae48d764e8b73d505
- 3eb4428540d07bdf2bae1d0709b3b17013178391

Plus a15affe6071e6abd42f6eb310f1cdc4e4d3246a0 but that's probably superceded by 4cbf51a7bdfa0469337f98c034a02dbf37f9e61d

They could probably be simply cherry-picked to recover them.

TallFurryMan, 2025-07-12:
7f612f0 and 3eb4428 seem to have been superseded.

963c8ac might still be relevant, not sure.