indilib/indi-3rdparty #1020: indi-eqmod could not get the correct  EQ-8R zero position. DEC readout is 80

yecloud, 2025-01-10:
**Describe the bug**
Driver: indi_eqmod_telescope
Connection: USB
OS: Linux 5.15.0-102-generic #112-Ubuntu SMP  x86_64 
DEC readout is 80 after the mount restarted at park position.

**To Reproduce**

1. Run driver, DEC readout is 90
2. Power off the mount and shutdown indiserver
3. Power on the mount and start indiserver
4. The DEC readout is 80
5. The only way to reset it to 90 is to perform a find home.

**Expected behavior**
The DEC readout should be 90.

**Desktop (please complete the following information):**
 - OS:Linux 5.15.0-102-generic #112-Ubuntu SMP  x86_64 
 - Version 1.3

knro, 2025-01-10:
You didn't set custom parking? what is the index files on startup? I have EQ-8 that I use often and never experienced this.

yecloud, 2025-01-10:
> You didn't set custom parking? what is the encoder index/step on startup? I have EQ-8 that I use often and never experienced this.

no custom parking.

[image]

stevelcb, 2025-01-21:
Also happens if a meridian flip has occurred, without turning the power on and off. Logs here:

[drive.google.com/drive/folders/1Oov1igh_...pmnzn4nh?usp=sharing](https://drive.google.com/drive/folders/1Oov1igh_KsMYR6q6t_Mb6A_upmnzn4nh?usp=sharing)

stevelcb, 2025-02-19:
Please note that this report is for the EQ8-r mount. Not the older EW8.
Thanks