indilib/indi-3rdparty #1028: Player-One Filter Wheel driver frequently crashes on restart of indiserver

vpnels, 2025-01-22:
**Describe the bug**
I am using Player-One Filter Wheel attached to a Player-One Poseidon-M camera. The chain is properly configured and runs correctly. However, the Player-One Filter Wheel driver frequently crashes on startup, but especially on re-starts of indiserver. It often takes a few restarts to get past a crash and have it start properly. 

My system is EKOS on StellarMate Pro. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. From EKOS press start. 
2. Observe on console that indiserver is starting and drivers are loading. 
3. Observe error message “Driver Crash” INDI Driver indi_playerone_wheel crashed. Restart it?
4. Repeat until error does not occur

**Expected behavior**
I expected that the profile would load all drivers and run. 

**Screenshots**
See screenshot

[image]

**Desktop (please complete the following information):**
 - OS: StellarMate Pro 1.8.6 (PRO-3)
 - INDI Library 2.1.1 Protocol 1.7

**Log Files**
Not yet available. Will have later.

vpnels, 2025-01-22:
Where do I find the pertinent logs? I looked in /var/logs but don’t see any pertinent logs.

knro, 2025-01-22:
For crashes, I'd say use [Ekos Debugger](https://stellarmate.com/support/tutorials/ekos-debugger.html) as you can get the backtrace which would be useful to know where the crash occured.

promero11, 2025-01-24:
Hi knro,
Thank you for quality drivers for Player One. I have this problem, too, and it is pretty severe.  In using the previous version of Stellarmate, I thought I had my disconnect issues using ASCOM, NINA native driver licked, but now, after installing Stellatmate 1.8.6 on Kubuntu 24.04 on a mini NUC PC, its bad. I tried to look at the logs folder in the program, but no logs are being created, even in verbose mode. I am running two identical cameras/filter wheels. The only other thing I can think of is a bad USB C connector on the cameras combined with really low temperatures this time of year.

Clear Skies,

Paull

hiro3110i, 2025-01-25:
@vpnels What EFW product are you using?
I have tested EFW driver with 7x36mm and 7x2" wheel on Raspberry Pi4/Ubuntu Mate 22.01 and Pi5/Ubuntu 24.01.
One possible reason for this issue is the compatibility of the PC's USB port, which may depend on the power supply capacity, especially if other devices (such as a camera) are connected at the same time.
If there is a setting for the power supply capacity of the USB port, set it appropriately to ensure sufficient power, or use a USB hub with a high power supply capacity.

promero11, 2025-01-25:
hiro3110i,

Thanks for looking in to this quickly. I run a Poseidon camera and7x36mm EFW. The disconnects are hardware based, I believe, as evidenced by no disconnects after:

1) I removed the back panels of the Player One cameras and found the EFW connectors misaligned, resulting in a poor connection when the cover is on. (see attacvhed photos).

2) I switched from a mini NUC PC running on 12 volts to a NUC PC running on 120v AC and subsequently turning all ports power in the BIOS.

Clear Skies,
Paul

[image]

[image]

hiro3110i, 2025-01-26:
@promero11 Thank you for sharing photos. I don't know sure it is in their expectation.
Anyway, you should contact to customer support after February 10.

closed 2025-08-04 by knro