indilib/indi-3rdparty #1029: Player-One Poseidon-M Camera has light band at top of dark frame

vpnels, 2025-01-22:
Describe the bug
I am using a Player-One Poseidon-M camera. The chain is properly configured and runs correctly. However, dark frames from the camera have a light band at the top of the image. This was seen in Windows native driver and ASCOM driver when this camera was introduced in 2022. The native driver and ASCOM driver were fixed for this problem and work perfectly under NINA. I am migrating to EKOS/INDI on a new StellarMate Pro. I recognize the same problem on INDI now. 

My system is EKOS on StellarMate Pro.

To Reproduce
Exact steps to reproduce the behavior.

Make sure optical chain is completely blocked from light to take a dark frames
Start the profile with the Player-One Poseidon-M as the imaging camera
Press the Camera menu icon on top right
Press the Capture button and initiate a 2 second exposure
Wait for captured image to appear 
Observe lighter bar at top of image

If camera is in Normal mode, the lighter bar is wider. 
If the camera is in Low Noise mode, the lighter bar is narrower, but still present. 

The presence of this lighter bar causes calibration problems when stacking images in PixInsight. 

Expected behavior
I expected that the dark frame is evenly dark with no lighter bar at top edge. 

Screenshots
See screenshot

[image]

No logs. Still trying to figure out how to get the logs.

vpnels, 2025-01-22:
The dark frame exposure was 2 seconds. Camera in Normal mode. Gain 100. Offset 20. The band does not depend on gain.

knro, 2025-01-22:
@hiro3110i Can you please take a look?

hiro3110i, 2025-01-25:
@vpnels Thank you for reporting the issue.
I'm using Poseidon-C with indi driver on Raspberry Pi4 and Pi5.
I think it is not a issue of indi driver code itself.

From my experience, Poseidon camera needs a little bit much electric power.
I saw such kind of band on my image when there is a power shortage.
If the PC's USB bus power supply capacity is low, using a USB hub that can supply power may be the solution.
Or, you should check firmware version of Poseidon camera.
Please contact to PlayerOne support.

vpnels, 2025-01-25:
Hi Hiroshi,

Of course, P1 support is away until February 9 for Chinese New Year. I have been already in contact with them.

About the Poseidon-M firmware. Where do you look for firmware? How do you upgrade? There is no firmware for the Poseidon-M on the P1 support and software web page.

Also, the USB port that I'm using is one of the USB3 (blue) ports on the Stellarmate Pro. I am providing power to the SMP from a SM 10A AC supply. One of the key motivations to switching to the SMP is cable management and clutter reduction. I don't want to have to add yet another box to the telescope (powered USB hub).

hiro3110i, 2025-01-26:
@vpnels They don't release firmware updates publicly.
In most cases, the factory settings work fine.
However, in certain circumstances, some measures may be necessary.
I recommend contacting support for assistance.