indilib/indi-3rdparty #1034: SVBony CCD not able to connect anymore after latest update

RoP09, 2025-02-02:
**Describe the bug**
I am using a SVBony SV305M Pro and until yesterday night - during a session while everything was working fine - I saw that there was an update provided for INDI and Stellarmate OS (KStars) and gave it a try.
After the update was done and a reboot was performed I could no longer connect to my SVBony camera (which is my guiding cam). The rest of the system seemed to work fine

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect to SVBony CCD camera (SVBony SV305M Pro)

**Expected behavior**
Camera  connects and works correctly

**Actual behaviour **
Camera cannot connect, instead an error message is shown.
INFO 226.306300 sec : Session log file /home/stellarmate/.indi/logs/2025-02-01/indi_svbony_ccd/indi_svbony_ccd_21:47:24.log
ERROR 251.349792 sec : Error Initializing the CCD (SVB_ERROR_INVALID_INDEX).

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Stellarmate OS  / Raspberry Pi 5
 - KStars 3.7.5

**Log Files**
INFO 226.306300 sec : Session log file /home/stellarmate/.indi/logs/2025-02-01/indi_svbony_ccd/indi_svbony_ccd_21:47:24.log
ERROR 251.349792 sec : Error Initializing the CCD (SVB_ERROR_INVALID_INDEX).

knro, 2025-02-02:
@jctk Can you please take a look? is this a regression?

jctk, 2025-02-02:
@knro 
Upgrading with the following commands reproduced this problem.
```bash
$ sudo apt update
$ sudo apt dist-upgrade
```

Deleting the following files solved the problem.
~/.indi/SVBONY*.xml
~/.indi/SVBONY*.xml.default

It is unlikely that #1021 is caused by the SVBONY Camera SDK being replaced with 1.13.2 from Vesion 1.12.8.
This problem did not occur during my development.

Have you made any changes to the INDI classes or anything else since #1021?

jctk, 2025-02-02:
@knro 
The cause is the SDK update in #1021.
Some parameters are missing in SDK 1.13.2, which causes an error when opening the old configuration xml.
For SVBONY SV405CC, Sharpness is gone in 1.13.2.
I don't know for SV305M Pro because I don't own it.

RoP09, 2025-02-02:
> [@knro](https://github.com/knro) Upgrading with the following commands reproduced this problem.
> 
> $ sudo apt update
> $ sudo apt dist-upgrade
> Deleting the following files solved the problem.
> 
> ```
> ~/.indi/SVBONY*.xml
> ~/.indi/SVBONY*.xml.default
> ```

does not work for me. If I delete both these files I am  getting exactly the same error as before.

jctk, 2025-02-02:
Could you attach a debug log for INDI, Capture and CCD?

See following instructions for debug log:
https://www.indilib.org/support/logs-submission.html

jctk, 2025-02-02:
Can you please attach the log files in the date subfolder under the following directory?
```
~/.local/share/kstars/logs/yyyy-mm-dd
```

RoP09, 2025-02-02:
Here's the log (not much inside though):

[log_17-52-07.txt](https://github.com/user-attachments/files/18633119/log_17-52-07.txt)

jctk, 2025-02-03:
@RoP09 
Thank you for providing the log file.
If you don't mind, could you please help me with the following 1 to 5?

1.
I have built an indi_svbony_ccd file that outputs a slightly more detailed log.
https://drive.google.com/file/d/15WMs7PYIGhBioW_iN1EDuP0GF8WzUDx-/view?usp=sharing

Place this file in `/usr/bin/indi_svbony_ccd` instead.
Also run `chmod +x indi_svbony_ccd`.

Can you attach the log file obtained with this file?
Please save the original file as an alias so that it can be restored.

2.
Can you please confirm if SV305M Pro can be used with the previous SVBONY Camera SDK?
https://drive.google.com/file/d/1kehqae_W2JLcRbWLWwJP9PcIo4XrVScs/view?usp=sharing

Please save the attached `libSVBCameraSDK.so.1.12.8` in `/lib/aarch64-linux-gnu/` and create a symbolic link with the following command.
This will cause indi_svbony_ccd to read the SDK 1.12.8 files.
```bash
stellarmate@sm5:/lib/aarch64-linux-gnu $ sudo ln -s libSVBCameraSDK.so.1.12.8 libSVBCameraSDK.so.1
```
3.
SVBONY Camera SDK automatically saves and restores the configuration with the Auto Save Configuration feature.
Files output by different versions of the SDK may not be compatible. (This happened a few years ago)

Please delete the files found with the following command.
```bash
stellarmate@sm5:~ $ sudo find / -name 'U*.bin' | grep 'AST_Cfg_\(SAVE\|A\)\.bin'
```

Here are the configuration files left in my environment. (no problems at this time)
```
/opt/FireCapture_v2.7/U3SM400C-AST_Cfg_SAVE.bin
/opt/FireCapture_v2.7/U3SM400C-AST_Cfg_A.bin
/home/stellarmate/U2SM201C-AST_Cfg_A.bin
/home/stellarmate/U2SM201C-AST_Cfg_SAVE.bin
/home/stellarmate/U3SM1001C-AST_Cfg_SAVE.bin
/home/stellarmate/U3SM400C-AST_Cfg_SAVE.bin
/home/stellarmate/U3SM900M-AST_Cfg_A.bin
/home/stellarmate/U3SM900C-AST_Cfg_SAVE.bin
/home/stellarmate/U3SM1001C-AST_Cfg_A.bin
/home/stellarmate/U3SM400C-AST_Cfg_A.bin
/home/stellarmate/U3SM900M-AST_Cfg_SAVE.bin
/home/stellarmate/U3SM900C-AST_Cfg_A.bin
/home/stellarmate/Projects/svb_test/U3SM400C-AST_Cfg_SAVE.bin
/home/stellarmate/Projects/svb_test/U3SM900M-AST_Cfg_A.bin
/home/stellarmate/Projects/svb_test/U3SM400C-AST_Cfg_A.bin
/home/stellarmate/Projects/svb_test/U3SM900M-AST_Cfg_SAVE.bin
/home/stellarmate/Projects/build/indi-3rdparty/U3SM400C-AST_Cfg_SAVE.bin
/home/stellarmate/Projects/build/indi-3rdparty/U3SM900M-AST_Cfg_A.bin
/home/stellarmate/Projects/build/indi-3rdparty/U3SM400C-AST_Cfg_A.bin
/home/stellarmate/Projects/build/indi-3rdparty/U3SM900M-AST_Cfg_SAVE.bin
```

4.
Check the SVBONY Camera SDK that is automatically loaded.
Execute the following command and tell us the output.

```bash
stellarmate@sm5:~ $ sudo ldconfig -p | fgrep libSVB
```

Example in my environment:
```
        libSVBCameraSDK.so.1.13.2 (libc6,AArch64) => /lib/aarch64-linux-gnu/libSVBCameraSDK.so.1.13.2
        libSVBCameraSDK.so.1 (libc6,AArch64) => /lib/aarch64-linux-gnu/libSVBCameraSDK.so.1
        libSVBCameraSDK.so (libc6,AArch64) => /lib/aarch64-linux-gnu/libSVBCameraSDK.so
```

5.
Check the SVBONY Camera SDK installed.
Please execute the following command and tell us the output.
```bash
stellarmate@sm5:~ $ ls -l /lib/aarch64-linux-gnu/libSVB*
```

Example in my environment
```
lrwxrwxrwx 1 root root      20  2月  1 22:05 /lib/aarch64-linux-gnu/libSVBCameraSDK.so -> libSVBCameraSDK.so.1
lrwxrwxrwx 1 root root      25  2月  1 22:05 /lib/aarch64-linux-gnu/libSVBCameraSDK.so.1 -> libSVBCameraSDK.so.1.13.2
-rw-r--r-- 1 root root 5262928  2月  1 22:05 /lib/aarch64-linux-gnu/libSVBCameraSDK.so.1.13.2
```

RoP09, 2025-02-03:
Here is the log again, the rest to follow:

[log_17-30-02.txt](https://github.com/user-attachments/files/18645375/log_17-30-02.txt)

RoP09, 2025-02-03:
edit: I can confirm the old SDK works, the symbolic link was not deleted it seems, had to manually do that

jctk, 2025-02-04:
> edit: I can confirm the old SDK works, the symbolic link was not deleted it seems, had to manually do that

Yes. Symbolic links should remove old links.

> Old SDK does not make a difference it seems, same error.

Do you know what version of the SVB Camera SDK you used to use with the SV305M Pro without problems?

Looking at the logs I see a possible problem with SVBOpenCamera in the API of SVBONY Camera SDK 1.13.2.
The last indi_svbony_ccd source code change was on Sep. 14, 2024 and I believe I can rule out indi_svbony_ccd as the cause.
However, I am a bit puzzled that the problem reproduces in SDK 1.12.8.
Was it an SDK older than 1.12.8 that was working fine?

I will report this issue to my contact at SVBONY.
If it is fixed the SVBONY Camera SDK will be committed.

RoP09, 2025-02-04:
Sorry maybe my last answer was misleading. What i wanted to express is that the SDK Version 1.12.8 you provided is working fine with the camera (after i deleted the old symbolic link pointing to the v1.13.2 and creating a new one to 1.12.8).

knro, 2025-02-04:
I merged #1038 , can you check again @RoP09 ?

RoP09, 2025-02-04:
> I merged [#1038](https://github.com/indilib/indi-3rdparty/pull/1038) , can you check again [@RoP09](https://github.com/RoP09) ?

is that any different from the version @jctk  posted above? (https://drive.google.com/file/d/15WMs7PYIGhBioW_iN1EDuP0GF8WzUDx-/view?usp=sharing)

If no then the log is already attached here https://github.com/user-attachments/files/18645375/log_17-30-02.txt

jctk, 2025-02-04:
There is no need to retest.

The log file has already been received with indi_svbony_ccd, which has made the same changes as #1038.

jctk, 2025-02-05:
I have contacted my contact at SVBONY about this issue. It has been shared with their engineers.
I am hopeful that they will recognize it as a bug and fix it soon.

jctk, 2025-02-07:
@RoP09 
Attached is the corrected SDK and indi_svbony_ccd.
Could you please check if it works properly with SV305M Pro?

libSVBCameraSDK.so.1.13.4
https://drive.google.com/file/d/1Jwte_zos0tE3vhhIMobUkQF9Q88hHQN-/view?usp=sharing

indi_svbony_ccd
https://drive.google.com/file/d/15WMs7PYIGhBioW_iN1EDuP0GF8WzUDx-/view?usp=sharing

RoP09, 2025-02-10:
Will try tonight, hopefully!

RoP09, 2025-02-10:
1.13.4 seems to work fine so far, I can successfully connect to the camera. Howeevr I could not download the indi_svbony_ccd from your drive share (but it works without that using the previous version from above...)

jctk, 2025-02-11:
Thanks for testing with the latest SDK.
Now that the issue is resolved, could you please close this issue?
Already it has been merged into the repository.

The shared link for infi_svbony_ccd may be misconfigured.The cause of this issue is the SDK, so you don't need to be able to download that.I shared that with you just in case.

closed 2025-02-11 by RoP09

jctk, 2025-02-19:
The Google Drive references in this issue have been removed. This is because this issue has been resolved.
If you need it, please build it yourself or obtain it from your distribution.

marklangster, 2025-09-03:
This issue is still happening for me on mac with a fresh install of kstars (no previous configuration to cause issues with parameters). I am using a SVBony 305C camera with the 1.13.4 SDK.

[           org.kde.kstars.ekos] - Connecting  "SVBONY CCD SV305C"
[           org.kde.kstars.ekos] - "Connecting INDI devices..."
[           org.kde.kstars.indi] - SVBONY CCD SV305C : "[DEBUG] Attempting to open SVBONY CCD SV305C (CameraID=1)... "
[           org.kde.kstars.indi] - SVBONY CCD SV305C :  "[ERROR] Error Initializing the CCD (SVB_ERROR_INVALID_INDEX). "