indilib/indi-3rdparty #1039: Image corruption on Altair cameras

aaronwmorris, 2025-02-06:
Image data is corrupted with Altair cameras with default configuration where `CCD_CAPTURE_FORMAT` is set to `INDI_RGB`.

Changing `CCD_CAPTURE_FORMAT` to `INDI_RAW` resolves the problem.  Strangely, changing back to the original setting `INDI_RGB` the issue does not occur again.  Only by restarting the indiserver, restoring the defaults, does the problem re-occur.

 - OS:  Debian 12 bookworm
 - Arch: aarch64  (Raspberry Pi 4)
 - INDI version: 2.1.1
 - Cameras
     - GPCAM3 224c
     - 178C

Example output:

[image]
There is no lens on this camera, so there is no usable image.  You can see there is a black bar at the top, with corrupted data under.

aaronwmorris, 2025-02-06:
Linking https://github.com/aaronwmorris/indi-allsky/issues/1840

Here you can see how the image data is corrupted.

knro, 2025-02-07:
@touptek Can you please check?

closed 2025-09-05 by aaronwmorris