indilib/indi-3rdparty #1041: Problems with Playerone in Guide tab

alepensato, 2025-02-13:
@hiro3110i @knro the original problem is again alive, now with Ubuntu or Mint installed on a 64bit Ryzen processor.

It is present in the Guide tab, I tried with different seconds of exposure and seems that with a value lower than 1sec the looping is ok, but if you use 1sec or more.... the loop hangs and take ony the first capture.

I tried to replace the lated libplayerone with previous versions, but the problem is always present.

The problem does not occour on my Raspbian 64bit on my Rasberry 4

 _Originally posted by @alepensato in [#684](https://github.com/indilib/indi-3rdparty/issues/684#issuecomment-2657687787)_

hiro3110i, 2025-02-14:
@alepensato Thank you for reporting. Could you please let me know more detail of your system?
Platform (CPU, memory, mother board), OS, KStars, Ekos, Indi, playerone driver and PlayerOne camera model, etc.

I usually test PlayerOne drivers on Raspberry Pi4/Ubuntu Mate 22.04 and Pi5/Ubuntu 24.04.
I tested kstars-bleeding 3.7.4 and indi-full 2.1.1 with Sedna-M for guiding camera again.
The problem you mentioned does not appear on my test benches.
There may be a system-specific issue with the 64-bit Ryzen processor.
There may be issues with USB compatibility between the camera and your PC.
It may be necessary to fix the issue with the camera's internal firmware or the camera SDK, but since this is a rare case, I think it would be a good idea to share this with the PlayerOne engineers so they can analyze the problem.

@knro Are the stable versions that can be updated with apt from your PPA repository kstars-bleeding 3.7.4 and indi-full 2.1.1?
Are the latest kstars-bleeding 3.7.5 and indi-full 2.1.2 not yet distributed through your PPA now?

knro, 2025-02-14:
Both [libplayerone](https://code.launchpad.net/~mutlaqja/+recipe/libplayerone-stable) and [indi-player](https://code.launchpad.net/~mutlaqja/+recipe/indi-playerone-stable) are built based on INDI 2.1.2 on Feb 5th.

hiro3110i, 2025-02-14:
@knro I'm sorry, I can't update kstars-bleeding and indi-full from your PPA.
Could you please confirm your PPA repository?

knro, 2025-02-14:
What's the problem exactly?

alepensato, 2025-02-14:
> [@alepensato](https://github.com/alepensato) Thank you for reporting. Could you please let me know more detail of your system? Platform (CPU, memory, mother board), OS, KStars, Ekos, Indi, playerone [...]

Hi,
I tested with Ubuntu 24.X and Linux Mint 22.x with latest Kstars 3.6.5 from ppa:mutlaqja/ppa repos and from ppa.stellarmate.com repos. I have an HP notebook with AMD Ryzen 5 CPU and 16GB of RAM and with a Playerone Sedna-M

hiro3110i, 2025-02-15:
@knro I tried "apt update" and "apt upgrade" again, indi-full was updated to 2.1.2 but kstars-bleeding was still 3.7.4.
We cannot update to the latest kstars through ppa:mutlaqja/ppa repo.

@alepensato kstars 3.6.5 is very old version, so please test with the latest kstars and indi.
However, ppa:mutlaqja/ppa repo is something wrong now. But you can install previous version kstars 3.7.4.
To be honest, the code body of indi-playerone has not changed since the version that solved the problem you pointed out in the past. Only PlayerOne SDK was updated. As far as I know, newer SDK will be more stable.

knro, 2025-02-15:
I restarted the [build process](https://code.launchpad.net/~mutlaqja/+recipe/kstars-stable) so hopefully it will build for arm64 this time.

alepensato, 2025-02-15:
> [@knro](https://github.com/knro) I tried "apt update" and "apt upgrade" again, indi-full was updated to 2.1.2 but kstars-bleeding was still 3.7.4. We cannot update to the latest kstars through [...]

@hiro3110i sorry for the mistake i intended the 3.7.5 version

alepensato, 2025-02-15:
Now seems ok with lastest update    6:3.7.5+202502150813~ubuntu24.04.1

hiro3110i, 2025-02-15:
@alepensato I'm glad it was resolved.
I think there was probably a version mismatch between the camera SDK, indi driver, and/or kstars.

@knro Could you please confirm [build process](https://code.launchpad.net/~mutlaqja/+recipe/kstars-stable)?
amd64 was built soon but arm64 didn't start build process.

hiro3110i, 2025-02-15:
@knro Additional information.
The arm64 version of kstars-bleeding-data was built for Ubuntu 24.04.1.
However, kstars-bleeding and kstars-bleeding-dbg were not built.
It seems that the build script stops midway.

hiro3110i, 2025-02-16:
@knro Or, kstars-bleeding-data build completed because it is common to amd64, but the arm64 code seems not to be built.

hiro3110i, 2025-02-17:
@knro Now we can update kstars-bleeding for arm64. Thank you very much.
This topic should be closed.

alepensato, 2025-02-17:
Thanks for your work

closed 2025-02-17 by alepensato