indilib/indi-3rdparty #1050: Compilation error with atik driver

duke164, 2025-03-16:
Hi everyone,

After a lot try to connect my ATIK EFW 2 filter wheel and having read a lot of old thread (some years ago) about connection problem and I've tried to build the library for the device.
I mention that I can connect my filter wheel from windows so it is not a device problem. I need to connect my device with my setup based on RPI4 (I can connect my others devices without too much problems)
When I connect my device my usb port recognize the device (lsusb give some information about the device)
I'm running on RPi4 on MATE unbuntu 22.04
I've install the latest INDI library 
my libatik library is 2.2.9+202412030942~ubuntu22.04.1
I've followed the instructions given in the root README of 3rdparty
But I've got a lot of error because of operand type incompatibility
Some examples ; the 2 first lines 
Projects/indi-3rdparty/indi-atik/atik_wheel.cpp: In member function ‘bool ATIKWHEEL::setupParams()’:
/Projects/indi-3rdparty/indi-atik/atik_wheel.cpp:143:17: error: no match for ‘operator[]’ (operand types are ‘INumberVectorProperty’ {aka ‘_INumberVectorProperty’} and ‘int’)
  143 FilterSlotNP[0].setMin(1);
 ---------------------^
Projects/indi-3rdparty/indi-atik/atik_wheel.cpp:144:17: error: no match for ‘operator[]’ (operand types are ‘INumberVectorProperty’ {aka ‘_INumberVectorProperty’} and ‘int’)
  144 FilterSlotNP[0].setMax(numOfFilter);
  ---------------------^

Any ideas or what information could be useful to give a diagnostic?

Best regards,

NB : I'm not  a C programmer. I can understand because i made code in other language.

knro, 2025-03-17:
You need to update indi-core first since the filter properties were updated.

duke164, 2025-03-17:
Thanks a lot for your quick answer. I'll check as soon as I can get the time to.
I keep you inform about the result.
I'm not sur to solve the connection problem with my filterwheel but It is somewhat my last chance.

duke164, 2025-03-21:
I've tried to reinstall INDI library, compile it and reinstall the atik_wheel driver.
I've gotten the same error.
I do all of the actrions because the driver downlloaded and installed is 3.0 and I hope that 3.1 driver could perhaps help me to connect succesfully the atik EFW wheel
Hereafter the sequence

duke164@duke164-desktop:~/Projects/indi$ 

-- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found CFITSIO 4.0.0: /usr/lib/aarch64-linux-gnu/libcfitsio.so
-- Found INDI: /usr/lib/aarch64-linux-gnu/libindidriver.so;/usr/lib/aarch64-linux-gnu/libindiAlignmentDriver.so (found version "2.1.2") 
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Found USB1: /usr/lib/aarch64-linux-gnu/libusb-1.0.so (found version "1.0.25") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found Atik Library: /usr/lib/aarch64-linux-gnu/libatikcameras.so
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/duke164/Projects/build/indi-atik

duke164@duke164-desktop:~/Projects$ sudo apt-get -y install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libraw-dev libdc1394-dev libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev
[sudo] password for duke164: 
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
build-essential est déjà la version la plus récente (12.9ubuntu3).
libboost-dev est déjà la version la plus récente (1.74.0.3ubuntu7).
libboost-regex-dev est déjà la version la plus récente (1.74.0.3ubuntu7).
libfftw3-dev est déjà la version la plus récente (3.3.8-2ubuntu8).
libftdi1-dev est déjà la version la plus récente (1.5-5build3).
libgps-dev est déjà la version la plus récente (3.22-4ubuntu2).
libjpeg-dev est déjà la version la plus récente (8c-2ubuntu10).
libcfitsio-dev est déjà la version la plus récente (4.0.0-1).
libdc1394-dev est déjà la version la plus récente (2.2.6-4).
libftdi-dev est déjà la version la plus récente (0.20-4ubuntu1).
liblimesuite-dev est déjà la version la plus récente (20.10.0+dfsg-5).
libnova-dev est déjà la version la plus récente (0.16-5).
librtlsdr-dev est déjà la version la plus récente (0.6.0-4).
cmake est déjà la version la plus récente (3.22.1-1ubuntu1.22.04.2).
git est déjà la version la plus récente (1:2.34.1-1ubuntu1.12).
libcurl4-gnutls-dev est déjà la version la plus récente (7.81.0-1ubuntu1.20).
libraw-dev est déjà la version la plus récente (0.20.2-2ubuntu2.22.04.1).
libtiff-dev est déjà la version la plus récente (4.3.0-6ubuntu0.10).
libusb-1.0-0-dev est déjà la version la plus récente (2:1.0.25-1ubuntu2).
zlib1g-dev est déjà la version la plus récente (1:1.2.11.dfsg-2ubuntu9.2).
libgphoto2-dev est déjà la version la plus récente (2.5.31+202502051718~ubuntu22.04.1).
libavcodec-dev est déjà la version la plus récente (7:4.4.2-0ubuntu0.22.04.1+esm6).
libavdevice-dev est déjà la version la plus récente (7:4.4.2-0ubuntu0.22.04.1+esm6).
libgsl-dev est déjà la version la plus récente (2.7.1+dfsg-3ubuntu0.22.04.1~esm1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 2 non mis à jour.

uke164@duke164-desktop:~/Projects$ sudo apt-get -y install libindi-dev
[... 21 lines cut ...]
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Found USB1: /usr/lib/aarch64-linux-gnu/libusb-1.0.so (found version "1.0.25") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found Atik Library: /usr/lib/aarch64-linux-gnu/libatikcameras.so
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/duke164/Projects/build/indi-atik
Then the error
duke164@duke164-desktop:~/Projects/build/indi-atik$ make -j4
[ 50%] Building CXX object CMakeFiles/indi_atik_wheel.dir/atik_wheel.cpp.o
[ 50%] Building CXX object CMakeFiles/indi_atik_ccd.dir/atik_ccd.cpp.o
/home/duke164/Projects/indi-3rdparty/indi-atik/atik_wheel.cpp: In member function ‘bool ATIKWHEEL::setupParams()’:
/home/duke164/Projects/indi-3rdparty/indi-atik/atik_wheel.cpp:143:17: error: no match for ‘operator[]’ (operand types are ‘INumberVectorProperty’ {aka ‘_INumberVectorProperty’} and ‘int’)
  143 |     FilterSlotNP[0].setMin(1);

In the next message I'll will put some more information about my unsuccessful connection test

duke164, 2025-03-21:
Here some command I've test following an other thread to help to diagnose.
I reinstall INDI +3rd drivers by the normal process

duke164@duke164-desktop:~$ sudo apt-get install indi-full gsc
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
gsc est déjà la version la plus récente (1.3+202212170706~ubuntu22.04.1).
indi-full est déjà la version la plus récente (2.1.1+202412060837~ubuntu22.04.1).
0 mis à jour, 0 nouvellement installés, 0 à enlever et 2 non mis à jour.
duke164@duke164-desktop:~$ apt search libindi1 && apt info libindi1
En train de trier... Fait
Recherche en texte intégral... Fait
libindi1/jammy,now 2.1.2+202502010701~ubuntu22.04.1 arm64  [installé, automatique]
  Instrument-Neutral Device Interface library -- shared library

Package: libindi1
Version: 2.1.2+202502010701~ubuntu22.04.1
Priority: optional
Section: libs
Source: libindi
Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Installed-Size: 3 077 kB
Depends: libc6 (>= 2.34), libcfitsio9 (>= 3.480~), libcurl3-gnutls (>= 7.16.2), libfftw3-double3 (>= 3.3.5), libgcc-s1 (>= 3.3.1), libgsl27 (>= 2.7.1), libgslcblas0 (>= 2.7.1), libjpeg8 (>= 8c), libnova-0.16-0 (>= 0.16), libogg0 (>= 1.0rc3), libstdc++6 (>= 12), libtheora0 (>= 1.0), libusb-1.0-0 (>= 2:1.0.9), libxisf, zlib1g (>= 1:1.1.4), libindi-data (>= 2.1.2+202502010701~ubuntu22.04.1), indi-bin (>= 2.1.2+202502010701~ubuntu22.04.1)
Conflicts: libindi-plugins, libindi0, libindialignmentdriver1, libindidriver1
Replaces: libindi-plugins, libindi0, libindialignmentdriver1, libindidriver1
Download-Size: 704 kB
APT-Manual-Installed: no
APT-Sources: https://ppa.launchpadcontent.net/mutlaqja/ppa/ubuntu jammy/main arm64 Packages
Description: Instrument-Neutral Device Interface library -- shared library
 INDI (Instrument-Neutral Device Interface) is a distributed XML-based
 control protocol designed to operate astronomical instrumentation.
 INDI is small, flexible, easy to parse, scalable, and stateless.
 It supports common DCS functions such as remote control, data acquisition,
 monitoring, and a lot more.
 .
 This package contains the shared library.

duke164@duke164-desktop:~$ apt search indi-atik && apt info indi-atik 
En train de trier... Fait
Recherche en texte intégral... Fait
indi-atik/jammy,now 3.0+t202502050653~ubuntu22.04.1 arm64  [installé, automatique]
  INDI Driver for Atik cameras and filter wheels.

indi-atik-dbg/jammy 3.0+t202502050653~ubuntu22.04.1 arm64
  INDI Driver for Atik cameras and filter wheels.

Package: indi-atik
Version: 3.0+t202502050653~ubuntu22.04.1
Priority: extra
Section: science
Maintainer: Jasem Mutlaq <mutlaqja@ikarustech.com>
Installed-Size: 109 kB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 3.3.1), libindi1, libstdc++6 (>= 11), libatik
Download-Size: 30,6 kB
APT-Manual-Installed: no
APT-Sources: https://ppa.launchpadcontent.net/mutlaqja/ppa/ubuntu jammy/main arm64 Packages
Description: INDI Driver for Atik cameras and filter wheels.
 Driver for Atik cameras and filter wheels.
 .
 This driver is compatible with any INDI client such as KStars or Xephem.
 
with device plugged

duke164@duke164-desktop:~$ lsusb    
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c52b Logitech, Inc. Unifying Receiver
**Bus 001 Device 004: ID 0403:af01 Future Technology Devices International, Ltd **Atik Filter Wheel****
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Everything seems to be ok but whenb i try to launch the server I get the same behavior with the device plugged or not
d
uke164@duke164-desktop:~$ indiserver -v indi_atik_wheel
2025-03-21T20:45:24: startup: indiserver -v indi_atik_wheel
2025-03-21T20:45:24: Driver indi_atik_wheel: pid=4397 rfd=6 wfd=6 efd=7
2025-03-21T20:45:24: listening to port 7624 on fd 5
2025-03-21T20:45:24: Local server: listening on local domain at: @/tmp/indiserver
I have to kill the process

In EKOS the connection do not succeed also.

So i have no more idea to test something else.
Any help would be appreciated.

duke164, 2025-07-14:
Not sure what you want me to do. I saw your SDK update but I don't know how to implement it. I'll try to next week-end.
Thank you for your help.

duke164, 2025-07-15:
Ok I think I've understood. I'll test it next week end.

closed 2025-08-04 by knro