indilib/indi-3rdparty #1055: ASI676MC incorrect bayer pattern/offsets

aaronwmorris, 2025-03-20:
I have several users of indi-allsky reporting receiving random images with a purple hue from their ASI676MC.  The issue occurs at random.

https://github.com/aaronwmorris/indi-allsky/issues/2007

I am fairly certain that the issue is being caused by the image sensor being read from the wrong offset.  When the debayer process is applied, the wrong colors are rendered.  In the reported images, the residual bayer pattern is still visible due to the mismatch.  (OpenCV is used for debayering)

 - OS:  Debian 12 bookworm
 - INDI:  2.1.1
 - Arch:  aarch64 (arm64)

omeganon, 2025-03-20:
I’m experiencing the same.

knro, 2025-03-21:
Isn't this a ZWO SDK issue then? Anything filed with ZWO directly?

jimmauck, 2025-03-21:
> [@jimmauck](https://github.com/jimmauck) mentioned he opened a ticket with ZWO, but I do not have a link to that.
> 

I have received request acknowledgement from ZWO, but no correspondence yet.

[image]

aaronwmorris, 2025-09-05:
Closing this out for now.  We can revisit later if needed.

closed 2025-09-05 by aaronwmorris