indilib/indi-3rdparty #1058: Compilation error in AHP driver

nouspiro, 2025-04-01:
**Describe the bug**
Can't compile latest 2.1.3 on Ubuntu 24.04

**To Reproduce**
Compile with AHP enabled

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04
 - Version [2.1.3]

**Log Files**

```
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp: In member function ‘virtual bool AHPGTBase::updateProperties()’:
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:199:55: error: ‘get_acceleration_angle’ was not declared in this scope; did you mean ‘ahp_gt_set_acceleration_angle’?
  199 |         GTDEConfigurationNP[GT_ACCELERATION].setValue(get_acceleration_angle(1) * 180.0 / M_PI);
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~
      |                                                       ahp_gt_set_acceleration_angle
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:203:21: error: ‘get_mount_flags’ was not declared in this scope; did you mean ‘ahp_gt_set_mount_flags’?
  203 |         int fork = (get_mount_flags() & isForkMount) ? 1 : 0;
      |                     ^~~~~~~~~~~~~~~
      |                     ahp_gt_set_mount_flags
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:214:73: error: too many arguments to function ‘int ahp_gt_get_pwm_frequency()’
  214 |         GTConfigurationNP[GT_PWM_FREQ].setValue(ahp_gt_get_pwm_frequency(0) * 700 + 1500);
      |                                                 ~~~~~~~~~~~~~~~~~~~~~~~~^~~
In file included from /home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:11:
/usr/include/ahp/ahp_gt.h:569:16: note: declared here
  569 | DLL_EXPORT int ahp_gt_get_pwm_frequency(void);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp: In member function ‘virtual bool AHPGTBase::ISNewNumber(const char*, const char*, double*, char**, int)’:
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:263:37: error: too many arguments to function ‘void ahp_gt_set_pwm_frequency(int)’
  263 |             ahp_gt_set_pwm_frequency(0, (GTConfigurationNP[GT_PWM_FREQ].getValue() - 1500) / 366);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/ahp/ahp_gt.h:726:17: note: declared here
  726 | DLL_EXPORT void ahp_gt_set_pwm_frequency(int value);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp: In member function ‘virtual bool AHPGTBase::ISNewSwitch(const char*, const char*, ISState*, char**, int)’:
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:284:56: error: ‘GTFlags’ does not name a type; did you mean ‘GT1Flags’?
  284 |                     ahp_gt_set_mount_flags(static_cast<GTFlags>(0));
      |                                                        ^~~~~~~
      |                                                        GT1Flags
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:289:56: error: ‘GTFlags’ does not name a type; did you mean ‘GT1Flags’?
  289 |                     ahp_gt_set_mount_flags(static_cast<GTFlags>(0));
      |                                                        ^~~~~~~
      |                                                        GT1Flags
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:311:53: error: ‘GTSteppingMode’ does not name a type; did you mean ‘GT1SteppingMode’?
  311 |             ahp_gt_set_stepping_mode(0, static_cast<GTSteppingMode>(GTRASteppingModeSP.findOnSwitchIndex()));
      |                                                     ^~~~~~~~~~~~~~
      |                                                     GT1SteppingMode
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:317:53: error: ‘GTSteppingConfiguration’ does not name a type; did you mean ‘GT1SteppingConfiguration’?
  317 |             ahp_gt_set_stepping_conf(0, static_cast<GTSteppingConfiguration>(GTRAWindingSP.findOnSwitchIndex()));
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                     GT1SteppingConfiguration
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:323:47: error: ‘GTFeature’ does not name a type; did you mean ‘GT1Feature’?
  323 |             ahp_gt_set_feature(0, static_cast<GTFeature>(GTRAGPIOConfigSP.findOnSwitchIndex()));
      |                                               ^~~~~~~~~
      |                                               GT1Feature
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:335:53: error: ‘GTSteppingMode’ does not name a type; did you mean ‘GT1SteppingMode’?
  335 |             ahp_gt_set_stepping_mode(1, static_cast<GTSteppingMode>(GTDESteppingModeSP.findOnSwitchIndex()));
      |                                                     ^~~~~~~~~~~~~~
      |                                                     GT1SteppingMode
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:341:53: error: ‘GTSteppingConfiguration’ does not name a type; did you mean ‘GT1SteppingConfiguration’?
  341 |             ahp_gt_set_stepping_conf(1, static_cast<GTSteppingConfiguration>(GTDEWindingSP.findOnSwitchIndex()));
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                     GT1SteppingConfiguration
/home/nou/astro-soft-stable/indi-3rdparty/indi-eqmod/ahp-gt/ahpgtbase.cpp:347:47: error: ‘GTFeature’ does not name a type; did you mean ‘GT1Feature’?
  347 |             ahp_gt_set_feature(1, static_cast<GTFeature>(GTDEGPIOConfigSP.findOnSwitchIndex()));
      |                                               ^~~~~~~~~
      |                                               GT1Feature
make[2]: *** [indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/build.make:118: indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/ahp-gt/ahpgtbase.cpp.o] Error 1
make[2]: *** Čaká sa na nedokončené úlohy....
/home/nou/astro-soft-stable/indi-3rdparty/indi-ahp-xc/indi_ahp_xc.cpp: In member function ‘virtual bool AHP_XC::Connect()’:
/home/nou/astro-soft-stable/indi-3rdparty/indi-ahp-xc/indi_ahp_xc.cpp:1148:27: error: too few arguments to function ‘int ahp_xc_connect(const char*, int)’
 1148 |     if(0 != ahp_xc_connect(serialConnection->port()))
      |             ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/nou/astro-soft-stable/indi-3rdparty/indi-ahp-xc/indi_ahp_xc.h:25,
                 from /home/nou/astro-soft-stable/indi-3rdparty/indi-ahp-xc/indi_ahp_xc.cpp:36:
/usr/include/ahp/ahp_xc.h:267:16: note: declared here
  267 | DLL_EXPORT int ahp_xc_connect(const char *port, int high_rate);
      |                ^~~~~~~~~~~~~~
```

knro, 2025-04-01:
@iliaplatone How do get this compiled under 24.04? libahp-gt-dev installed? we need to update instructions

nouspiro, 2025-04-01:
ok previously ahp was installed from indi sources and I had them left over previosly. So removing these files installed by indi disable ahp but installing libahp-gt-dev from apt will result in these errors.

iliaplatone, 2025-04-01:
The aho gt library version is old, you should use the upstream version from launchpad ppa:iliaplatone/ppa

nouspiro, 2025-04-01:
Ok it is solved then.

closed 2025-04-01 by nouspiro

nouspiro, 2025-04-01:
Did you compiled previous version of INDI? Because then you most likely have outdated left over include files in $INSTALL_PREFIX/include/ahp. Make sure that you don't have libahp-gt-dev or libahp-xc-dev installed. Delete agp_gt.h and ahp_xc.h from include/ahp and it will work again.

iliaplatone, 2025-04-01:
Debian testing and unstable have the correct version 1.7.3
Ubuntu still uses the 1.6.3, which is older and not working

aaronwmorris, 2025-04-01:
@nouspiro  @iliaplatone No, this was a new install on Debian 12 bookworm stable which apparently has 1.6.1.

nouspiro, 2025-04-01:
Unistall ahp packges. Then check `cmake -L . | grep AHP` it should print `AHP_GT_INCLUDE_DIR:PATH=AHP_GT_INCLUDE_DIR-NOTFOUND` 

Without AHP build is working fine for me.

iliaplatone, 2025-04-01:
@aaronwmorris libahp-gt 1.7.3 is not backward compatible, should depend on 1.7.3 or higher in control

iliaplatone, 2025-04-01:
See PR #1060

aaronwmorris, 2025-04-01:
I appreciate the info.  I am not really blocked by this, it is just a bug that needs to be fixed in the build system.  I tried the change, but it did not really fix the problem, but there may be more going on.

I still think this issue needs to be reopened until the build issue is fixed.

iliaplatone, 2025-04-01:
Try installing v1.7.3 from ppa:iliaplatone/ppa

aaronwmorris, 2025-04-11:
@knro I think the build of this driver should be disabled if the right version of the library is not found.

28446, 2025-04-27:
I think this bug should remain open, until indi-3rdparty compiles.
If the version of libahp-gt in

felipe@sirio:~/Indi$ tar -t -f indi-3rdparty-2.1.3.tar.gz | grep  libahp
indi-3rdparty-2.1.3/debian/libahp-gt/
indi-3rdparty-2.1.3/debian/libahp-gt/changelog
indi-3rdparty-2.1.3/debian/libahp-gt/compat
indi-3rdparty-2.1.3/debian/libahp-gt/control
indi-3rdparty-2.1.3/debian/libahp-gt/copyright
indi-3rdparty-2.1.3/debian/libahp-gt/rules
indi-3rdparty-2.1.3/debian/libahp-gt/source/
indi-3rdparty-2.1.3/debian/libahp-gt/source/format
indi-3rdparty-2.1.3/debian/libahp-xc/
indi-3rdparty-2.1.3/debian/libahp-xc/changelog
indi-3rdparty-2.1.3/debian/libahp-xc/compat
indi-3rdparty-2.1.3/debian/libahp-xc/control
indi-3rdparty-2.1.3/debian/libahp-xc/copyright
indi-3rdparty-2.1.3/debian/libahp-xc/rules
indi-3rdparty-2.1.3/debian/libahp-xc/source/
indi-3rdparty-2.1.3/debian/libahp-xc/source/format

is correct, the error is not due to its version.
To avoid any misunderstanding, I removed all references to this version and installed

libahp-gt1:amd64 (1.7.3-1) and  libahp-gt-dev:amd64 (1.7.3-1)[here are the headers, in Debian at least] from Debian repositories (testing) and the result does not change:

felipe@sirio:~/Indi/indi-3rdparty-2.1.3/build/indi-3rdparty$ make
[  3%] Built target indi_eqmod_telescope
[  3%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/ahp-gt/ahpgtbase.cpp.o
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp: In member function ‘virtual bool AHPGTBase::updateProperties()’:
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:198:52: error: ‘get_max_speed’ was not declared in this scope
  198 |         GTDEConfigurationNP[GT_MAX_SPEED].setValue(get_max_speed(1));
      |                                                    ^~~~~~~~~~~~~
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:199:55: error: ‘get_acceleration_angle’ was not declared in this scope; did you mean ‘ahp_gt_set_acceleration_angle’?
  199 |         GTDEConfigurationNP[GT_ACCELERATION].setValue(get_acceleration_angle(1) * 180.0 / M_PI);
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~
      |                                                       ahp_gt_set_acceleration_angle
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:203:21: error: ‘get_mount_flags’ was not declared in this scope; did you mean ‘ahp_gt_set_mount_flags’?
  203 |         int fork = (get_mount_flags() & isForkMount) ? 1 : 0;
      |                     ^~~~~~~~~~~~~~~
      |                     ahp_gt_set_mount_flags

The compilation error does not appear in the indi_eqmod_telescope target:

felipe@sirio:~/Indi/indi-3rdparty-2.1.3/build/indi-3rdparty$ make indi_eqmod_telescope

[100%] Built target indi_eqmod_telescope

But in indi_ahpgt_telescope:

felipe@sirio:~/Indi/indi-3rdparty-2.1.3/build/indi-3rdparty$ make indi_ahpgt_telescope

[  0%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/ahp-gt/ahpgtbase.cpp.o
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp: In member function ‘virtual bool AHPGTBase::updateProperties()’:
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:198:52: error: ‘get_max_speed’ was not declared in this scope
  198 |         GTDEConfigurationNP[GT_MAX_SPEED].setValue(get_max_speed(1));
      |                                                    ^~~~~~~~~~~~~
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:199:55: error: ‘get_acceleration_angle’ was not declared in this scope; did you mean ‘ahp_gt_set_acceleration_angle’?
  199 |         GTDEConfigurationNP[GT_ACCELERATION].setValue(get_acceleration_angle(1) * 180.0 / M_PI);
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~
      |                                                       ahp_gt_set_acceleration_angle
/home/felipe/Indi/indi-3rdparty-2.1.3/indi-eqmod/ahp-gt/ahpgtbase.cpp:203:21: error: ‘get_mount_flags’ was not declared in this scope; did you mean ‘ahp_gt_set_mount_flags’?
  203 |         int fork = (get_mount_flags() & isForkMount) ? 1 : 0;
      |                     ^~~~~~~~~~~~~~~
      |                     ahp_gt_set_mount_flags
make[3]: *** [indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/build.make:121: indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/ahp-gt/ahpgtbase.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:1432: indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1439: indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/rule] Error 2
make: *** [Makefile:202: indi_ahpgt_telescope] Error 2

Currently, the only way to build indi-3rdparty-2.1.3 is to remove all references to libahp-* from the system:

felipe@sirio:~/Indi/indi-3rdparty-2.1.3/build/indi-3rdparty$ cmake -L . | grep AHP
-- sudo command: /usr/bin/sudo
-- setcap command: /usr/sbin/setcap
AHP_GT_INCLUDE_DIR:PATH=AHP_GT_INCLUDE_DIR-NOTFOUND
WITH_AHP_GT:BOOL=ON
WITH_AHP_XC:BOOL=ON

PS. It's been raining here for several days, so I thought I'd update Indi. I have no way to go out and try the mount control, can anyone confirm that even without indi_ahpgt_telescope, indi_eqmod_telescope works correctly? Otherwise I'll have to think about a roll back.

Thanks