indilib/indi-3rdparty #1063: Touptek and related driver are broken

gy-lehel, 2025-04-04:
in v2.1.3 the touptek driver is broken when it comes to doing exposures with the camera (585).

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Try to take a frame
3. just hangs without logs or anything

**Expected behavior**
As in 2.1.2.1 a frame is taken and showed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: arch linux
 - Version N/A

**Log Files**

gy-lehel, 2025-04-04:
[2025-04-04T22:28:53.669 CEST DEBG ][           org.kde.kstars.indi] - Starting INDI Driver "indi_toupcam_ccd"
[2025-04-04T22:28:53.677 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: FIFO: start indi_toupcam_ccd"
[2025-04-04T22:28:53.677 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: FIFO: Starting driver indi_toupcam_ccd"
[2025-04-04T22:28:53.678 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: pid=137540 rfd=11 wfd=11 efd=12"
[2025-04-04T22:28:53.678 CEST DEBG ][           org.kde.kstars.indi] - Adding managed driver "Touptek"
[2025-04-04T22:28:53.711 CEST DEBG ][           org.kde.kstars.indi] - Received new device ToupTek ATR585C
[2025-04-04T22:28:53.711 CEST DEBG ][           org.kde.kstars.indi] - Driver "Touptek" is adding device ToupTek ATR585C (heuristic match)
[2025-04-04T22:28:53.711 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: snooping on Wave 150i.EQUATORIAL_EOD_COORD"
[2025-04-04T22:28:53.712 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: snooping on Wave 150i.EQUATORIAL_COORD"
[2025-04-04T22:28:53.712 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: snooping on Wave 150i.TELESCOPE_INFO"
[2025-04-04T22:28:53.712 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: snooping on Wave 150i.GEOGRAPHIC_COORD"
[2025-04-04T22:28:53.712 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: snooping on Wave 150i.TELESCOPE_PIER_SIDE"
[2025-04-04T22:28:53.712 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: snooping on SQM.SKY_QUALITY"
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - INDIListener: New device "ToupTek ATR585C"
[2025-04-04T22:28:53.753 CEST INFO ][           org.kde.kstars.ekos] - Ekos received a new device:  "ToupTek ATR585C"
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < CONNECTION >
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < DRIVER_INFO >
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < POLLING_PERIOD >
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < DEBUG >
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < SIMULATION >
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < CONFIG_PROCESS >
[2025-04-04T22:28:53.753 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < ACTIVE_DEVICES >
[2025-04-04T22:28:53.785 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2025-04-04T20:28:53: Driver indi_toupcam_ccd: model: ATR585C, name: ATR585C, maxspeed: 2, preview: 2, maxfanspeed: 1indi_toupcam_ccd dispatch error: oneSwitch requires 'device' attribute"
[2025-04-04T22:28:53.803 CEST INFO ][           org.kde.kstars.indi] - ToupTek ATR585C :  "[INFO] Debug is enabled. "
[2025-04-04T22:28:53.835 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < DEBUG_LEVEL >
[2025-04-04T22:28:53.835 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < LOGGING_LEVEL >
[2025-04-04T22:28:53.836 CEST DEBG ][           org.kde.kstars.indi] - < ToupTek ATR585C >: < LOG_OUTPUT >
[2025-04-04T22:28:53.877 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] Toggle Debug Level -- Alignment Subsystem "
[2025-04-04T22:28:54.164 CEST INFO ][           org.kde.kstars.ekos] - Connecting to "ToupTek ATR585C"
[2025-04-04T22:28:54.422 CEST DEBG ][           org.kde.kstars.ekos] - Connecting  "ToupTek ATR585C"
[2025-04-04T22:28:54.425 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] Using default recorder (SER) "
[2025-04-04T22:28:54.425 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] Using default encoder (RAW) "
[2025-04-04T22:28:54.425 CEST INFO ][           org.kde.kstars.indi] - ToupTek ATR585C :  "[INFO] ToupTek ATR585C connect "
[2025-04-04T22:28:54.426 CEST INFO ][           org.kde.kstars.ekos] - "ToupTek ATR585C" Version: "2.2" Interface: 2 is connected.
[2025-04-04T22:28:54.457 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] Pixel format 100 is supported by SER recorder. "
[2025-04-04T22:28:54.457 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] Pixel format 100 is supported by RAW encoder. "
[2025-04-04T22:28:54.474 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] eventCallBack: 0x00000001 "
[2025-04-04T22:28:54.729 CEST INFO ][           org.kde.kstars.ekos] - "ToupTek ATR585C" is connected and ready.
[2025-04-04T22:28:54.734 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] eventCallBack: 0x00000001 "
[2025-04-04T22:28:54.734 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] eventCallBack: 0x00000001 "
[2025-04-04T22:28:54.797 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] eventCallBack: 0x00000001 "
[... cut ...]
[2025-04-04T22:29:04.624 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] eventCallBack: 0x00000001 "
[2025-04-04T22:29:18.156 CEST INFO ][   org.kde.kstars.ekos.capture] - "[ToupTek ATR585C] CCD capture stopped"
[2025-04-04T22:30:17.427 CEST INFO ][           org.kde.kstars.indi] - ToupTek ATR585C :  "[INFO] FPS rate limit is set to unlimited "
[2025-04-04T22:30:17.429 CEST INFO ][           org.kde.kstars.indi] - ToupTek ATR585C :  "[INFO] FPS rate limit is set to unlimited "
[2025-04-04T22:30:54.646 CEST INFO ][   org.kde.kstars.ekos.capture] - "[ToupTek ATR585C] CCD capture stopped"

gy-lehel, 2025-04-04:
Two tries to take a 1s frame with the camera

balonchiks, 2025-04-05:
Have exactly the same problem after updating ATR585C to the latest v4.49 firmware. Everything has been working fine with the v4.39 firmware

knro, 2025-04-05:
@touptek Please check

gy-lehel, 2025-04-05:
Based on my experience from the past 5 months, they issue firmware and sdk in pairs, there is never cross-functionality, one always has to have a matching pair.
They only offer latest firmware and sdk on their webpage, so a rollback is not possible (found out too late, now I am stuck with it).
Firmware can only be updated from windows.... (big-big minus, one has to buy crappy windoze to do a 3 minute task?).

But foremost: They seem to publish their firmware updates without *any* testing, at least that is the fact that can be deduced based on the experience gained during the last 5 months, since I have this camera. Errors, crashes, faults that are glaringly obvious within seconds after trying to use the updated stuff, strangely are never found before releasing a firmware-sdk pair.

gy-lehel, 2025-04-05:
I already tried reverting 9fee17ef but that did not fix it.

gy-lehel, 2025-04-05:
Also tried downgrading the firmware, unfortunately it is not supported by tupptekkk

balonchiks, 2025-04-05:
After whole day trying I have managed to get it working by installing the latest version of INDI and Kstars from [here](https://gitea.nouspiro.space/nou/astro-soft-build). Make sure to use the build-soft-latest.sh version of the script. This will install latest beta version of Kstars together with INDI and INDI 3rd party drivers which seems to contain the working driver for the Touptek cameras

balonchiks, 2025-04-08:
I've also received a reply from ToupTek this morning:

`please download the SDK package from our official website:https://www.touptek-astro.com/downloads?atfWidgetNav=box_sdk Inside the downloaded package, you’ll find the necessary driver files for various Linux distributions. Please manually replace the corresponding files in your KStars installation directory with those from the SDK package that match your PC system.`

Not sure if this helps you, haven't tried it myself yet

gy-lehel, 2025-04-08:
I removed all "official" packages, and compiled indi and indi-3rdparty from git. For the latter I had to use -DWITH_AHP_GT=Off -DWITH_AHP_XC=Off in order to be able to compile. As of now no change, touptek camera still not working, hangs while taking a frame. Will try the touptek-suggested solution, too.

gy-lehel, 2025-04-08:
I can confirm, by downloading the SDK packages, copying the ligtoupcam.so file from the linux/x64 directory to /usr/lib with a new name, and redirecting the respective symlinks to it, camera started working again.

gy-lehel, 2025-04-08:
Due to having only a raspi3 I am not able to test witn any other architecture, is there someone, who runs this on arm?

gy-lehel, 2025-04-08:
Patching the SDK on a system running 2.1.3 results in a toupcam driver crash.

knro, 2025-04-09:
Can you check if https://github.com/indilib/indi-3rdparty/pull/1067 resolves your issue on x86_64 ?

closed 2025-04-11 by gy-lehel