indilib/indi-3rdparty #1064: SKywatcher 150i excessive guiding speeds, not possible to reduce them

gy-lehel, 2025-04-05:
**Describe the bug**
Skywather 150i mount, indi 2.1.3 and  2.1.2.1 and phd2 2.6.13.

Mount connects, both with generic eqmod and 150i driver. Slewing is good, though error are higher than my old EQ5-OnStep config, star alignment did not really change this.

Starting up phd2 first calibration was successful, took some frames, then came the meridian flip. Calibration was instantly lost (even though both in indi and phd2 was told to re-use it), in the subsequent calibration the first guiding pulse triggers a several degrees runaway eastward movement of the mount. Since then it is impossible to calibrate, as this behavior is now consistent, with both drivers. Setting guiding speed in the is ignored. My guess is, that skywatcher again buggered the eqmod protocol, creating yet another flavor to it.

Indi polar alignment: mount moves double the amount of the pre-set 15/30 degrees with both drivers, before, and after aforementioned observation.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect indi with drivers
2. Start polar alignment from meridian
3. See the mount slew below the horizon for 30 degree setting.

1. Connect indi with drivers
2. Start calibration assistant from PHD2, slew to meridian, start calibration
3. See mount run away with 2..4x estimated guiding speed

**Expected behavior**
Possible calibration, and turning commanded amount of degrees on polar alignment.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: arch linux x86_64
 - Version n/a (rolling release)

**Log Files**

gy-lehel, 2025-04-08:
The mount moves 45 degrees instead of 30, ending up below the horizon during polar alignment. This behavior is consistent both for eqmod, and i150 drivers.

gy-lehel, 2025-04-08:
[2025-04-09T01:09:48.545 CEST INFO ][     org.kde.kstars.ekos.align] - "Starting Polar Alignment Assistant process v3.0 ..."
[2025-04-09T01:09:48.545 CEST INFO ][     org.kde.kstars.ekos.align] - "Clearing mount Alignment Model..."
[2025-04-09T01:09:48.547 CEST DEBG ][           org.kde.kstars.indi] - ISD:Telescope: UnParking... 

[2025-04-09T01:09:48.549 CEST DEBG ][     org.kde.kstars.ekos.align] - Capture&Solve - Target RA: "13:39:51"  DE: "+00° 24' 33\""
[2025-04-09T01:09:48.549 CEST DEBG ][     org.kde.kstars.ekos.align] - Capture&Solve - Destination RA: "13:39:51"  DE: "+00° 24' 33\""
[2025-04-09T01:09:48.549 CEST DEBG ][     org.kde.kstars.ekos.align] - Align state changed to "In Progress"
[2025-04-09T01:09:48.549 CEST DEBG ][   org.kde.kstars.ekos.capture] - Align State changed from "Idle" to "In Progress"
[2025-04-09T01:09:48.549 CEST DEBG ][   org.kde.kstars.ekos.capture] - Setting resume alignment after flip to true
[2025-04-09T01:09:48.551 CEST INFO ][     org.kde.kstars.ekos.align] - "Capturing image..."
[2025-04-09T01:09:48.592 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "Align: list cleared "
[2025-04-09T01:09:48.593 CEST INFO ][           org.kde.kstars.indi] - Wave 150i :  "[INFO] Telescope already unparked. "
[2025-04-09T01:09:48.594 CEST DEBG ][           org.kde.kstars.indi] - ToupTek ATR585C : "[DEBUG] eventCallBack: 0x00000001 "
[2025-04-09T01:09:49.291 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] Compute local time: lst=13.59817519 (13:35:53.43) - julian date=2460774.46514245 "
[2025-04-09T01:09:49.292 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":j1\", 4 bytes written "
[2025-04-09T01:09:49.292 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=A6A18E\", 8 bytes read "
[2025-04-09T01:09:49.293 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] GetRAEncoder() = 9347494 "
[2025-04-09T01:09:49.293 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":j2\", 4 bytes written "
[2025-04-09T01:09:49.293 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=A60F80\", 8 bytes read "
[2025-04-09T01:09:49.293 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] Current encoders RA=9347494 DE=8392614 "
[2025-04-09T01:09:49.294 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":f1\", 4 bytes written "
[2025-04-09T01:09:49.294 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=111\", 5 bytes read "
[2025-04-09T01:09:49.294 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":f2\", 4 bytes written "
[2025-04-09T01:09:49.294 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2025-04-09T01:09:49.294 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[DEBUG] Scope RA (13:39:52) DE ( 0:24:33) Aligned RA (13:39:52) DE ( 0:24:33) AZ (178:38:30) ALT (42:54:58), PierSide (West) "
[2025-04-09T01:09:50.294 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] Compute local time: lst=13.59845445 (13:35:54.44) - julian date=2460774.46515406 "
[2025-04-09T01:09:50.295 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":j1\", 4 bytes written "
[2025-04-09T01:09:50.295 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=D3A18E\", 8 bytes read "
[2025-04-09T01:09:50.295 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] GetRAEncoder() = 9347539 "
[2025-04-09T01:09:50.295 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":j2\", 4 bytes written "
[2025-04-09T01:09:50.295 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=A60F80\", 8 bytes read "
[2025-04-09T01:09:50.296 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] Current encoders RA=9347539 DE=8392614 "
[2025-04-09T01:09:50.296 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":f1\", 4 bytes written "
[2025-04-09T01:09:50.296 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=111\", 5 bytes read "
[2025-04-09T01:09:50.296 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":f2\", 4 bytes written "
[2025-04-09T01:09:50.296 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2025-04-09T01:09:50.297 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[DEBUG] Scope RA (13:39:52) DE ( 0:24:33) Aligned RA (13:39:52) DE ( 0:24:33) AZ (178:38:51) ALT (42:54:58), PierSide (West) "
[2025-04-09T01:09:51.297 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] Compute local time: lst=13.59873378 (13:35:55.44) - julian date=2460774.46516566 "
[2025-04-09T01:09:51.298 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":j1\", 4 bytes written "
[2025-04-09T01:09:51.298 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] read_eqmod: \"=00A28E\", 8 bytes read "
[2025-04-09T01:09:51.298 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[SCOPE] GetRAEncoder() = 9347584 "
[2025-04-09T01:09:51.298 CEST DEBG ][           org.kde.kstars.indi] - Wave 150i : "[COMM] dispatch_command: \":j2\", 4 bytes written "
[... cut ...]
[2025-04-09T01:10:10.368 CEST DEBG ][           org.kde.kstars.indi] - Image received. Mode: "Align" Size: 16594560
[2025-04-09T01:10:10.483 CEST INFO ][     org.kde.kstars.ekos.align] - "Image received."
[2025-04-09T01:10:10.574 CEST DEBG ][     org.kde.kstars.ekos.align] - Align state changed to "In Progress"
[2025-04-09T01:10:10.574 CEST DEBG ][   org.kde.kstars.ekos.capture] - Setting resume alignment after flip to true
[2025-04-09T01:10:10.688 CEST INFO ][     org.kde.kstars.ekos.align] - Stopping Polar Alignment Assistant process...

knro, 2025-04-09:
Let's clarify one thing. Are we talking about **Slew** speeds or **Guide** speeds, or both??

gy-lehel, 2025-04-09:
We are talking abot two different things:
- When trying to do polar alignment, mount consistently moves more, than the set value. There are two options 15 and 30 degrees of motion. When choosing 15, the movement is approximately 25-30 degrees per step. For 30 degrees the movement is 45-50 degrees. The logs above cover such a failed polar alignment (2nd slew ended up pointing the mount into the wall)
- When starting a calibration with phd2, the mount starts to move with a very "high" speed, the distance covered usually with 4+ guiding steps is covered with a single one, resulting in a failed calibration. There were instances, when it was possible to calibrate, and the mount behaved normally, but suddenly, after a meridian flit this behavior popped up, and ever since then, it is consistent. According both to the synscan app, and the hand controller guiding speed is set to 0.5x.
- I also observed occasions, when the pre-set slew speed of the moint was ignored, and the mount slewed with its maximum speed, even though the requested speed was lower I do not have logs of that instance.

I tested both the eqmod generic driver, and the 150i driver, the behavios seemed to be consistent.

knro, 2025-04-09:
Ok I'm not sure why it behaves this way. Skywatcher added new commands, but I think existing commands should work as expected. Maybe now the speed is different, perhaps don't select "Max speed" in PAA, but select a lower slew speed? However, the guide issue is quite odd. You check the guide rate in INDI Control Panel?

Try to send a pulse from Guide module in Ekos and watch the changes in RA/DE to get a sense of high fast it is responding to the guide pulses.

gy-lehel, 2025-04-09:
For the excessive lew, while polar aligning: it seems to be so, that the alignment algorithm makes a wrong assumption of the speed of the slew, when the speed is set to "custom" in the drop-down. When set to a concrete value, the amount is as expected.

The guide speed behavior I was not yet able to reproduce this evening with the 150i driver