indilib/indi-3rdparty #1066: Support mount Celestron Advanced GT (CG5) via WiFi

svalat, 2025-04-05:
**Did you make sure there is no existing driver for your device? Please check [supported devices portal](https://indilib.org/devices].**

There is already CelestronAUX but it requires some patches which I discuss here (not yet at the end of the process, need help to fix an issue).

**Provide link to product page**

The Celestron CG5 / Advanced GT is an old german familly mount from Celestron.
 
 - [Provide URL to the product description and details.](https://www.celestron.com/products/cg-5-computerized-mount#description)
 - https://www.astroshop.de/fr/equatorial-avec-goto/celestron-monture-as-gt/p,7946
 
Manual is here : 

- https://s3.amazonaws.com/celestron-site-support-files/support_files/1145296473_advancedsgt.pdf

I'm using the Celestron WiFi dongle to connect the mount to my PC : 

 - <https://www.celestron.com/products/skyportal-wifi-module?_pos=1&_sid=5a9de468e&_ss=r>

**Describe device family**

* Mounts

**Provide device communication protocol**

Already supported by CelestronAUX.

**Hardware Sample**

No.

**Status**

After patching a bit the driver to add the required values : 

```diff
From 45a7d506730cbe43256a2b63ba18da59e68bb8ef Mon Sep 17 00:00:00 2001
From: Sebastien Valat <sebastien.valat.dev@orange.fr>
Date: Sat, 5 Apr 2025 15:40:55 +0200
Subject: [PATCH] Add support for Mount Celestron Advanced GT (CG5)

---
 indi-celestronaux/celestronaux.cpp            | 15 ++++++++++++---
 indi-celestronaux/celestronaux.h              |  3 ++-
 indi-celestronaux/indi_celestronaux.xml.cmake |  8 ++++++++
 3 files changed, 22 insertions(+), 4 deletions(-)

diff --git a/indi-celestronaux/celestronaux.cpp b/indi-celestronaux/celestronaux.cpp
index e3f25204..3d2a2fee 100644
--- a/indi-celestronaux/celestronaux.cpp
+++ b/indi-celestronaux/celestronaux.cpp
@@ -232,6 +232,7 @@ bool CelestronAUX::initProperties()
     if (strstr(getDeviceName(), "CGX") ||
             strstr(getDeviceName(), "CGEM") ||
             strstr(getDeviceName(), "Advanced VX") ||
+            strstr(getDeviceName(), "Advanced GT") ||
             strstr(getDeviceName(), "Wedge"))
     {
         // Force equatorial for such mounts
@@ -240,10 +241,13 @@ bool CelestronAUX::initProperties()
             m_MountType = EQ_FORK;
     }
 
-    if (m_MountType == ALT_AZ)
+    if (m_MountType == ALT_AZ) {
+       fprintf(stderr, "=====> SET ZENITH %s\n", getDeviceName());
         SetApproximateMountAlignment(ZENITH);
-    else
+    } else {
+       fprintf(stderr, "=====> SET NORTH CELESTO POLE\n");
         SetApproximateMountAlignment(m_Location.latitude >= 0 ? NORTH_CELESTIAL_POLE : SOUTH_CELESTIAL_POLE);
+    }
 
     //    MountTypeSP[ALT_AZ].fill("ALTAZ", "AltAz", m_MountType == ALT_AZ ? ISS_ON : ISS_OFF);
     //    MountTypeSP[EQ_FORK].fill("FORK", "EQ Fork", m_MountType == EQ_FORK ? ISS_ON : ISS_OFF);
@@ -456,8 +460,13 @@ void CelestronAUX::formatModelString(char *s, int n, uint16_t model)
         snprintf(s, n, "Nexstar Evolution");
     else if (model == MountVersion::CGX)
         snprintf(s, n, "CGX");
-    else
+    else if (model == MountVersion::Advanced_GT) {
+        snprintf(s, n, "Advanced GT");
+        fprintf(stderr, "===============> Advanced GT\n");
+    } else {
+        fprintf(stderr, "===============> DEVICE = %d = %x\n", model, model);
         snprintf(s, n, "Unknown");
+    }
     /* TODO: Missing xx needs to be infered.
       0x05xx : 'CGE',
       0x06xx : 'Advanced GT'
diff --git a/indi-celestronaux/celestronaux.h b/indi-celestronaux/celestronaux.h
index a5194c41..9db10896 100644
--- a/indi-celestronaux/celestronaux.h
+++ b/indi-celestronaux/celestronaux.h
@@ -92,7 +92,8 @@ class CelestronAUX :
             Series_GT         = 0x1283,
             AVX               = 0x1485,
             Evolution_Nexstar = 0x1687,
-            CGX               = 0x1788
+            CGX               = 0x1788,
+            Advanced_GT       = 0x0682
         };
 
         // Previous motion direction
diff --git a/indi-celestronaux/indi_celestronaux.xml.cmake b/indi-celestronaux/indi_celestronaux.xml.cmake
index 160daf4f..91fe1c7b 100644
--- a/indi-celestronaux/indi_celestronaux.xml.cmake
+++ b/indi-celestronaux/indi_celestronaux.xml.cmake
@@ -41,5 +41,13 @@
       <driver name="Celestron AUX">indi_celestron_aux</driver>
       <version>@CAUX_VERSION_MAJOR@.@CAUX_VERSION_MINOR@</version>
    </device>
+   <device label="Celestron Advanced GT Wired" manufacturer="Celestron">
+      <driver name="Celestron AUX">indi_celestron_aux</driver>
+      <version>@CAUX_VERSION_MAJOR@.@CAUX_VERSION_MINOR@</version>
+   </device>
+   <device label="Celestron Advanced GT WiFi" manufacturer="Celestron">
+      <driver name="Celestron AUX">indi_celestron_aux</driver>
+      <version>@CAUX_VERSION_MAJOR@.@CAUX_VERSION_MINOR@</version>
+   </device>
 </devGroup>
 </driversList>
-- 
2.43.0
```

I'am able to connect to the mount, looks to get it in place by using the joint patch, but currently failing to turn over AXIS1. From home position (to polar) I cannon run in one direction along axis1 (AD).

Looks a problem linked to the limits / encoder.  At the INIT they are in the panel of EKOS **"mount info"** :

```
Encoder_axis_1 = 0
Angle_axis_1 = 0 or 360
```

I think this is the root of the issue, when I go in the blocked direction encoder loopack from 0 to max and angle go from 0 to 360. But I don't yet find my way to solve it. As you might know the driver better than me, **can I get a bit of help ?**

[image]

svalat, 2025-04-05:
I tried patching the angles, it looks to be related to the encoder value, not the angle.

svalat, 2025-04-12:
Hum, looks there is an error I didn't noticed before when starting the driver : 

```
Dispatch command error(-1): INDI: Could not find property HORIZONTAL_COORD in Celestron Advanced GT WiFi
<setNumberVector device="Celestron Advanced GT WiFi" name="HORIZONTAL_COORD" state="Idle" timeout="0" timestamp="2025-04-12T14:47:37">
    <oneNumber name="AZ">
360
    </oneNumber>
    <oneNumber name="ALT">
45.399999999999998579
    </oneNumber>
</setNumberVector>
```

svalat, 2025-04-12:
And the problem appear when I try to turn Est (right arrow). I'm in north hemisphere.

svalat, 2025-04-12:
Hum, by searching it looks there might be a command which is missing to be sent to the dongle wifi. 

If I follow the given procedure it looks to work : 

1. Connect with Sky Safari or CWPI on the dongle
2. Make fast align (no star) so it init the mount
3. Disconnect Sky Safari  or CWPI
4. Connect with Ekos/Indi

Then it is OK and the encoders now start at middle value not anymore 0 which solve the issue. Now I can move any direction and use GoTo.

There is probably a command to be send to the dongle so it reset its encoder reading ?

svalat, 2025-04-22:
I progressed, if I follow the procedure : 

**Init procedure CPWI**

- Init with CWPI & quick align (no star)
- Connect with Ekos

**Pointing issue**

Then if I select a star like Alkaïd, it goes to the right position, but in kstars the cursor go in the wrong one, other side of the sky. Same if I point Arcturus after that, the mount goes in right position but Kstars cursor is wrong side of the map.

Notice I didn't yet tried to model the mount. But problem if I want to model it the plate solving will be hard because the point is far away from where the mount really point.

There might be some coordinate issue in the driver, between ALT/AZ & RA/DE or something like this. Or a problem with the initial value of the encoders like reported in previous message. Does it suppose the encoders starts to 0 first ?

**Init procedure with phone**

If I make the same with SkySafari from the phone to init instead of CPWI I got strange results, again problem of encoder value starting to 0 , then block to turn over east side.

I'm not sure if this is the mount which block to turn (the dongle wifi) or kstars.

svalat, 2025-04-22:
**Success night without GOTO**

I succeeded yesterday to make a night by not using the goto but just resyncing the mount by hand + plate solving and using only the array movement to go to the position + the map, then it was fine.

As soon as I used GOTO I loosed my position (again, I didn't yet tried to model, this is probably the problem).

knro, 2025-04-23:
Thank you for making the changes. We can definitely add "Advanced GT" to the code. However, the encoder values returned by the mount should make sense when it is at its home position. When you turned on logging ,what was the response from the mount for both axis? I guess it matches what was reported in the INDI driver.

I have AVX and Evolution, and I noticed that sometimes the encoder values reported is not correct and a power cycle is required. Never used CPWI before setting up any mount so this shouldn't really be required to operate the mount under this driver.

commit ab859c94 (2025-05-11): * Add experimental adaptive tuner for Az-Alt tracking.
* Add Advanced GT support (Thanks to Sébastien Valat for the patch). Fixes #1066
closed 2025-05-11 by knro