indilib/indi-3rdparty #1068: Canon DSLR is left unresponsive after aborting capture

MurrayVK3, 2025-04-10:
**Describe the bug**
My Canon 550D is left in an unresponsive state if I cancel an autofocus in Ekos (nearly always) or abort a capture (sometimes).  The only way to clear the problem is to power cycle the camera and restart indiserver.  

**To Reproduce**
Start an autofocus and click the stop button or allow autofocus to fail itself (eg no stars found).   There is no immediate indication the camera has gone away until I try another capture or disconnect in the inidi control panel.  These actions just hang waiting on the camera.  
In the attached log I started an autofocus, stopped it then clicked capture image which kept restarting. The camera had crashed.

**Expected behavior**
The camera should remain responsive after an abort.

**Desktop (please complete the following information):**
 - OS:  Linux
 - Debian 12 "Bookworm"
 The  system the camera is plugged into and which runs indiserver is a Gigabyte Brix with no desktop, no gvfs etc so I'm pretty sure the USB port is not being grabbed by anyone else

[log_11-53-57.txt.zip](https://github.com/user-attachments/files/19676685/log_11-53-57.txt.zip)

MurrayVK3, 2025-04-10:
I've just realized the focuser was incorrectly set to the camera, not  OnStep.  Same outcome but for completeness here's a B version of the log with the correct hardware.

[log_12-40-14_B.txt.zip](https://github.com/user-attachments/files/19677159/log_12-40-14_B.txt.zip)

MurrayVK3, 2025-04-12:
I get the same error using gphoto2 --capture-image.  Cancel that half way through (CTRL-C) and the camera is left in a non-responsive state.   Looks like a bug in either libgphoto2 or the camera's firmware.

closed 2025-04-12 by MurrayVK3

calhaus, 2025-06-25:
Did you fixed the problem?
It happens to me with a canon rebel t6i (eos750d), single shots works, but if i start a framing sequence from a normal way out in autofocus, after i stop it the camera stops working until i restart the indi driver...