indilib/indi-3rdparty #1073: PlayerOne Poseidon Binning Error

JimTheEngineer1, 2025-04-17:
**Describe the bug**
Last 2-3 nights I have imaged, when I go to stack using AS!4, I get an immediate width mismatch error. It looks like each time 2 frames around the meridian flip record in Bin 2 instead of Bin 1. So I just deleted these frames. I will turn the logs on next session and keep the frames next time. Has anyone else had this issue?

**To Reproduce**
1. Run driver on SMPro using Bin 1
2. Perform meridian flip
3. Check files captures for Bin 2

**Expected behavior**
Continued Bin 1 photos being captured

**Screenshots**
No screenshots yet. Awaiting clear skies

**Desktop (please complete the following information):**
 - OS: StellarMate
 - Version 2.7.54

**Log Files**
No log files yet. Awaiting clear skies.

Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

hiro3110i, 2025-04-18:
@JimTheEngineer1 Thank you for your report.
I haven't been able to test it in a meridian flip situation.

@knro Could there be some correlation between the binning settings in the camera driver and meridian flip?

knro, 2025-04-18:
I don't think there is any relationship there

JimTheEngineer1, 2025-04-18:
The forecasts are telling me no clear skies until 27/4. So I may just turn alignment and focus off and get some frames tonight with the dome slaving off and the scope lens cap on (like I’m taking darks). It’s not going to represent the actual conditions, but maybe it will repeat and I will have some logs and the files that are binned wrong.

JimTheEngineer1, 2025-04-18:
BTW - my stacking program was DSS not AS!4…

hiro3110i, 2025-04-18:
@JimTheEngineer1 Thank you very much for your time.

JimTheEngineer1, 2025-04-19:
I captured 124 dark images last night tracking M51. This was simulated since the skies were cloudy. The  meridian flip occurred around 0836 UTC - I’m in Houston. The shutter was closed the entire time but the dome was tracking as well as the mount. I had to disable focus and alignment routines. All frames captured were correct bin 1. So the issue has either been solved or was due to other commands related to focus or alignment routines that are incidentally done at the meridian flip. The forecasts show cloudy skies until at least Apr 28.

JimTheEngineer1, 2025-04-19:
SMPro wants me to do an update. I won’t do this unless you would like me to. Not sure what’s in the latest update.

JimTheEngineer1, 2025-04-22:
What is the correct INDI driver to use with the Poseidon camera?
PlayerOne CCD
PlayerOne Camera 1
PlayerOne Camera 2
PlayerOne Camera 3

JimTheEngineer1, 2025-04-24:
I seem to be using PlayerOne Camera 1 now, but a while ago I was using PlayerOne CCD. I don’t ever recall changing this. So I was wondering if this was the cause. I’m still not seeing clear skies at through May 3, so testing will be delayed.

hiro3110i, 2025-04-24:
@JimTheEngineer1 Much of the code is the same regardless of which driver you use, so if you have any problems, they can be assumed to be common issues.

closed 2026-01-29 by knro