indilib/indi-3rdparty #1077: Help request. ASI power failure to install on Astroberry ASIair pro

Riaandw79, 2025-04-22:
Hi there.

I have attempted all methods found with web search. Somehow all fail to install Asi power to astroberry running on an ASIair pro (RPI 4).
I have no Linux experience and appart from reading the errors I do not know how to resolve them. Any assistance appreciated. I'm sure I'm missing something simple. Feel free to rip me a new one.

**For normal use install from the INDI PPA with:

`sudo apt install indi-asi-power`
This is dependent on the pigpiod daemon which normally runs as a systemd service. The daemon and service are installed with INDI package libpigpiod

To build from source you need to install required libraries and headers before compiling. See [INDI site](http://indilib.org/download.html) for more details. In addition you need to install pigpio client libraries and headers

`sudo apt -y install libpigpiod-if-dev libpigpiod-if2-1 pigpio-tools libpigpiod
`
```
cd ~/Projects
git clone https://github.com/indilib/indi-3rdparty.git
cd indi-3rdparty
```
Compile and install the driver

```
mkdir -p ~/Projects/build/indi-asi-power
cd ~/Projects/build/indi-asi-power
cmake -DCMAKE_INSTALL_PREFIX=/usr ~/Projects/indi-3rdparty/indi-asi-power
make
sudo make install
```**

Here what I get

>_Checking out files: 100% (3033/3033), done.
astroberry@astroberry:~/Projects $ sudo apt -y install libpigpiod-if-dev libpigpiod-if2-1 pigpio-tools libpigpiod
Reading package lists... Done
Building dependency tree
Reading state information... Done
libpigpiod-if-dev is already the newest version (1.79-1+rpt1).
libpigpiod-if2-1 is already the newest version (1.79-1+rpt1).
pigpio-tools is already the newest version (1.79-1+rpt1).
The following packages were automatically installed and are no longer required:
  indi-astroberry-amh libdav1d3 libglu1-mesa libwxgtk3.0-0v5
  skychart-data-dso skychart-data-pictures skychart-data-stars
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libpigpiod
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
1 not fully installed or removed.
Need to get 0 B/113 kB of archives.
After this operation, 362 kB of additional disk space will be used.
(Reading database ... 162455 files and directories currently installed.)
Preparing to unpack .../libpigpiod_0.2+deb10u1_armhf.deb ...
Unpacking libpigpiod (0.2+deb10u1) ...
Replacing files in old package pigpiod (1.79-1+rpt1) ...
dpkg: error processing archive /var/cache/apt/archives/libpigpiod_0.2+deb10u1_armhf.deb (--unpack):
trying to overwrite '/usr/include/pigpio.h', which is also in package libpigpio-dev 1.79-1+rpt1
Errors were encountered while processing:
/var/cache/apt/archives/libpigpiod_0.2+deb10u1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
astroberry@astroberry:~/Projects $ cd indi-3rdparty
astroberry@astroberry:~/Projects/indi-3rdparty $ mkdir -p ~/Projects/build/indi-asi-power
astroberry@astroberry:~/Projects/indi-3rdparty $ cd ~/Projects/build/indi-asi-power
astroberry@astroberry:~/Projects/build/indi-asi-power $ cmake -DCMAKE_INSTALL_PREFIX=/usr ~/Projects/indi-3rdparty/indi-asi-power
-bash: cmake: command not found
astroberry@astroberry:~/Projects/build/indi-asi-power $ cmake -DCMAKE_INSTALL_PREFIX=/usr ~/Projects/indi-3rdparty/indi-asi-power
-bash: cmake: command not found
astroberry@astroberry:~/Projects/build/indi-asi-power $ make
make: *** No targets specified and no makefile found.  Stop.
astroberry@astroberry:~/Projects/build/indi-asi-power $ sudo make install_

Tried this with result.
[... 404 lines cut ...]
install -m 0644 p*.1                           /usr/local/man/man1
install -m 0755 -d                             /usr/local/man/man3
install -m 0644 p*.3                           /usr/local/man/man3
ldconfig
ldconfig: file /usr/lib/arm-linux-gnueabihf/libmtpricoh.so.9 is truncated

ldconfig: file /usr/lib/arm-linux-gnueabihf/libRicohCameraSDKCpp.so.1.1.0 is truncated
ldconfig: file /usr/lib/arm-linux-gnueabihf/libRicohCameraSDKCpp.so is truncated

ldconfig: file /usr/lib/arm-linux-gnueabihf/libmtpricoh.so is truncated
ldconfig: file /usr/lib/arm-linux-gnueabihf/libmtpricoh.so.9.3.0 is truncated

ldconfig: file /usr/lib/arm-linux-gnueabihf/libRicohCameraSDKCpp.so.1 is truncated
astroberry@astroberry:~/pigpio-master $ sudo apt install indi-asi-power
Reading package lists... Done
Building dependency tree
Reading state information... Done
indi-asi-power is already the newest version (0.95+deb10u1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
indi-asi-power : Depends: libpigpiod but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
astroberry@astroberry:~/pigpio-master $ sudo systemctl enable pigpiod
astroberry@astroberry:~/pigpio-master $ sudo systemctl start pigpiod
astroberry@astroberry:~/pigpio-master $_

Riaandw79, 2025-04-22:
I found it... Under Ekos. Drivers/others/asipower

knro, 2025-08-04:
Please note that this driver is no longer maintained.