indilib/indi-3rdparty #1090: USB3 Time out issue on Playerone Poseidon-M

SparkyNZL, 2025-05-18:
**Describe the bug**
When plugged into the USB3 of the SMPro  and when taking a picture from within the Indi Control Panel --> Playerone Poseidon-M Pro --> Main Control,  Pressing Set, exposes correct first time everytime, but then if you press set again to take another exposure, its usually hands at 0.10sec.

This is then epressed when taking images for say PAA, first image works every time, 2nd timesout , again same behaviour when say taking Flats, these might get 3 or 4 images taken, but just about always ends in a timeout issue, this never occurs when plugged into a USB2 port.

If i unplug the camera from the USB3 Port and use a USB2 port, this does not occur. I have may other devices working off USB3 with no issue. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. From within Indi Control Panel
2. Select PlayerOne Poseidon-M Pro tab
3. Select Main Control Tap
4. Press Set (on the Expose line)
5. First image always exposes correctly
6. 2nd image just hangs.
7. Does not do this with USB2

**Expected behavior**
Expose should work every time regardless of which USB2 or USB3 port 

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: SMPro 1.8.8
 - CM4

**Log Files**
No errors in DMESG
time out error in Kstars log

These errors do not occur when plugged into USB2

Have tested and swapped the USB Cables,
Checked noise on the 12v Power supply 
(Have not checked the noise on the power pins on the USB3 port)

hiro3110i, 2025-05-28:
@SparkyNZL Thank you for reporting.
Could you please let me know what version of linux, indilib and PlayerOne driver are you using?
Do you have only one camera connected?
If you have two or more cameras connected at the same time, please check the optical train settings.
Conflicting camera and driver settings may cause operational problems.

I'm not familiar with SMPro and CM4.
I always use Poseidon-C with KStars+Ekos on Raspberry Pi4 (Ubuntu 22.04 Mate and 24.04) / Pi5 (Ubuntu 24.04) .
Is the USB3 bus power sufficient?
If USB3 bus power is configurable, please set it so that sufficient power is supplied.
You can test using a USB3 hub that can supply external power to isolate the problem.

Is 12V power supplied regardless of whether the cooling system is on or off?
Depending on the system, the order in which USB bus power and 12V power are supplied can be an issue.

From past troubleshooting, timeouts are more likely to be caused by hardware than software issues.

SparkyNZL, 2025-05-30:
@hiro3110i Thanks for the reply.

So on further testing it works fine on a Pi4 and a Pi400, but not on a Pi5 or the SMP CM4 based system. I have also raised this with  PlayerOne also.

They are tested with and without equipment load (eg alone, and with full astronomy load of Mount, EAF, Rotator, EFW, Guide Camera. With and without 12v, I check the 12v external on the Oscilloscope and is nice and clean, I have not check for noise on the USB3 ports.

Responds the same way regardless of 1 or 2 Cameras connected

PlayerOne Artemis-M Pro works fine on USB3

As mentioned, the device works fine under USB2

USB3 Power on the SMP is ~400mA, on the Pi4 is ~100mA 

Linux is Debian Bullseye
Stellarmate Pro and StellarMateOS on PI updated to latest version.

I also backed out the SMP to 1.8,1 image, this still had the same issue.

(SMPro = StellarMate Pro)
(CM4 is the Raspberry Compute Module version 4)

I think this answers all your questions, it only every 2nd image that is affected using the method mentioned above.

hiro3110i, 2025-05-31:
@SparkyNZL Thank you for further information.
I met same situation on the RasPi5 with Ubuntu 24.04 (no problems with 23.10 at that time). 
I had been tried on RasPi5 for a mean time and found stable conditions finally.
I have been working with PlayerOne engineers to debug this issue.
The most important point is the latest SDK (which is fixed the issue and using in the latest indi PlayerOne driver).
For your reference, I'm using Poseidon-C with Sedna-M or Neptune-C II (Mono setting).
I'm using Saturn-C with Sedna-M or Neptune-C II in my another settings.

Could you please check again the setting of "FastExposure" which is another important point?
FastExposure is useful for long exposure but your exposure time is 1 second.
In such short exposure time, image download is too late for next exposure.
Poseidon need longer time to download a image due to higher resolution.
Therefore, what may not be a problem with other cameras may be a problem with Poseidon.

hiro3110i, 2025-06-01:
@SparkyNZL Can you check the version of PlayerOne driver which is in the General Info Tab of camera driver?
Second exposure issue for RasPi5 was solved since SDK version 3.8.0 (indi_playerone_ccd version 1.18).

Regarding to RasPi5, we should unlock USB3 bus power limitation.
usb_max_current_enable=1 should be written in the boot settings (/boot/firmware/config.txt).
USB3 on the RasPi5 is very fast, making autofocus and plate solving quick and easy,
but it is more complicated to set up than the RasP4 just now.

SparkyNZL, 2025-06-01:
@hiro3110i  It is 1.19 driver and SDK is 3.8.1 - With these, the issue is still present.

thanks so much.

hiro3110i, 2025-06-02:
@SparkyNZL Thank you for additional testing.
Poseidon-M is something different from my Poseidon-C.
I'll ask to check it to PlayerOne engineer.

By the way, RasPi4 has no issue, right?

SparkyNZL, 2025-06-03:
Correct, I have tested this on a USB3 port on a Pi400 and works fine. 

Cheers

⁣Get BlueMail for Android ​