indilib/indi-3rdparty #1101: indi_toupcam_ccd timeout factor not providing enough buffer for short exposures

tyevans, 2025-06-03:
**Describe the bug**

The check added [here](https://github.com/indilib/indi-3rdparty/commit/4e690584d63380a9c6cf749b2b3fb7fb653f777c#diff-2b8da889bf450ebc2fafa773d270480d8216adf5d9d181a10ba47a54a997e457R1589
) causes touptek cameras of sufficient resolution to always time out at low exposure values:

```
        if (elapsed > m_ExposureRequest * m_TimeoutFactorN.value)
```

(from my own situation) an orion g10 mono running at 8176x5616 generates a 42MB fits file.  
Interfacing that with an n150 powered minipc the normal download time sits in the 400-900ms range.

with a timeout factor of 1.3 (the max allowed), that provides an optimistic lower bounds of 300ms for exposures and an observed bounds closer to 1s.

This issue does not occur on long exposures at all, for obvious reasons.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver on a touptek camera with sufficient resolution (?)
2. attempt to capture an image with an exposure in the 100-500ms range
3. Note the failure in the console

**Expected behavior**
Touptek users should be able to take short exposures.

I can add logs or anything else necessary, but hopefully the issue makes enough sense.

A lower bounds should probably be put in place (though I notice this check didn't exist at all in the previous approach).

knro, 2025-06-03:
Thank you. Perhaps a quick fix is to have an std::max(1, m_ExposureRequest * m_TimeoutFactorN.value) so that it never reports timeout under 1 second

tyevans, 2025-06-05:
@knro that fixes the issue for sure.  sorry for the wait.  appreciate the quick turn around on the fix!

mireianievas, 2025-06-22:
Sorry but this is not fixed. Even with the maximum factor (1.3) I still get continuous timeouts on the 462m (which on the guide module seem to be ignored, luckily) and when using short exposures of a few seconds also on the 2600C. Unfortunately, this really brings it to almost impossible to take bias or flats, unless done in short interactive batches.