indilib/indi-3rdparty #1108: Touptek driver became defunct with 2.1.4

gy-lehel, 2025-06-21:
**Describe the bug**
Touptek camera driver broke again in 2.1.4 exposures are possible to start, but never finish successfully `indi_toupcam_ccd`

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Take a shot
3. See system trying to take the shot several times, then abort

In my experience touptek issues firmware and driver in pairs, and they seem to prefer implementing NBC-s in both. They also prohibit a reversal of the in-camera firmware, once installed a newr one, one is stuck there. 57.26598.20240928 seems to be completely defunct with the 2025 january firmware.

Is there any way we might debug, and fix the driver instead of waiting on yet another trashy drop from touptek? Is there any debugging guide, etc that I could make use of, and activerly take part in fixing this?

**Expected behavior**
Shot taken, and saved to the disk

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: arch linux
 - Version 2.1.4

**Log Files**
To be added later, they do not contain anything useful...

mireianievas, 2025-06-22:
I think this is related to #1101 . A proper fix should probably come from @touptek , but as a quick fix I think increasing that max timeout factor to something absurdly high (1.3 is ok for the guide cam it seems, but insufficient for a IMX571 sensor) should do the trick. Let me try :)

mireianievas, 2025-06-22:
Actually with @knro  's changes in the main / dev branch it works fine. They can be easily backported to the 2.1.4 (I did) if you want to keep the rest of the indi code in the released tag.

gy-lehel, 2025-06-22:
Definitely a different issue: trying to take a 2 sec exposure, it starts, but never finishes. No logs are written.

TallFurryMan, 2025-06-22:
Would it be possible for you both to provide the duration of the exposure and the duration of the capture (that is, exposure plus frame download)? It appears the SDK will count frame download inside the exposure duration, which will require a constant additional delay to the timeout factor. If the timeout factor that I introduced causes issues and that @knro extended, I will implement a way to disable it (say when the factor is 1) and make that the default.

gy-lehel, 2025-06-22:
I am pretty sure, that the issue I see in 2.1.4 is similar to the one I reported in 2.1.3, and is related to sdk vs firmware incompatibility in touptek domain. Unfortunately the touptek firmware of the 585 models got updated at least 4 times since release (only camera to have so many updates since release), and in all cases earlier SDK-s stopped working, one had to switch to the latest. As of now the latest FW is from jan 26.