indilib/indi-3rdparty #1109: Sequence hangs after aborting exposure (Canon EOS 750D / T6i)

calhaus, 2025-06-25:
### Bug: Sequence hangs after aborting exposure (Canon EOS 750D / T6i)

**Summary:**  
When using the INDI gphoto driver with my Canon EOS 750D / Rebel T6i, aborting an exposure sequence (using Ekos or INDI Control Panel) causes the driver to get stuck. After aborting, single exposures still work, but any further sequence (multi-frame capture) hangs indefinitely until I restart the INDI driver. Using the same camera with other programs (like OpenLiveStacker) via gphoto2/libgphoto2 does not show this problem.

**How to reproduce:**
1. Start a multi-frame capture (sequence) in Ekos/INDI with the Canon EOS 750D.
2. Abort the sequence before it finishes.
3. Try to start a new sequence (multi-frame capture).
4. The driver hangs and does not trigger the sequence. Only restarting the driver solves the issue.
5. Single frame capture still works after aborting.

**Expected:**  
After aborting a sequence, it should be possible to start a new sequence without restarting the driver.

**Observed:**  
After abort, only single exposures work. Any new sequence hangs until the INDI driver is restarted.

**Notes:**
- This does not happen with other software (e.g., OpenLiveStacker) using gphoto2 directly.
- The camera does not need to be power cycled; only the INDI driver needs to be restarted.
- The issue seems to be in the driver's internal state after an abort, rather than libgphoto2 or the camera firmware.

---

### Technical Root Cause (Proposed by github's copilot)

- The driver `indi-gphoto` (`gphoto_driver.cpp`) implements `gphoto_abort_exposure` which signals a thread to abort exposure, but **does not perform a deep reset of the driver's internal state after an abort**.
- If the camera or libgphoto2 does not clean up properly, the internal flags, buffers, or widgets might remain in an inconsistent state, causing subsequent sequences to hang.
- Only a full driver restart clears this state, as observed.

---

### **Proposed Patch / Solution (Proposed by github's copilot)**

A possible fix is to add a function that resets the internal state of the driver after aborting, e.g.:

```c++
void gphoto_reset_state(gphoto_driver *gphoto)
{
    // Reset command flags
    gphoto->command = 0;
    gphoto->is_aborted = false;
    gphoto->bulb_mode = false;

    // Limpa ficheiro de imagem em memória, se existir
    if (gphoto->camerafile)
    {
        gp_file_free(gphoto->camerafile);
        gphoto->camerafile = nullptr;
    }

    // Releitura dos widgets principais para garantir valores atuais
    if (gphoto->exposure_widget)
        gphoto_read_widget(gphoto->exposure_widget);
    if (gphoto->iso_widget)
        gphoto_read_widget(gphoto->iso_widget);
    if (gphoto->format_widget)
        gphoto_read_widget(gphoto->format_widget);
    if (gphoto->bulb_widget)
        gphoto_read_widget(gphoto->bulb_widget);

    // Opcional: revalidar estado de conexão com câmara
    // gp_camera_exit(gphoto->camera, gphoto->context);
    // gp_camera_init(gphoto->camera, gphoto->context);
}
```

Then, call this function at the end of `gphoto_abort_exposure`:

```c++
int gphoto_abort_exposure(gphoto_driver *gphoto)
{
    // ... existing abort logic ...
    gphoto_reset_state(gphoto);
    return GP_OK;
}
```

**This should help ensure the driver is ready for a new sequence after aborting.**

---

**System Info:**  
- Camera: Canon EOS 750D / Rebel T6i
- OS: [your OS/version]
- INDI version: [exact version]
- indi-gphoto version: [exact version/commit]
- libgphoto2 version: [exact version]

**Logs:**  
[Attach logs with debug enabled if possible]

---

Thank you for your work!

calhaus, 2025-06-27:
I would like to propose adding a new property to the indi-gphoto driver, called "Abort Graceful," which would appear in the Main Control tab (just below "Shutter Release"). This property would let users choose the abort behavior for sequence exposures ("framing" mode):

When "Abort Graceful" is enabled (default):
**If the user aborts during a sequence, the driver will allow the current exposure to complete and only then terminate the sequence, behaving like a single exposure abort.**
When "Abort Graceful" is disabled:
The abort command will immediately interrupt the current exposure, matching the current/legacy behavior.
This gives users flexibility: safe and camera-friendly aborts for short exposures, or immediate stops when needed. The implementation is entirely in the driver, so it does not require any changes to Ekos/KStars or the client side.

knro, 2025-06-27:
I'm looking into this today, hopefully no need to introduce any new properties.

calhaus, 2025-06-27:
Thank you, a lot, and thank you for taking the time. 
the property suggestion is because this doesn't happen to everyone, so the abort is working for most people, for me trying this way (no luck compiling for now), so i can try to stop a sequence like it's a single frame, and also if it stops like a single frame for me its ok because i don't do guiding and i use exposures of 30sec max, so the wait to stop would not be that much and single frames work just fine...

calhaus, 2025-06-27:
Tested it now, still the same behavior... :/

knro, 2025-06-28:
Are you using force BULB in Options?

closed 2025-08-04 by knro