indilib/indi-3rdparty #1113: ATIK 314L Not Recognized

paulinskis, 2025-06-29:
**Background**
I've been attempting (so far, unsuccessfully) to make an Atik 314L+ work under Linux (Mint, 22.1). Earlier posts and screenshots from libindi website indicate that the 314 should be supported and work. I've also tested this camera under Windows, and managed to get it to work - however (and this may be the cause of the issue), I've had to specifically select one of the older drivers to get Windows to enumerate the camera as "Atik 314L", instead of "Synoptics USB 2.0 HR Chemi Camera (A)". Bear in mind that I've only tried to use the camera on Windows to help verify that it is operational - I'd love to get it working under Linux.

**Describe the bug**
Running `indi_atik_ccd` under KStars or indiserver does not seem to work with Atik 314L. It seems that the driver is unable to complete the initialisation. The camera is recognized by the driver as "Synoptics USB 2.0 HR Chemi Camera (A)", instead of "Atik 314L" (perhaps some of you who have used this camera/driver before could confirm that this should be the case? Also, I cannot see any driver crashes happening.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver, e.g. under indiserver: `indiserver -vvv indi_atik_ccd`
2. Wait
3. Camera/driver never completes the initialisation (or, if used via KStars, you'd see `"Unable to establish: + Atik. Please ensure the device is connected and powered on."`. Again, I believe the camera is functional as I was able to take exposures on Windows, control the cooler, etc.

**Expected behavior**
Driver to connect to the camera successfully.

**Desktop (please complete the following information):**
 - OS: Linux Mint (22.1)
 - Latest `indi-atik` installed via `apt`: `indi-atik/noble,now 3.0+t202506011248~ubuntu24.04.1 amd64`

**Log Files**

Here's the log output of KStars/Ekos:
```
Resetting Ekos Manager...
"Starting INDI services..."
INDI: Starting local drivers...
Starting INDI Server:  ("-v", "-p", "7624", "-m", "1024", "-r", "0", "-f", "/tmp/indififo454a0d5f") -f "/tmp/indififo454a0d5f"
INDI: INDI Server started locally on port  7624
INDI Server Started?  true
Starting INDI Driver "indi_atik_ccd"
Scheduler Ekos status is 1
Adding managed driver "Atik"
INDIListener: Adding a new client manager to INDI listener..
INDI: Connecting to local INDI server on port  7624  ...
INDI server connected.
"INDI services started on port 7,624."
"2025-06-29T12:44:27: startup: /usr/bin/indiserver -v -p 7624 -m 1024 -r 0 -f /tmp/indififo454a0d5f"
"2025-06-29T12:44:27: listening to port 7624 on fd 5"
"2025-06-29T12:44:27: Local server: listening on local domain at: @/tmp/indiserver"
"2025-06-29T12:44:27: FIFO: start indi_atik_ccd"
"2025-06-29T12:44:27: FIFO: Starting driver indi_atik_ccd"
"2025-06-29T12:44:27: Driver indi_atik_ccd: pid=8298 rfd=9 wfd=9 efd=10"
"2025-06-29T12:44:27: Client 8: new arrival from 127.0.0.1:40396 - welcome!"
""
"2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD"
"2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Telescope Simulator.EQUATORIAL_COORD"
"2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Telescope Simulator.TELESCOPE_INFO"
"2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD"
Received new device Synoptics USB 2.0 HR Chemi Camera (A)
INDI Server:  "2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE"
INDI Server:  "2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE"
INDI Server:  "2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION"
INDI Server:  "2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on Focuser Simulator.FOCUS_TEMPERATURE"
INDI Server:  "2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on CCD Simulator.FILTER_SLOT"
INDI Server:  "2025-06-29T12:44:30: Driver indi_atik_ccd: snooping on CCD Simulator.FILTER_NAME"
INDI Server:  "2025-06-29T12:44:30: "
< Synoptics USB 2.0 HR Chemi Camera (A) >: < DEBUG >
< Synoptics USB 2.0 HR Chemi Camera (A) >: < CONNECTION >
INDI Server:  "Driver indi_atik_ccd: snooping on SQM.SKY_QUALITY"
INDI Server:  ""
Synoptics USB 2.0 HR Chemi Camera (A) >: < DRIVER_INFO >
Synoptics USB 2.0 HR Chemi Camera (A) >: < POLLING_PERIOD >
Synoptics USB 2.0 HR Chemi Camera (A) >: < SIMULATION >
Synoptics USB 2.0 HR Chemi Camera (A) >: < CONFIG_PROCESS >
Synoptics USB 2.0 HR Chemi Camera (A) >: < ACTIVE_DEVICES >
"Unable to establish:\n+ Atik\nPlease ensure the device is connected and powered on."
```

Running `indiserver -vvv indi_atik_ccd` produces the following:
```
[... 72 lines cut ...]
Driver indi_atik_ccd: read defSwitchVector Synoptics USB 2.0 HR Chemi Camera (A) CONFIG_PROCESS Idle rw
 CONFIG_LOAD='Off'
 CONFIG_SAVE='Off'
 CONFIG_DEFAULT='Off'
 CONFIG_PURGE='Off'
Driver indi_atik_ccd: read defSwitchVector Synoptics USB 2.0 HR Chemi Camera (A) CONFIG_PROCESS Idle rw
 CONFIG_LOAD='Off'
 CONFIG_SAVE='Off'
 CONFIG_DEFAULT='Off'
 CONFIG_PURGE='Off'
Driver indi_atik_ccd: read defTextVector Synoptics USB 2.0 HR Chemi Camera (A) ACTIVE_DEVICES Idle rw
 ACTIVE_TELESCOPE='Telescope Simulator'
 ACTIVE_ROTATOR='Rotator Simulator'
 ACTIVE_FOCUSER='Focuser Simulator'
 ACTIVE_FILTER='CCD Simulator'
 ACTIVE_SKYQUALITY='SQM'
Driver indi_atik_ccd: read defTextVector Synoptics USB 2.0 HR Chemi Camera (A) ACTIVE_DEVICES Idle rw
 ACTIVE_TELESCOPE='Telescope Simulator'
 ACTIVE_ROTATOR='Rotator Simulator'
 ACTIVE_FOCUSER='Focuser Simulator'
 ACTIVE_FILTER='CCD Simulator'
 ACTIVE_SKYQUALITY='SQM'
```

Please let me know what other information would be useful! Thanks.

knro, 2025-07-02:
Weird, why would it detect the camera as `Synoptics USB 2.0 HR Chemi Camera (A)` which is what is reported in the logs above? @TallFurryMan any ideas?

knro, 2025-07-02:
It's not an Ekos issue. Just the Device name when @paulinskis started the server from console.

TallFurryMan, 2025-07-02:
Well, the device name doesn't seem to prevent the standalone indiserver test from working correctly. So that seems unrelated.

paulinskis, 2025-07-02:
I think I've just found something interesting and am now also starting to believe that the name matters. There are two ways of getting the _name_ of the device/camera.

1. Call `ArtemisDeviceName` - this is what's being done currently, and it returns the above `Synoptics USB 2.0 HR Chemi Camera (A)`. This call can be made prior to running `ArtemisConnect`
2. Call `ArtemisProperties` and check the `Description` property. In my case, I got the _correct_ `ATIK-314L`. However, this can only be done after acquiring a handle by calling `ArtemisConnect` first.

There are two lines in the `indi_atik_ccd.cpp` that push the result of `ArtemisDeviceName` to the rest of system (sorry I'm new to this and not entirely familiar with the inner workings of indi).

```
cameras.push_back(std::unique_ptr<ATIKCCD>(new ATIKCCD(cameraName, i)));
cameraNames.push_back(pName);
```

I've just made a quick change so that the above uses the `Description` returned by `ArtemisProperties` - Ekos connected right away.

[image]

[image]

Now, the question remains, would this change break things for other cameras - @knro @TallFurryMan do any of you have other ATIK cameras that we could use to test this idea?

TallFurryMan, 2025-07-03:
So because the name reported by the driver does not match what Ekos expect from its XML database, the camera is not recognised.

Do we consider this to be a bug in the SDK? That for 314L the name is provided in the description and not in the name? It's entirely possible their new line of products for the medical market got an overlap in their implementation.